raspberrypi/bookworm-feedback #342: Cannot copy/paste text "PACKAGE" on first attempt, double copy operation is required in order to copy it

qrp73, 2024-12-28:
I found strange issue with text editors on Raspi Bookworm Labwc. 
It happens with running clipman to store clipboard content between app sessions.

I'm able to copy/paste usual text, but when I trying to copy/paste text which starts with word "PACKAGE" or with its first letters (even just one letter "P"), it don't works on first attempt but it works if you copy it twice in a row.

Interestingly is that when I trying to copy paste letter P from other words, it works ok. The issue happens when you copy letter P from word PACKAGE. Attempt to copy "ACKAGE" part also works ok. It don't works when you copy first letters of word PACKAGE from the first letter.

I didn't noticed such behavior before today update, so it is possible that it was added with some update.
But there is also [similar bug in Raspi Bookworm Labwc with copy/paste files in pcmanfm](https://github.com/raspberrypi/bookworm-feedback/issues/303), it always need to use double copy in order to be able to paste the file.

## Steps to reproduce
1) Install clipman ```sudo apt install clipman```

2) Start it from console with ```/usr/bin/wl-paste -t text --watch clipman store``` (as alternative you can add this line to ```~/.config/labwc/autostart```: ```/usr/bin/wl-paste -t text --watch clipman store &```

3) Open Text Editor

4) Type two lines:
```
PACKAGE
0123456
```

5) Select 0123 or other text and do clipboard copy operation (just to reset clipboard content)

6) Select text "PACKAGE" and do clipboard copy operation only once (do not do this operation twice in a row)

7) Try to paste it into a new line

**Expected result:** the text "PACKAGE" appears in a new line

**Actual result:** nothing happens - BUG

8) Try to copy text "PACKAGE" for the second time and make sure that now paste operation works

9) Try to select "ACKAGE" text and copy/paste it, make sure it works with first copy operation

10) Try to select "P" letter from word "PACKAGE" and copy paste it, make sure it don't works on first attempt.

qrp73, 2024-12-28:
further investigation shows that executing command
```
$ echo "SOME_TEXT" | clipman store
```
and then trying to do paste text in the text editor works ok

But executing command
```
$ echo "PACKAGE" | clipman store
```
and then trying to do paste text in the text editor doing nothing. 

In both cases the text appears in the history file ```~/.local/share/clipman.json```, but word "PACKAGE" is failed to paste.

So, it looks like something with wayland/labwc configuration is wrong on Raspberry Pi OS Bookworm.

qrp73, 2024-12-28:
further investigation shows that the issue within wl-copy code.

## Steps to reproduce:
1) Open console and execute: 
```$ wl-paste -t text --watch /bin/wl-copy```

2) Open text editor and type two lines:
```
PACKAGE
0123456
```

3) Try to copy and paste word "0123456", make sure it works ok

4) Now try to copy and paste word "PACKAGE"

**Expected result:** the word is copied and pasted ok

**Actual result:** nothing happens - BUG

lurch, 2024-12-29:
~~Unless you can prove otherwise, I'm going to assume that this is a bug in `clipman` rather than a bug in Raspberry Pi OS Bookworm. If you can reproduce this _without_ using clipman, then I'll take a more in-depth look.~~

Ahh, forget that, I've been able to replicate it using just [wl-clipboard](https://github.com/bugaevc/wl-clipboard).
(Oh, it seems that you were also looking at `wl-clipboard` at the same time I was :slightly_smiling_face: )

I was able to produce the following minimal test-case:
1. Have two Terminal windows and one Text Editor window open 
2. In the first Terminal window, type `echo -n PB | wl-copy`
3. In the Text Editor, select Edit -> Paste. Result: `PB` gets pasted into the editor window.
4. In the second Terminal window, type `wl-paste`. Result `PB` gets printed.
5. In the first Terminal window, type `echo -n PA | wl-copy`
6. In the Text Editor, select Edit -> Paste. Result: Nothing happens!
7. In the second Terminal window, type `wl-paste`. Result `PA` gets printed.

Using some of the hints in https://github.com/bugaevc/wl-clipboard/issues/112 I was able to dig a bit deeper; try running `export WAYLAND_DEBUG=1` in the first Terminal window and then repeat the above sequence. In steps 3 and 4 above, the first Terminal window displays `{Default Queue} zwlr_data_control_source_v1#11.send("text/plain;charset=utf-8", fd 4)`. However at step 6 the first Terminal window doesn't display anything, and at step 7 the first Terminal window displays `{Default Queue} zwlr_data_control_source_v1#11.send("application/vnd.hp-HPGL", fd 4)`. This suggests that the Wayland clipboard is identifying `PA` and `PACKAGE` as https://www.fileformat.info/info/mimetype/application/vnd.hp-HPGL/index.htm instead of plain-text. And therefore the Text Editor window is presumably thinking "There's `application/vnd.hp-HPGL` data in the clipboard, but I don't know how to interpret that, so I'm not going to paste anything".

The workaround for this is to force the clipboard-data to be plain-text by doing `echo -n PA | wl-copy --type text/plain` and then you **can** paste `PA` (or `PACKAGE`) into the Text Editor window. So it seems the "fix" for your original problem would be to force `clipman` to flag the copied data as `text/plain` mime-type rather than have it automatically inferring `application/vnd.hp-HPGL` mime-type. And as `clipman` isn't something that we directly recommend (see #303 ), this is something that you'll have to work out for yourself. With a bit of luck, maybe this will also give you a route to solving your other copy'n'paste issue.

closed 2024-12-29 by lurch

qrp73, 2024-12-29:
>> EDIT: And If I change the command from your previous comment to wl-paste -t text/plain --watch /bin/wl-copy then your "PACKAGE" example then works.

NO, IT DON'T FIX IT AT ALL. 

With ```wl-paste -t text/plain --watch /bin/wl-copy``` I get exactly the same issue.

Please reopen the issue.

lurch, 2024-12-29:
> NO, IT DON'T FIX IT AT ALL.

Sorry, my mistake. I just tried writing up a detailed reproduction of this, and it seems that I must have got confused. Although in hindsight, it seems like running `wl-copy` as the result of a `wl-paste --watch` could cause an infinite loop, because every time you run `wl-copy` that'll cause a change to the clipboard-contents, which will trigger the `wl-paste --watch` again? :thinking: 

As I've said before, and as I also said in #303 , the fact that you have to "select copy twice" sounds like a clipman bug, so you should report it to the clipman maintainers. The https://github.com/bugaevc/wl-clipboard/issues/112 that I linked to also implies that there are problems with clipman. (Unfortunately it seems like the [original clipman repo](https://github.com/yory8/clipman) no longer exists. But I found a [fork](https://github.com/chmouel/clipman) that [suggests](https://github.com/chmouel/clipman?tab=readme-ov-file#related-software) that [clipmon](https://git.sr.ht/~whynothugo/clipmon) might be a better alternative? :shrug: )

You might find that running `wl-paste -l` and `wl-paste` helps you to track down what's going wrong with `wl-paste` and `clipman`?

qrp73, 2024-12-29:
> As I've said before, and as I also said in https://github.com/raspberrypi/bookworm-feedback/issues/303 , the fact that you have to "select copy twice" sounds like a clipman bug, so you should report [...]

I wrote my own clipman alternative, but it has the same issue due to a no way to copy paste a plain text with wl-copy/wl-paste, because it leads to issue with some words, like "PACKAGE" and the option --type text/plain don't fix it.