raspberrypi/bookworm-feedback #343: Broken copy/paste functionality (floating bug)

qrp73, 2024-12-29:
I often get some strange behavior when copy/paste functionality fails. 
I discover the issue more deep and found that there is something broken with wl-copy/wl-paste commands.

The issue is floating, so don't expect exactly the same results. You will get different issues on each attempt to do steps to reproduce, but always it will fails to perform copy/paste operation.

## Steps to reproduce:
1) Open console and execute: 
```$ /usr/bin/wl-paste -t text/plain --watch /bin/wl-copy```

2) Open text editor and type two lines:
```
PACKAGE
0123456
```

3) Try to copy and paste word "0123456", make sure it works ok

**Expected result:** the word is copied and pasted ok

**Actual result:** sometimes it works ok, sometimes it paste wrong text, sometimes nothing happens - BUG

4) Now try to copy and paste word "PACKAGE"

**Expected result:** the word is copied and pasted ok

**Actual result:** always nothing happens - BUG

There is something strange with word "PACKAGE", because it always fails to copy it.

See also related issues:
- https://github.com/raspberrypi/bookworm-feedback/issues/303 (the same issue with copy/paste files in pcmanfm)
- https://github.com/raspberrypi/bookworm-feedback/issues/342 (closed by mistake, still reproducible, provided solution is mistaken and don't works)

qrp73, 2024-12-29:
What I'm looking for - is just a way to keep clipboard content when window is closed. For example popup window is open I copy text, move focus to another window and first window is closed (app terminated). I need to paste the text, but it is lost.

There is a clipman app in raspberry pi repository, but there is a problem to copy some strings like "PACKAGE". I wrote my own alternative to clipman but got the same issue. And I don't see the way how to fix it or at least to find some workaround. Because something is going wrong in wl-copy/wl-paste commands.

Consolatis, 2024-12-29:
You are still not giving wl-copy a mime type of text/plain and thus its auto detection kicks and and misrepresents the mime type as debugged by @lurch at https://github.com/raspberrypi/bookworm-feedback/issues/342#issuecomment-2564572411. At the same time you tell wl-paste to only accept the mime type text/plain. Please read the man pages of both.

lurch, 2024-12-29:
> `/usr/bin/wl-paste -t text/plain --watch /bin/wl-copy`
...
sometimes it works ok, sometimes it paste wrong text, sometimes nothing happens

See my comment [here](https://github.com/raspberrypi/bookworm-feedback/issues/342#issuecomment-2564588608)

qrp73, 2024-12-29:
But I don't see the way - how I can provide mime type for wl-copy?

Here is test code with clipman replacement in C#:
```
// sudo apt install mono-runtime
// mcs -out:"clipman.exe" -debug clipman.cs
using System;
using System.IO;
using System.Linq;
using System.Text;
using System.Diagnostics;

class Program
{
    static void Main()
    {
        var text = Console.In.ReadToEnd() ?? string.Empty;
        Console.WriteLine("data: {0}", string.Join("", Encoding.UTF8.GetBytes(text).Select(arg=>arg.ToString("x2"))));
        
        var homeDirectory = Environment.GetFolderPath(Environment.SpecialFolder.UserProfile);
        var filePath = Path.Combine(homeDirectory, ".local", "share", "clipman.txt");
        var textOld = File.ReadAllText(filePath);

        if (text != textOld) {
            File.WriteAllText(filePath, text);
            Console.WriteLine("store");
        }
        serveText("/bin/wl-copy", "", text);
    }
    
    static void serveText(string cmd, string args, string text) {
        using (var process = new Process())
        {
            process.StartInfo.FileName = cmd;
            process.StartInfo.Arguments = args;
            process.StartInfo.RedirectStandardInput = true;
            process.StartInfo.UseShellExecute = false;
            process.Start();
            using (var writer = process.StandardInput)
            {
                writer.Write(text);
            }
            process.WaitForExit();
        }
    }
}
```

Testing:
1) start ```/usr/bin/wl-paste -t text/plain --watch /usr/bin/mono clipman.exe```
2) Copy/paste "0123456" and "PACKAGE" and get exactly the same behavior when "PACKAGE" is failed to copy

I can process any mime type from wl-paste and can pass any mime type to wl-copy, but how I can do that?

And why it fails on a plain text copy/paste?

lurch, 2024-12-29:
> But I don't see the way - how I can provide mime type for wl-copy?

I already answered this [here](https://github.com/raspberrypi/bookworm-feedback/issues/342#issuecomment-2564572411) - `wl-copy --type text/plain`

qrp73, 2024-12-29:
> I already answered this [here](https://github.com/raspberrypi/bookworm-feedback/issues/342#issuecomment-2564572411) - `wl-copy --type text/plain`

This is wrong answer and it don't works. Just try it. 

The command ```wl-copy --type text/plain``` will copy text "wl-copy --type text/plain" into clipboard.

I read manual, but I don't see the way on how to pass mime type to wl-copy. I see such option only for wl-paste.

Is there any way to ask wl-paste to provide some universal format which will include mime type and orignial data from clipboard with no modification/conversion? And to pass this data to wl-copy?

qrp73, 2024-12-29:
It looks that I found the solution. I'm added ```--type text/plain``` option for wl-copy and added check if the text passed from wl-paste is the same as on previous call, if it is the same I just skip the call to wl-copy in order to break the loop.

At a glance, it works and doing all what I need. Still not sure that all is ok, needs to test it a little bit more. But if all is ok, then it can be fixed in clipman code with adding ```--type text/plain``` option and text check to break the call loop. And at a glance it looks that it fix the issue with copy/paste file in pcmanfm. :)

lurch, 2024-12-29:
> This is wrong answer and it don't works. Just try it.

It does work, and I **have** tried it. To give a more fully-expanded version of my test-case in https://github.com/raspberrypi/bookworm-feedback/issues/342#issuecomment-2564572411

Have two Terminal windows and one Text Editor window open
1. In the first Terminal window, type `echo -n PB | wl-copy`
2. In the Text Editor, select Edit -> Paste. Result: `PB` gets pasted into the editor window.
3. In the second Terminal window, type `wl-paste`. Result: `PB` gets printed.
4. In the first Terminal window, type `echo -n PA | wl-copy`
5. In the Text Editor, select Edit -> Paste. Result: Nothing happens!
6. In the second Terminal window, type `wl-paste`. Result: `PA` gets printed.
7. In the first Terminal window, type `echo -n PA | wl-copy -t text`
8. In the Text Editor, select Edit -> Paste. Result: Nothing happens!
9. In the second Terminal window, type `wl-paste`. Result `PA` gets printed.
10. In the first Terminal window, type `echo -n PA | wl-copy -t text/plain`
11. In the Text Editor, select Edit -> Paste. Result: `PA` gets pasted into the editor window.
12. In the second Terminal window, type `wl-paste`. Result `PA` gets printed.
13. In the first Terminal window, type `echo -n PACKAGE | wl-copy -t text/plain`
14. In the Text Editor, select Edit -> Paste. Result: `PACKAGE` gets pasted into the editor window. (which is the exact behaviour you've been asking for all along)
15. In the second Terminal window, type `wl-paste`. Result `PACKAGE` gets printed.

If you also run `wl-paste -l` at various points in the second terminal window, that'll help you see why some versions work and other versions don't.

lurch, 2024-12-29:
If you're storing things that you copy into the clipboard to `~/.local/share/clipman.txt` on disk, then hopefully you never copy sensitive data (e.g. passwords) into your clipboard and hopefully nobody gains access to your Raspberry Pi's SD card :wink:

lurch, 2024-12-29:
Also, I guess if you copied "0123456" to the clipboard and then rebooted, and then tried copying "0123456" to the clipboard again, the clipboard would remain empty?

qrp73, 2024-12-29:
> Also, I guess if you copied "0123456" to the clipboard and then rebooted, and then tried copying "0123456" to the clipboard again, the clipboard would remain empty?

yes, but if you want to keep it between boot sessions, you can call the app from autostart. This is implemented in restore command in clipman. But this doesn't matter for me. I just need to keep clipboard within the same boot session, as it works on windows.

Here is working replacement for clipman, it don't use history, just store single copy of clipboard value:
```
// sudo apt install mono-runtime
// mcs -out:"clipman.exe" -debug clipman.cs
// /usr/bin/wl-paste -t text/plain --watch /usr/bin/mono clipman.exe
using System;
using System.IO;
using System.Linq;
using System.Text;
using System.Diagnostics;

class Program
{
    static void Main()
    {
        var text = Console.In.ReadToEnd() ?? string.Empty;
        
        var homeDirectory = Environment.GetFolderPath(Environment.SpecialFolder.UserProfile);
        var filePath = Path.Combine(homeDirectory, ".local", "share", "clipman.txt");
        var textOld = File.ReadAllText(filePath);

        if (text != textOld) {
            File.WriteAllText(filePath, text);
            serveText("/bin/wl-copy", "--type text/plain", text);
        }
    }
    
    static void serveText(string cmd, string args, string text) {
        using (var process = new Process())
        {
            process.StartInfo.FileName = cmd;
            process.StartInfo.Arguments = args;
            process.StartInfo.RedirectStandardInput = true;
            process.StartInfo.UseShellExecute = false;
            process.Start();
            using (var writer = process.StandardInput)
            {
                writer.Write(text);
            }
            process.WaitForExit();
        }
    }
}
```

Looks like it works good. So, now I can forgot about these issue with lost clipboard value and requirement for double copy. :)

If you can fix clipman source code just add these fixes ("--type text/plain" option and conditional wl-copy call only when text is changed).

qrp73, 2024-12-29:
Update: it seems that it forgot the value in clipboard if I close the editor :(

Sometime the value still remains, sometimes it is lost. Cannot find the reason why it happens...

Consolatis, 2024-12-29:
> and conditional wl-copy call only when text is changed).

An alternative approach for that in a proper clipboard manager (e.g. one that doesn't use wl-copy / wl-paste but implements the [wlr_data_control_unstable_v1](https://gitlab.freedesktop.org/wlroots/wlr-protocols/-/blob/master/unstable/wlr-data-control-unstable-v1.xml) protocol itself) is to announce an additional custom mime-type which allows the clipboard manager to detect and ignore its own offers.

qrp73, 2024-12-29:
> An alternative approach for that in a proper clipboard manager

is there any "proper clipboard manager" for rasbperry pi bookworm labwc?

Consolatis, 2024-12-29:
Basically instead of storing and replacing any offer you see you could only store the last offer and only set it if there is no current offer. I don't know if that is something that is supported by wl-paste though, at least when testing with my old version of it the watch doesn't trigger when closing the application holding the current offer (used `wl-paste --watch date` to confirm, triggers on new offers but doesn't trigger when closing the application holding the last offer).

qrp73, 2024-12-29:
How can I can notify my app when current offer disappears due to app close?

Consolatis, 2024-12-29:
Just tried the date example above again with the master branch of wl-clipboard: that one does actually call the watch command when loosing the offer. Thus the input for your tool on offer disappear should be an empty string if your version is new enough.

qrp73, 2024-12-29:
No, on my version from raspberry pi repository it don't notify when app closed. 

But I found workaround I create lock file before calling wl-copy in order to know next time if this is self feedback or a new event. If text is changed the lock is removed even when exists, otherwise wl-copy is called only when lock doesn't exists otherwise lock is removed and allows to catch a new callback even for the same text. 

Also I replaced ```~/.local/share``` folder with ```/dev/shm``` to avoid unwanted wear on the SD card.

```
// sudo apt install mono-runtime
// mcs -out:"clipman.exe" -debug clipman.cs
// /usr/bin/wl-paste -t text/plain --watch /usr/bin/mono clipman.exe
using System;
using System.IO;
using System.Linq;
using System.Text;
using System.Diagnostics;

class Program
{
    static void Main() {
        var text = Console.In.ReadToEnd() ?? string.Empty;
        Console.WriteLine("notify");
        
        //var homeDirectory = Environment.GetFolderPath(Environment.SpecialFolder.UserProfile);
        //var folder = Path.Combine(homeDirectory, ".local", "share");
        var folder = "/dev/shm";
        var filePath = Path.Combine(folder, "clipman.txt");
        var fileLock = Path.Combine(folder, "clipman.txt.lock");
        
        var textOld = File.Exists(filePath) ? File.ReadAllText(filePath) : null;
        
        if (text != textOld) {
            File.WriteAllText(filePath, text);
            
            if (File.Exists(fileLock)) {
                File.Delete(fileLock);
            }
        }
        if (!File.Exists(fileLock)) {
            File.WriteAllText(fileLock, string.Empty);
            
            serveText("/bin/wl-copy", "--type text/plain", text);
        } else {
            File.Delete(fileLock);
        }
    }
    
    static void serveText(string cmd, string args, string text) {
        using (var process = new Process())
        {
            process.StartInfo.FileName = cmd;
            process.StartInfo.Arguments = args;
            process.StartInfo.RedirectStandardInput = true;
            process.StartInfo.UseShellExecute = false;
            process.Start();
            using (var writer = process.StandardInput)
            {
                writer.Write(text);
            }
            process.WaitForExit();
        }
    }
}
```

Works great :)

lurch, 2024-12-29:
> > An alternative approach for that in a proper clipboard manager
> 
> is there any "proper clipboard manager" for rasbperry pi bookworm labwc?

If you read my comments more carefully, you'll have seen that I mentioned an alternative "clipboard manager" here: https://github.com/raspberrypi/bookworm-feedback/issues/342#issuecomment-2564588608
But I've not tried it myself, as I have no need for such a thing.

> Works great :)

Great, I'll close this and your other "copy paste" issue then :+1:

closed 2024-12-29 by lurch

qrp73, 2024-12-29:
It appears, that ```--type text/plain``` option has a downside effect. 
When I copy text "ТЕСТ" on web-site text box, it will be pasted as "\u0422\u0415\u0421\u0422"...

lurch, 2024-12-29:
> > Great, I'll close this and your other "copy paste" issue then
> 
> Please do not close issues without confirmation that they have been fixed.

I closed _this_ issue because you said "Works great :)" and I closed your _other_ copy-paste issue because you said "And at a glance it looks that it fix the issue with copy/paste file in pcmanfm. :)".

> but as you can see, the bug is still present on Raspberry Pi OS with the latest updates.

As myself and Simon have already told you several times, this isn't a bug in Raspberry Pi OS, it's just the way that the clipboard works in Wayland. If you choose to use a 3rd-party clipboard manager (which isn't even maintained any more) which has bugs, or your own self-written clipboard manager, then it's up to you to deal with those. Despite your claims about this being "the most annoying issue", you're the _only_ person complaining about this, and the solution is simple: `apt remove clipman`.

qrp73, 2024-12-29:
While it may seem like this issue is not widely reported, the core of the problem lies in the fact that most users are unaware of how to approach solving it. Many don’t even reach the point of identifying clipman as a potential solution.

Personally, it took me a considerable amount of time to research and discover that clipman could address the issue, followed by additional effort to troubleshoot and find workarounds for errors in  clipman and wl-paste.

For less technically inclined users, the problem with clipboard persistence may feel insurmountable, as they may not see any viable path to a resolution. This is why I believe the issue deserves more attention, even if it seems to affect only a small subset of users who take the time to report it.