raspberrypi/bookworm-feedback #344: Pressing Alt+Fx keys in virtual terminal leads to a false triggering of Alt+Ctrl+Fx shortcuts (virtual terminal switch)

qrp73, 2024-12-30:
I need to process Alt+F1,Alt+F2,...,Alt+Fx keys in my console application. But for some unknown reason pressing these key shortcuts leads to a false triggering of Alt+Ctrl+F1,Alt+Ctrl+F2,...Alt+Ctrl+Fx keys which switching virtual terminal.

I'm not sure, is this is a bug with Alt+Ctrl+Fx shortcut handling due to ignore Ctrl key state, or there is some config file which defines  duplication of Alt+Ctrl+Fx keys for Alt+Fx keys. 

I don't see where these keys are defined. Can you help?

ghollingworth, 2024-12-31:
Ctrl-Alt-Fn or Alt-Fn are Linux defaults for these things, this isn't a bug...

dumpkeys from the console will show you what is mapped there

closed 2024-12-31 by ghollingworth

qrp73, 2024-12-31:
omg, dumpkeys give me 500 kilobytes output with this (see a small cut below). What is this? 
I just need to disable Alt+Fx shortcuts to not duplicate Alt+Ctrl+Fx on a virtual terminal. How can I do it?
```
keymaps 0-127
keycode   1 = Escape          
	alt	keycode   1 = Meta_Escape     
	shift	alt	keycode   1 = Meta_Escape     
	altgr	alt	keycode   1 = Meta_Escape     
	shift	altgr	alt	keycode   1 = Meta_Escape     
	control	alt	keycode   1 = Meta_Escape     
	shift	control	alt	keycode   1 = Meta_Escape     
	altgr	control	alt	keycode   1 = Meta_Escape     
	shift	altgr	control	alt	keycode   1 = Meta_Escape     
	alt	shiftl	keycode   1 = Meta_Escape     
	shift	alt	shiftl	keycode   1 = Meta_Escape     
	altgr	alt	shiftl	keycode   1 = Meta_Escape     
	shift	altgr	alt	shiftl	keycode   1 = Meta_Escape     
	control	alt	shiftl	keycode   1 = Meta_Escape     
	shift	control	alt	shiftl	keycode   1 = Meta_Escape     
	altgr	control	alt	shiftl	keycode   1 = Meta_Escape     
	shift	altgr	control	alt	shiftl	keycode   1 = Meta_Escape     
	alt	shiftr	keycode   1 = Meta_Escape     
	shift	alt	shiftr	keycode   1 = Meta_Escape     
	altgr	alt	shiftr	keycode   1 = Meta_Escape     
	shift	altgr	alt	shiftr	keycode   1 = Meta_Escape     
	control	alt	shiftr	keycode   1 = Meta_Escape     
	shift	control	alt	shiftr	keycode   1 = Meta_Escape     
	altgr	control	alt	shiftr	keycode   1 = Meta_Escape     
	shift	altgr	control	alt	shiftr	keycode   1 = Meta_Escape     
	alt	shiftl	shiftr	keycode   1 = Meta_Escape     
	shift	alt	shiftl	shiftr	keycode   1 = Meta_Escape     
	altgr	alt	shiftl	shiftr	keycode   1 = Meta_Escape     
	shift	altgr	alt	shiftl	shiftr	keycode   1 = Meta_Escape     
	control	alt	shiftl	shiftr	keycode   1 = Meta_Escape     
	shift	control	alt	shiftl	shiftr	keycode   1 = Meta_Escape     
	altgr	control	alt	shiftl	shiftr	keycode   1 = Meta_Escape     
	shift	altgr	control	alt	shiftl	shiftr	keycode   1 = Meta_Escape     
	alt	ctrll	keycode   1 = Meta_Escape     
	shift	alt	ctrll	keycode   1 = Meta_Escape     
	altgr	alt	ctrll	keycode   1 = Meta_Escape     
	shift	altgr	alt	ctrll	keycode   1 = Meta_Escape     
	control	alt	ctrll	keycode   1 = Meta_Escape     
	shift	control	alt	ctrll	keycode   1 = Meta_Escape     
	altgr	control	alt	ctrll	keycode   1 = Meta_Escape     
	shift	altgr	control	alt	ctrll	keycode   1 = Meta_Escape     
	alt	shiftl	ctrll	keycode   1 = Meta_Escape     
	shift	alt	shiftl	ctrll	keycode   1 = Meta_Escape     
	altgr	alt	shiftl	ctrll	keycode   1 = Meta_Escape     
	shift	altgr	alt	shiftl	ctrll	keycode   1 = Meta_Escape     
	control	alt	shiftl	ctrll	keycode   1 = Meta_Escape     
	shift	control	alt	shiftl	ctrll	keycode   1 = Meta_Escape     
	altgr	control	alt	shiftl	ctrll	keycode   1 = Meta_Escape     
	shift	altgr	control	alt	shiftl	ctrll	keycode   1 = Meta_Escape     
	alt	shiftr	ctrll	keycode   1 = Meta_Escape     
	shift	alt	shiftr	ctrll	keycode   1 = Meta_Escape     
	altgr	alt	shiftr	ctrll	keycode   1 = Meta_Escape     
	shift	altgr	alt	shiftr	ctrll	keycode   1 = Meta_Escape     
	control	alt	shiftr	ctrll	keycode   1 = Meta_Escape     
	shift	control	alt	shiftr	ctrll	keycode   1 = Meta_Escape     
	altgr	control	alt	shiftr	ctrll	keycode   1 = Meta_Escape     
	shift	altgr	control	alt	shiftr	ctrll	keycode   1 = Meta_Escape     
	alt	shiftl	shiftr	ctrll	keycode   1 = Meta_Escape     
	shift	alt	shiftl	shiftr	ctrll	keycode   1 = Meta_Escape     
	altgr	alt	shiftl	shiftr	ctrll	keycode   1 = Meta_Escape     
	shift	altgr	alt	shiftl	shiftr	ctrll	keycode   1 = Meta_Escape     
	control	alt	shiftl	shiftr	ctrll	keycode   1 = Meta_Escape     
	shift	control	alt	shiftl	shiftr	ctrll	keycode   1 = Meta_Escape     
	altgr	control	alt	shiftl	shiftr	ctrll	keycode   1 = Meta_Escape     
	shift	altgr	control	alt	shiftl	shiftr	ctrll	keycode   1 = Meta_Escape     
keycode   2 = one             
[... 83 lines cut ...]
	shift	altgr	control	alt	shiftr	ctrll	keycode   2 = Meta_exclam     
	shift	shiftl	shiftr	ctrll	keycode   2 = exclam          
	shift	altgr	shiftl	shiftr	ctrll	keycode   2 = exclam          
	shift	control	shiftl	shiftr	ctrll	keycode   2 = exclam          
	shift	altgr	control	shiftl	shiftr	ctrll	keycode   2 = exclam          
	alt	shiftl	shiftr	ctrll	keycode   2 = Meta_one        
	shift	alt	shiftl	shiftr	ctrll	keycode   2 = Meta_exclam     
	altgr	alt	shiftl	shiftr	ctrll	keycode   2 = Meta_one        
	shift	altgr	alt	shiftl	shiftr	ctrll	keycode   2 = Meta_exclam     
	control	alt	shiftl	shiftr	ctrll	keycode   2 = Meta_one        
	shift	control	alt	shiftl	shiftr	ctrll	keycode   2 = Meta_exclam     
	altgr	control	alt	shiftl	shiftr	ctrll	keycode   2 = Meta_one        
	shift	altgr	control	alt	shiftl	shiftr	ctrll	keycode   2 = Meta_exclam     
keycode   3 = two             
	shift	keycode   3 = at              
	shift	altgr	keycode   3 = at              
	control	keycode   3 = nul             
	shift	control	keycode   3 = nul             
	altgr	control	keycode   3 = nul             
	shift	altgr	control	keycode   3 = nul             
	alt	keycode   3 = Meta_two        
	shift	alt	keycode   3 = Meta_at         
	altgr	alt	keycode   3 = Meta_two        
....
```

lurch, 2025-01-01:
As Gordon has already said, this is just standard Linux behaviour. The same thing happens on my x64 Laptop running Ubuntu: if I'm on the graphical desktop then I need to press e.g. Ctrl+Alt+F3 to switch to the third virtual terminal, but once I'm at a virtual terminal I "only" need to press Alt+F4 to switch to the fourth virtual terminal.

qrp73, 2025-01-05:
yes, I see. 
But I'm looking for the way to remove Alt+Fn keys to allow user app using it. 
And I didn't find any kind of info on how to do it. 
Can you please help?

Consolatis, 2025-01-05:
https://unix.stackexchange.com/questions/146060/how-to-disable-alt-arrow-switching-of-virtual-consoles

lurch, 2025-01-05:
> But I'm looking for the way to remove Alt+Fn keys to allow user app using it (the same as it works when running in desktop terminal).
> And I didn't find any kind of info on how to do it.
> Can you please help?

This is the kind of question that you ought to be asking in the [forums](https://forums.raspberrypi.com/), rather than on the `bookworm-feedback` repo.