raspberrypi/bookworm-feedback #345: Continuous error spamming to the log file: "[ERROR] [../src/output-state.c:45] Failed to commit frame", сonsumes too much disk space and damages the SD card's write resource

qrp73, 2024-12-30:
## Steps to reproduce

1) boot to Raspberry Pi Bookworm with Labwc desktop
2) switch to a virtual terminal
3) type: ```tail -f .xsession-errors```

**Expected result:** no continuous error spamming

**Actual result:** the following error appears in the log about 33 times per second: [ERROR] [../src/output-state.c:45] Failed to commit frame

As result, the .xsession-errors is spammed with megabytes of the same error text, сonsumes disk space and damages the SD card's write resource.

ghollingworth, 2024-12-31:
You have not provided any reproduction information, as such I'll close this until it's provided.  Note, step 1 of any reproduction information must be "install a new Raspberry Pi OS (lite / desktop) (32bit / 64 bit)

qrp73, 2024-12-31:
I'm talking about Raspberry Pi OS Bookworm Desktop 64 bit with all latest updates:
```
Linux raspi 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux
```

qrp73, 2024-12-31:
Just tested it on a clean install of Raspberry Pi OS Desktop 64-bit and can reproduce this critical bug.

## Steps to reproduce
1) Install a new Raspberry Pi OS with desktop 64-bit, detailed info about image used for installation:
```
https://www.raspberrypi.com/software/operating-systems/#raspberry-pi-os-64-bit
Raspberry Pi OS with desktop
    Release date: November 19th 2024
    System: 64-bit
    Kernel version: 6.6
    Debian version: 12 (bookworm)
    Size: 1,179MB
    SHA256: ea6e68c48d14c3d78af5471c0b288bbf6522fdd775241f74d8295d106d344300
```
```
$ sha256sum ~/.cache/Raspberry\ Pi/Imager/lastdownload.cache
ea6e68c48d14c3d78af5471c0b288bbf6522fdd775241f74d8295d106d344300
```

2) Reboot and do ```sudo apt install && sudo apt upgrade```

3) Reboot and do ```sudo apt install && sudo apt upgrade``` again

4) Use taskbar update icon to install latest raspberry pi ui mods (which is not installed with sudo apt upgrade for some unknown reason

5) Reboot

6) Open takbar menu "Add / Remove plugins", remove "Update" icon and add "CPU", "CPU Temp" and "GPU" icons

7) switch to virtual terminal and execute ```tail -f .xsession-errors```

**Expected result:** no log spamming with thousands of errors

**Actual result:** the log is spammed with ```[ERROR] [../src/output-state.c:45] Failed to commit frame``` 30 times per second

Note: the same issue happens when you staying on desktop screen and display is going to sleep (display blank).

ghollingworth, 2024-12-31:
Does it happen on a lite image?  Or only on a desktop image when switched to the VT?

qrp73, 2024-12-31:
I didn't tested it on lite version, but since it is related to UI, I think it is relevant for desktop image

ghollingworth, 2024-12-31:
The logging message is coming from labwc which is triggering wlr_output_commit (with no display to commit to).  But the upstream code no longer has this line (this code has been changed recently).

Will have a look at it when we're all back in the office, it's not going to affect most people since they'll be running the desktop.  It only affects you because you've switched to the VC.

Not sure why you're getting 30 per second though, I only get 1 per second...

qrp73, 2024-12-31:
> Will have a look at it when we're all back in the office, it's not going to affect most people since they'll be running the desktop. It only affects you because you've switched to the VC.

The issue affects anyone who using labwc desktop with display sleep or with switching to a virtual terminal.

> Not sure why you're getting 30 per second though, I only get 1 per second...

I found that the issue occurs when you add the "CPU," "GPU," and "CPU Temp" widgets to the taskbar. Without these widgets, this error is raised rarely. However, when these widgets are added to the desktop taskbar, the error is logged about 30 times per second.

I discovered this because I noticed a strange decrease in disk space when the system was idle. The root cause turned out to be the rapid growth of the ```~/.xsession-errors``` file. The rest of the details you know from my posts above.

ghollingworth, 2025-01-06:
It looks like this is an issue across basically any X application as well...  It's not just limited to the panel, try running something like neverball and you'll see it...

We'll have a look at the labwc update to see if that fixes the error report.