raspberrypi/bookworm-feedback #346: Drag and drop from xarchiver to the desktop or to a pcmanfm window fails

probonopd, 2024-12-31:
To extract a file from a zip file, I'd like to double-click the zip file, then drag-and-drop the file to be extracted either on the desktop or on a file manager window.

However, in `Raspberry Pi reference 2024-11-19` this does not work.

As soon as the mouse cursor leaves the xarchiver window, the file icon that xarchiver is showing during a drag operation disappears.

I suspect this somehow has to do with Wayland as all involved windows are native Wayland windows, but I am not sure which component is at fault (Raspberry Pi OS itself, xarchiver, PCManFM, labwc, wlroots, wayland-protocols).

--

Thanks @Consolatis for having helped with the initial analysis.

probonopd, 2024-12-31:
Additional information: Drag-and-drop from a PCManFM window to the desktop works.

lurch, 2025-01-01:
Duplicate of https://github.com/raspberrypi/bookworm-feedback/issues/227
I did a bit of digging, and found these related issues, so this appears to be a wider problem, and not just limited to Raspberry Pi OS ?
https://bugs.kde.org/show_bug.cgi?id=408233
https://gitlab.gnome.org/GNOME/file-roller/-/issues/4
https://gitlab.freedesktop.org/wayland/wayland-protocols/-/issues/89

closed 2025-01-01 by lurch

Consolatis, 2025-01-02:
Wow, this is kind of annoying. Thanks for looking into the underlying issue. Looks like file-roller now intends to use fuse which seems like an overly complicated solution. I'll try to look into this issue over the next months. @probonopd feel free to ping me via IRC when I forget.

In the meantime, this comment from the file-roller issue may help:
> BTW, LXQt File Archiver is able to unpack files using DND, even with Nautilus. Problem solved for me. EDIT: Sadly, it's not perfect, it extracts what's chosen when dragging begins, and then copies files to the target from temporary location. Still better than nothing

probonopd, 2025-01-02:
https://gitlab.freedesktop.org/wayland/wayland-protocols/-/issues/89 really leaves me puzzled - who needs to do what now so that drag-and-drop can get fixed?

Consolatis, 2025-01-10:
Update: xarchiver just implemented a workaround by extracting to a temporary directory and then supplying the links of the extracted files to the drop target. Based on the [commit message](https://github.com/ib/xarchiver/commit/b7716fcc12d35ab4c8c36b94b879fe5e93d69c94) it appears to also work for the desktop but I didn't test it.