raspberrypi/bookworm-feedback #348: /etc/os-release should mention "Raspberry Pi OS" and version

probonopd, 2024-12-31:
`/etc/os-release` should mention "Raspberry Pi OS" and version. Is "Raspberry Pi reference 2024-11-19" the correct way to call the version number?

Currently there is only

```
user@raspberrypi:~ $ cat /etc/os-release 
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
```

There is an auxiliary file but most tools don't make use of it to identify the operating system:

```
user@raspberrypi:~ $ cat /etc/rpi-issue 
Raspberry Pi reference 2024-11-19
Generated using pi-gen, https://github.com/RPi-Distro/pi-gen, 891df1e21ed2b6099a2e6a13e26c91dea44b34d4, stage4
```

lurch, 2025-01-01:
Closing as duplicate of #204

There's no "formal" way of identifying a Raspberry Pi OS version - I usually just use the names and release-dates shown on https://www.raspberrypi.com/software/operating-systems/ e.g. "Raspberry Pi OS (64-bit) November 19th 2024". And I usually treat "Raspberry Pi OS with desktop" the same as "Raspberry Pi OS with desktop and recommended software", i.e. I only make a distinction between the "desktop" and "lite" versions.

closed 2025-01-01 by lurch

probonopd, 2025-01-02:
The thing is, how do I know a few months later which one from the website  I put onto that microSD card?

lurch, 2025-01-02:
> The thing is, how do I know a few months later which one from the website I put onto that microSD card?

I think the most reliable way is by looking at the contents of `/boot/firmware/issue.txt`.
AFAIK `/boot/firmware/issue.txt` doesn't get updated by `apt`:
```
$ dpkg -S /boot/firmware/issue.txt
dpkg-query: no path found matching pattern /boot/firmware/issue.txt
```
but `/etc/os-release` **does** get updated by `apt` (it's part of the `base-files` package):
```
$ dpkg -S /etc/os-release
base-files: /etc/os-release
$ ls -l /etc/os-release 
lrwxrwxrwx 1 root root 21 Oct 31 11:04 /etc/os-release -> ../usr/lib/os-release
$ dpkg -L base-files
/.
/bin
/boot
/dev
/etc
/etc/debian_version
/etc/default
/etc/dpkg
/etc/dpkg/origins
/etc/dpkg/origins/debian
/etc/host.conf
/etc/issue
/etc/issue.net
/etc/profile.d
/etc/skel
/etc/update-motd.d
/etc/update-motd.d/10-uname
/home
/lib
/proc
/root
/run
/sbin
/sys
/tmp
/usr
/usr/bin
/usr/games
/usr/include
/usr/lib
/usr/lib/os-release
/usr/sbin
/usr/share
/usr/share/base-files
/usr/share/base-files/dot.bashrc
/usr/share/base-files/dot.profile
/usr/share/base-files/dot.profile.md5sums
/usr/share/base-files/info.dir
/usr/share/base-files/motd
/usr/share/base-files/profile
/usr/share/base-files/profile.md5sums
/usr/share/base-files/staff-group-for-usr-local
/usr/share/common-licenses
/usr/share/common-licenses/Apache-2.0
/usr/share/common-licenses/Artistic
/usr/share/common-licenses/BSD
/usr/share/common-licenses/CC0-1.0
/usr/share/common-licenses/GFDL-1.2
/usr/share/common-licenses/GFDL-1.3
/usr/share/common-licenses/GPL-1
/usr/share/common-licenses/GPL-2
/usr/share/common-licenses/GPL-3
/usr/share/common-licenses/LGPL-2
/usr/share/common-licenses/LGPL-2.1
/usr/share/common-licenses/LGPL-3
/usr/share/common-licenses/MPL-1.1
/usr/share/common-licenses/MPL-2.0
/usr/share/dict
/usr/share/doc
/usr/share/doc/base-files
/usr/share/doc/base-files/README
/usr/share/doc/base-files/README.FHS
/usr/share/doc/base-files/changelog.gz
/usr/share/doc/base-files/copyright
/usr/share/info
/usr/share/lintian
/usr/share/lintian/overrides
/usr/share/lintian/overrides/base-files
/usr/share/man
/usr/share/misc
/usr/src
/var
/var/backups
/var/cache
/var/lib
/var/lib/dpkg
/var/lib/misc
/var/local
/var/lock
/var/log
/var/run
/var/spool
/var/tmp
/etc/os-release
/usr/share/common-licenses/GFDL
/usr/share/common-licenses/GPL
/usr/share/common-licenses/LGPL
/usr/share/doc/base-files/FAQ
```