raspberrypi/bookworm-feedback #349: update-initramfs error

svavassori, 2025-01-02:
Hello all,

I'm not sure if this is the correct place to report a problem with a script in a package, but in case it isn't, I'd appreciate any redirection to the correct site (I've looked to find something more specific, but I didn't find any).

Yesterday I updated a Raspberry PI I had and all the packages except `initramfs-tools` have been upgraded correctly (`apt-get update`, `apt-get upgrade` and `apt-get dist-upgrade`).
Unfortunately this package [`initramfs-tools_0.142+rpt2+deb12u1_all.deb`](https://archive.raspberrypi.org/debian/pool/main/i/initramfs-tools/initramfs-tools_0.142+rpt2+deb12u1_all.deb) kept failing when running the update script, exiting almost immediately with an exit code of 1.

After some hard time to find the root cause, I was able to track it down the issue and it is related to a raspberry-pi specific change, namely in `/usr/sbin/update-initramfs` this function has been added in `rpt2` version of the package:

```
get_latest_rpi_versions()
{
	version_list="$(
		prev_pref=""
		linux-version list |
		linux-version sort --reverse |
		while read -r version; do
			pref="$(echo $version | grep -o '^.\++rpt-rpi-')"
			if [ -z "$pref" ]; then
				break
			fi
			if [ -n "$prev_pref" ] && [ "$prev_pref" != "$pref" ]; then
				break
			fi
			if [ -e "${BOOTDIR}/initrd.img-$version" ]; then
				echo "$version"
			else
				break
			fi
			prev_pref="$pref"
		done
		)"
	if [ -z "${version_list}" ]; then
		return
	fi
	verbose "Latest RPi versions: ${version_list}"
}
```

I had installed only one kernel with version `6.1.0-rpi7-rpi-v6`, so the echo line is evaluated as: `echo '6.1.0-rpi7-rpi-v6' | grep -o '^.\++rpt-rpi-'` and it doesn't match, this means that `grep` returns an error code of `1` and since it's the last command in the pipeline, it's returned as a pipeline exit error, so the whole script stops in `sh`.

Executing the same script with `bash` instead of `sh` allowed a proper execution and initfs creation, so that was the quick-and-dirty workaround I used.

My suggestion for a proper fix is to change the `grep` part to NOT fail if it doesn't match and change also the regex to match other similar Raspberry PI kernels (like the one I had, `linux-image-6.1.0-rpi7-rpi-v6`).

`pref="$(echo $version | { grep -o '^.\+\(+rpt\|-rpi[0-9]\)-rpi-' || test $? = 1; })`

As a side note, I'd like to ask what's the difference between `rpi` and `rpt` kernels: I read something about Raspberry PI Trading, but I don't know if it's correct to have that as mainline kernel or if I should change my `apt/source.list` files (currently they're pointing to `http://mirrordirector.raspbian.org/raspbian/` and `http://archive.raspberrypi.org/debian/`)

Regards,
Sergio

XECDesign, 2025-01-03:
Thanks for the detailed report. Didn't know about the difference in `set -e` behaviour in `dash`. I think just `|| true` should be enough. I can't think of any reason it could return 2 or why it should be handled differently. Or am I missing something?

`pref="$(echo "$version" | grep -o '^.\+\(+rpt\|-rpi[0-9]\+\)-rpi-' || true)"` seems to work as expected for me.

The reason for different version format is to do with Debian's kernel packaging shenanigans. The explanation is a bit long.

The short version is that between 6.1 and 6.6, Debian's packaging has changed how it handles ABI changes and that affected how we version it as well.

Full version:

Take a look here - https://packages.debian.org/source/bookworm/linux. Notice that the kernel version they ship is 6.1.119, but the binary packages they produce contain 6.1.0-21 in the name.

Since the upstream kernel does not follow semantic versioning, the version number does not provide a way to keep track of the ABI. If it did, tools like DKMS could check whether they need to rebuild modules or not. So Debian decided that they will do it themselves. They take the major and minor version, set the patch version to 0 and append an ABI version. The packaging provides a script to generate the ABI metadata to keep track of changes. When the kernel builds, it checks whether the ABI has changed and throws an error until you manually bump it.

That was confusing, since the version number doesn't map onto the upstream version well and maintaining the ABI version was fiddly. The ABI changes pretty much any time the kernel is updated anyway. The newer packaging just uses the upstream version as the ABI version - https://packages.debian.org/source/trixie/linux

Previously, we followed the same procedure and used rpiN for the ABI version. Now we just add `+rpt` to signify that it's not the same as a stock Debian version with the same version number.

Furthermore, the packaging has the concept of feature sets and flavours. A feature set is an additional set of patches applied on top of the source. Flavours are various config changes. Previously, we applied all the pi-specific patches named 'rpi', but that left the patch out of the linux-source-* packages. Now that patch set is empty and the patches apply at the top level instead. Finally, v6, v7, v7l and so on are the flavours matching the appropriate defconfig options for various pi models.

Putting it all together, you end up with something like `6.6.62+rpt-rpi-v7l` - ABI version (upstream+rpt)-featureset-flavour.

svavassori, 2025-01-03:
I also didn't know the difference in behavior between `bash` and `dash` on that error handling until I just discovered it the other day... :-|

Thanks for the detailed explanation of the kernel versioning, I knew some of them due to Debian and I appreciated your more in-deep clarification.

About `grep`, an exit code of `1` means "no match" and sometimes it can be classified as error or not depending on the context, while `2` is always an error (e.g. unable to read from file). It's true that `|| true` behaves like the one suggested in this context but it will hide the different meaning between _no match is OK_ and _I don't care about any actual error_. You can find more details on this stack overflow [question](https://stackoverflow.com/questions/6550484/prevent-grep-returning-an-error-when-input-doesnt-match/49627999#49627999) that I used as model.

That is the reason for which I went for the more specific one: since it's a bit unusual, I think it will make people searching for that and discover those little differences with respect to the more general `|| true`. Nonetheless I'm perfectly fine with whichever option you prefer.

XECDesign, 2025-01-06:
Fixed version should be up now.

svavassori, 2025-01-07:
BTW, do you know where is the source repository for this raspberry-package?

XECDesign, 2025-01-08:
There's not github repo for it, but the source is in the apt repo.

svavassori, 2025-01-08:
OK, so just for curiosity: how the sources are packaged and built? I had the idea there was something like [debian salsa](https://salsa.debian.org)

svavassori, 2025-01-08:
Yes, it can be closed, thanks for the quick fix

closed 2025-01-08 by lurch