raspberrypi/bookworm-feedback #350: Pi 5, Node-Red, BLE and kernel 6.6.62-1 problem

GravityBendsTime, 2025-01-05:
I spent a couple of days troubleshooting this because I thought docker is the culprit. Finally I made a new OS image and installed node-red locally and found out:

Previous kernel 6.6.51+rpt-rpi-2712 and BLE scanning works fine with Node-Red both locally and in container.

apt full-upgrade installs new kernel 6.6.62 and BLE does not work anymore in Node-Red. Not locally, not in container. 
I also tried some additional noble and scanning nodes but all the same: "Unable to start BLE scan. Adapter state: unknown"

I reflashed the SD card image of 64-bit Bookworm and installed necessary packages for my Pi 5 and before full-upgrade everything works as it used to. After full-upgrade (I suppose that's what selecting update/install in GUI does) and new kernel, again no joy with bluetooth. Therefore I'm confident that the latest kernel is the culprit indeed.

It is noteworthy, that BT interface hci0 seems to be ok with the new kernel when checked with bluez package tools. Only difference I found is output of btmon. When everything works, btmon show some info and starts rapidly feeding the console with HCI events. With the new kernel btmon shows the few static info lines but no LE Advertising Reports or other events whatsoever.

I used apt-mark to hold four packages until there's a fix:
```
$ apt-mark showhold
linux-headers-rpi-2712
linux-headers-rpi-v8
linux-image-rpi-2712
linux-image-rpi-v8
```

And here's the /var/log/apt/history.log which shows what happened just before BLE stopped working with Node-red.
```
Start-Date: 2025-01-05  20:05:22
Commandline: packagekit role='update-packages'
Requested-By: pi (1000)
Install: linux-headers-6.6.62+rpt-rpi-v8:arm64 (1:6.6.62-1+rpt1, automatic), libwlroots-0.18:arm64 (0.18.1-1+rpt6, automatic), linux-headers-6.6.62+rpt-rpi-2712:arm64 (1:6.6.62-1+rpt1, automatic), linux-kbuild-6.6.62+rpt:arm64 (1:6.6.62-1+rpt1, automatic), linux-image-6.6.62+rpt-rpi-v8:arm64 (1:6.6.62-1+rpt1, automatic), linux-headers-6.6.62+rpt-common-rpi:arm64 (1:6.6.62-1+rpt1, automatic), rasputin:arm64 (0.2, automatic), libxcb-errors0:arm64 (1.0.1-2~bpo12+1+rpt1, automatic), linux-image-6.6.62+rpt-rpi-2712:arm64 (1:6.6.62-1+rpt1, automatic)
Upgrade: labwc:arm64 (0.7.4-1+rpt1, 0.8.1-1+rpt1), linux-headers-rpi-v8:arm64 (1:6.6.51-1+rpt3, 1:6.6.62-1+rpt1), raspberrypi-ui-mods:arm64 (1.20241112, 1.20241211), linux-headers-rpi-2712:arm64 (1:6.6.51-1+rpt3, 1:6.6.62-1+rpt1), linux-image-rpi-2712:arm64 (1:6.6.51-1+rpt3, 1:6.6.62-1+rpt1), linux-image-rpi-v8:arm64 (1:6.6.51-1+rpt3, 1:6.6.62-1+rpt1)
```

ghollingworth, 2025-01-06:
It looks like the kernel update has changed some interface that Node-Red uses.  It sounds most likely that the issue is with node-red, it might be worth checking whether there are any updates for it to work with the upstream kernel

GravityBendsTime, 2025-01-06:
> It looks like the kernel update has changed some interface that Node-Red uses. It sounds most likely that the issue is with node-red, it might be worth checking whether there are any updates for it to work with the upstream kernel

That is possible. But that does not explain differences in `btmon` output before and after kernel update. Before update locally, not related to node-red, btmon outputs a bazillion of BT events but after update none. That should be relatively easy to replicate.

ghollingworth, 2025-01-06:
Wouldn't surprise me to find the problem is they fixed a problem

https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=linux-6.6.y&id=830c03e58beb70b99349760f822e505ecb4eeb7e

GravityBendsTime, 2025-01-08:
> Wouldn't surprise me to find the problem is they fixed a problem
> 
> https://git.kernel.org/pub/scm/linux/kernel/git/stable/linux.git/commit/?h=linux-6.6.y&id=830c03e58beb70b99349760f822e505ecb4eeb7e

Thanks for looking into this. So it seems I'm out of luck with my current setup. Nod-red BLE implementation `noble` I have used has apparently been abandoned for a long time so I suppose this was inevitable. Luckily I found another way to do this with a python and dbus based gateway that integrates to pretty much anything, including node-red. 

FWIW, if anyone has this same problem and wants a future proof solution, look for https://gateway.theengs.io/

closed 2025-01-08 by GravityBendsTime