raspberrypi/bookworm-feedback #354: Chromium problems on latest Rpios

martk200, 2025-01-12:
I recently updated my rpi5 to a clean copy of Rpios. Everything worked well except that Chromium browser would not detect my webcam. For me this is important as I use Zoom quite a lot. It worked well with a slightly older version of Rpios.
 i have set permissions for Zoom on Chromium to ask for permission to use the webcam but it does not ask me for permission.
  I have 3 webcams and none of them work. One is a brand new purchase in the hope that it would solve my problems . It didn't. The built in microphone of the webcams work,after granting permission.
   Using the Zoom client under box86 is hopeless the audio from my microphone is unintelligible.
  
   All the webcams work on non browser applications such as Cheese.
   
    I have found a very untidy way to work around the problem by using Vivaldi browser on Zoom only. I don't like it  as my usual browser.
    
     I have put a comment into the rpi forums but only had one response. From someone else who is having the same problem. So it is not just me with the problem.

lurch, 2025-01-14:
Have you also tried using Firefox on Raspberry Pi OS? Does that give you similar problems?

ping @cillian64

martk200, 2025-01-15:
> David, thank you for your help. Your second option seems to have done the
> trick. I deleted the devices file and the webcam now works.
>  Once again thank you.
>
>  Regards
>
> Martin Knowles

Andrew, I also had the problem on Firefox. The webcam worked on Vivaldi
browser, but Chromium is my browser of choice

lurch, 2025-01-15:
@cillian64 Is step 2 likely to be a common "problem" that we ought to consider adding to the [documentation](https://www.raspberrypi.com/documentation/), or is it a fairly rare scenario?

cillian64, 2025-01-15:
> David, thank you for your help. Your second option seems to have done the
trick. I deleted the devices file and the webcam now works.
 Once again thank you.

Glad I could help!

> @cillian64 Is step 2 likely to be a common "problem" that we ought to consider adding to the [documentation](https://www.raspberrypi.com/documentation/), or is it a fairly rare scenario?

When you use the webcam in chromium/firefox there are two separate permission dialogs. One inside the browser to decide if this website can use the webcam, and one *outside* the browser to decide if this application (the browser) should be allowed to use the webcam.  If the latter of those is denied (which is easily done accidentally by closing the dialog or pressing escape) then the permission-denied is remembered and the user is never asked again.

I wouldn't exactly call it a common problem but I did see David P hit this a few weeks back, so it's not unheard of.

Instead of just documenting the "fix" (clearing the permission store) I think a better solution would be to either modify `xdg-permission-store` to not cache refusals (because I think it's very rare someone on rpi-os will be running an untrusted application they never want to allow camera access) or, even better, to update the permission dialog to have "allow once", "allow always", "deny once", "deny always" options and make closing the dialog equivalent to "deny once" instead of "deny always".

lurch, 2025-01-15:
> Instead of just documenting the "fix" (clearing the permission store) I think a better solution would be to either modify `xdg-permission-store` to not cache refusals (because I think it's very rare [...]

Sounds like that's an upstream thing (a quick search finds https://github.com/flatpak/xdg-desktop-portal/wiki/The-Permission-Store and https://github.com/flatpak/xdg-desktop-portal/blob/main/document-portal/xdg-permission-store.c ) so I'll close this issue as the user's original problem has been resolved.

closed 2025-01-15 by lurch

cillian64, 2025-01-15:
Yeh, it's not an rpi-os-specific issue, but as we're basically the only non-GNOME-non-KDE platform actually using pipewire for cameras, I don't think anybody else is going to fix it if we don't.