raspberrypi/bookworm-feedback #355: Screen rotation settings lost on monitor power cycle, but properly set upon boot/reboot

shauntaine, 2025-01-14:
Fresh install of RPi OS and used the screen configuration program to rotate my attached touchscreen display to portrait mode.  All works well and the settings are properly implemented on boot or restart.  However, a smart plug controls when my panel turns on or off.  If the monitor loses power, and then turns back on the display is reset to the default orientation.  This can be fixed either by rebooting or going through the screen configuration utility.  The settings only appear to revert after the monitor looses power together.  Simply using the monitor button to enter standby mode allows for settings to be retained.

lurch, 2025-01-15:
Can you please provide details about which touchscreen / panel / monitor it is that you're actually using? Is it connected over DSI or HDMI? Which model of Raspberry Pi are you using? Are you running the 32-bit or 64-bit version of Raspberry Pi OS Bookworm? Is everything [up to date](https://www.raspberrypi.com/documentation/computers/os.html#update-software) with `apt`?

shauntaine, 2025-01-15:
Apologies, should have included that up front:
Raspberry Pi 4B 8GB RAM
Operating System: Debian GNU/Linux 12 (bookworm)
Kernel: Linux 6.6.69-v8+
Architecture: arm64
Monitor: ASUS VT229H touch screen monitor
Monitor is reported as HDMI 1A in the raindrop screen configuration
Everything is up to date with apt.

All this system is doing is displaying a web browser pointed to a home
assistant dashboard.  The monitor is controlled by smart plug to determine
when the monitor is on or off.

Thanks

lurch, 2025-01-16:
Thanks for the extra info. On a fully up-to-date 64-bit install of Raspberry Pi OS I see kernel `6.6.62+rpt-rpi-v8` - are you using an `rpi-update` kernel?

I'm unable to recreate this problem with either of the Dell monitors I have on my desk at work - I set a 90 degree rotation in RPi OS, switched the monitor on and off using it's power-button, plugged and unplugged the HDMI cable, and plugged and unplugged the monitor's power cable, and in all cases the display comes back on with the same 90 degree rotation that I set. Any ideas @popcornmix or @6by9 ?

@shauntaine Why do you need to use a smart plug to physically power the monitor on and off? Would using the `wlopm` command to get the monitor to power on and off be a suitable solution? In the meantime, you can run e.g. `wlr-randr --output HDMI-A-1 --transform 90` to rotate your display, and that might be quicker (if you save it into a script file) than using the screen configuration GUI.

6by9, 2025-01-16:
Please confirm you are using labwc, and not X11.

We really need to know what events are being triggered by the monitor losing power.
The easiest initial thing is probably to run `udevadm monitor` over SSH, disconnect the monitor power, and post the output here. Stop `udevadm monitor` with ctrl-c, and restart it again. Connect the monitor power, and post the output here.

Followup step would be to do the same with `dmesg -w` to monitor the kernel logs, however the default logging settings are likely to be missing some of the required detail.

shauntaine, 2025-01-30:
Apologies for the delay.
Here is what I've done to remove the variables of my own unsupervised
troubleshooting.
1) Reinstalled Pi OS fresh.
2) Monitor rotation set in the screen configuration tool.
3) Screen rotation settings survive a reboot, but not power cycyling via a
smart plug.

udevadm monitor
Monitor power turned off
KERNEL[1108428.802782] change   /devices/platform/gpu/drm/card1 (drm)
UDEV  [1108428.821405] change   /devices/platform/gpu/drm/card1 (drm)
KERNEL[1108429.842170] change   /devices/platform/gpu/drm/card1 (drm)
UDEV  [1108429.844559] change   /devices/platform/gpu/drm/card1 (drm)
KERNEL[1108429.981758] change   /devices/platform/gpu/drm/card1 (drm)
UDEV  [1108429.985759] change   /devices/platform/gpu/drm/card1 (drm)
KERNEL[1108430.311143] change   /devices/platform/gpu/drm/card1 (drm)
UDEV  [1108430.316056] change   /devices/platform/gpu/drm/card1 (drm)

Monitor Power Restored
KERNEL[1108443.071776] change   /devices/platform/gpu/drm/card1 (drm)
UDEV  [1108443.079177] change   /devices/platform/gpu/drm/card1 (drm)
KERNEL[1108443.401907] change   /devices/platform/gpu/drm/card1 (drm)
UDEV  [1108443.410396] change   /devices/platform/gpu/drm/card1 (drm)
KERNEL[1108443.832669] change   /devices/platform/gpu/drm/card1 (drm)
UDEV  [1108443.834981] change   /devices/platform/gpu/drm/card1 (drm)
KERNEL[1108444.262408] change   /devices/platform/gpu/drm/card1 (drm)
UDEV  [1108444.264779] change   /devices/platform/gpu/drm/card1 (drm)

***@***.***:~ $ dmesg -w
[    0.000000] Booting Linux on physical CPU 0x0000000000 [0x410fd083]
[    0.000000] Linux version 6.6.62+rpt-rpi-v8 ***@***.***)
(gcc-12 (Debian 12.2.0-14) 12.2.0, GNU ld (GNU Binutils for Debian) 2.40)
#1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25)
[    0.000000] KASLR enabled
[    0.000000] random: crng init done
[    0.000000] Machine model: Raspberry Pi 4 Model B Rev 1.4
[    0.000000] efi: UEFI not found.
[    0.000000] Reserved memory: created CMA memory pool at
0x000000000e000000, size 512 MiB
[    0.000000] OF: reserved mem: initialized node linux,cma, compatible id
shared-dma-pool
[    0.000000] OF: reserved mem: 0x000000000e000000..0x000000002dffffff
(524288 KiB) map reusable linux,cma
[    0.000000] OF: reserved mem: 0x000000003ef653c0..0x000000003ef655bf (0
KiB) nomap non-reusable ***@***.***
[    0.000000] OF: reserved mem: 0x000000003ef65600..0x000000003ef656e1 (0
KiB) nomap non-reusable ***@***.***
[    0.000000] NUMA: No NUMA configuration found
[    0.000000] Faking a node at [mem 0x0000000000000000-0x00000000ffffffff]
[    0.000000] Faking a node at [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] NUMA: NODE_DATA [mem 0xfbffd3c0-0xfbffffff]
[    0.000000] NUMA: NODE_DATA [mem 0x1fefef3c0-0x1feff1fff]
[    0.000000] Zone ranges:
[    0.000000]   DMA      [mem 0x0000000000000000-0x000000003fffffff]
[    0.000000]   DMA32    [mem 0x0000000040000000-0x00000000ffffffff]
[    0.000000]   Normal   [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000000000000-0x000000003b3fffff]
[    0.000000]   node   0: [mem 0x0000000040000000-0x00000000fbffffff]
[    0.000000]   node   1: [mem 0x0000000100000000-0x00000001ffffffff]
[    0.000000] Initmem setup node 0 [mem
0x0000000000000000-0x00000000fbffffff]
[    0.000000] Initmem setup node 1 [mem
0x0000000100000000-0x00000001ffffffff]
[    0.000000] On node 0, zone DMA32: 19456 pages in unavailable ranges
[    0.000000] On node 1, zone Normal: 16384 pages in unavailable ranges
[    0.000000] percpu: Embedded 30 pages/cpu s85928 r8192 d28760 u122880
[    0.000000] pcpu-alloc: s85928 r8192 d28760 u122880 alloc=30*4096
[... 748 lines cut ...]
[   11.413047] Bluetooth: RFCOMM socket layer initialized
[   11.413066] Bluetooth: RFCOMM ver 1.11
[   79.082094] warning: `ThreadPoolForeg' uses wireless extensions which
will stop working for Wi-Fi 7 hardware; use nl80211
[   91.269773] v3d fec00000.v3d: MMU error from client CLE (4) at
0x84a1000, pte invalid
[  131.589993] logitech-hidpp-device 0003:046D:4024.0005: HID++ 2.0 device
connected.
[947493.920762] usb 1-1.3: USB disconnect, device number 3
[989818.684202] usb 1-1.3: new full-speed USB device number 5 using xhci_hcd
[989818.790582] usb 1-1.3: New USB device found, idVendor=2575,
idProduct=0401, bcdDevice=14.30
[989818.790606] usb 1-1.3: New USB device strings: Mfr=1, Product=2,
SerialNumber=0
[989818.790619] usb 1-1.3: Product: CoolTouchR System
[989818.790630] usb 1-1.3: Manufacturer: Weida Hi-Tech
[989818.817715] input: Weida Hi-Tech                CoolTouchR System
     Mouse as
/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.0/0003:2575:0401.0006/input/input19
[989818.818510] input: Weida Hi-Tech                CoolTouchR System
     as
/devices/platform/scb/fd500000.pcie/pci0000:00/0000:00:00.0/0000:01:00.0/usb1/1-1/1-1.3/1-1.3:1.0/0003:2575:0401.0006/input/input20
[989818.819184] hid-multitouch 0003:2575:0401.0006: input,hiddev96,hidraw0:
USB HID v1.10 Mouse [Weida Hi-Tech                CoolTouchR System
  ] on usb-0000:01:00.0-1.3/input0