raspberrypi/bookworm-feedback #358: Unable to do image-backup to external CIFS mounted filesystem.  zfdisk hangs

vaguy, 2025-01-16:
HW is Pi 3 Model B Rev 1.2B
Bookworm version is up-to-date -- Linux RPi2 6.6.62+rpt-rpi-v7 #1 SMP Raspbian 1:6.6.62-1+rpt1 (2024-11-25) armv7l GNU/Linux
dpkg -l | grep -i cifs
ii  cifs-utils                           2:7.0-2                          armhf        Common Internet File System utilities
ii  libsmbclient:armhf           2:4.17.8+dfsg-2           armhf        shared library for communication with SMB/CIFS servers
ii  samba                              2:4.17.8+dfsg-2           armhf        SMB/CIFS file, print, and login server for Unix
No issues with mounting a cifs file system under bookworm.
I can read, read/write/create files with cp commands
Here is bash -x output of image-backup
sudo bash -x image-backup -i /mnt/T3500-D/Bkup-RPi3/bookworm3.img,,1000
+ IMGNAME=/mnt/T3500-D/Bkup-RPi3/bookworm3.img
+ IMGSIZE=
+ IMGINCR=1000
+ shift 2
+ '[' 0 -gt 0 ']'
+ '[' '' = '' ']'
+ '[' /mnt/T3500-D/Bkup-RPi3/bookworm3.img '!=' '' ']'
+ IMGFILE=/mnt/T3500-D/Bkup-RPi3/bookworm3.img
+ [[ ! /mnt/T3500-D/Bkup-RPi3/bookworm3.img =~ ^/mnt/.*$ ]]
+ '[' -d /mnt/T3500-D/Bkup-RPi3/bookworm3.img ']'
+ '[' '' = '' ']'
+ IMGSIZE=7704
+ IRFSSIZE=7704
+ '[' 1000 = '' ']'
+ ADDMB=1000
+ '[' -f /mnt/T3500-D/Bkup-RPi3/bookworm3.img ']'
+ truncate -s 8615100416 /mnt/T3500-D/Bkup-RPi3/bookworm3.img
+ '[' 0 -ne 0 ']'
+ '[' dos = dos ']'
+ echo 'label: dos'
+ sfdisk /mnt/T3500-D/Bkup-RPi3/bookworm3.img

At this point sfdisk is hung, but I can Ctrl-C to get out of image-backup
If I rerun the image-backup script
    sudo bash -x image-backup -i /mnt/T3500-D/Bkup-RPi3/bookworm3.img,,1000
things hang a littler earlier
+ IMGNAME=/mnt/T3500-D/Bkup-RPi3/bookworm3.img
+ IMGSIZE=
+ IMGINCR=1000
+ shift 2
+ '[' 0 -gt 0 ']'
+ '[' '' = '' ']'
+ '[' /mnt/T3500-D/Bkup-RPi3/bookworm3.img '!=' '' ']'
+ IMGFILE=/mnt/T3500-D/Bkup-RPi3/bookworm3.img
+ [[ ! /mnt/T3500-D/Bkup-RPi3/bookworm3.img =~ ^/mnt/.*$ ]]
+ '[' -d /mnt/T3500-D/Bkup-RPi3/bookworm3.img ']'

I can't Ctrl-C out of it.  kill -9 no help.
I opened up anther terminal windows, I could tab my way to the image directory, but 'ls -l' hung
I  power cycled to clean things up.

Seems like a OS or CIFS bug, not image-backup.
image-backup works fine under stretch
I did notice that stretch uses version 2:6.7-1 of cifs-utils while bookworm uses 2:7.0-2

Any thoughts or suggestions?
Thx

vaguy, 2025-01-16:
The image-backup script is part of a suite of scripts from RonR-RPi-image-utils
https://github.com/seamusdemora/RonR-RPi-image-utils
That's not real important.
I've narrowed the issue down for you to 2 simple commands

##############################################################
#The first test is using a local USB device - sfdisk COMPLETES

pi@RPi2:~ $ **truncate -s 8636071936 /media/pi/Empty/test5.img** 
pi@RPi2:~ $ **echo "label:dos" | sfdisk /media/pi/Empty/test5.img**
sfdisk returns:
Checking that no-one is using this disk right now ... OK

Disk /media/pi/Empty/test5.img: 8.04 GiB, 8636071936 bytes, 16867328 sectors
Units: sectors of 1 * 512 = 512 bytes
Sector size (logical/physical): 512 bytes / 512 bytes
I/O size (minimum/optimal): 512 bytes / 512 bytes

Script header accepted.
Done.
Created a new DOS (MBR) disklabel with disk identifier 0x754e6c3d.

New situation:
Disklabel type: dos
Disk identifier: 0x754e6c3d

The partition table has been altered.
Syncing disks.

##############################################################
#The second test is using a CIFS mounted file system - sfdisk HANGS

#My mount command looks like
**sudo mount -t cifs -rw -o username=xxxxx,password=xxxxx,iocharset=utf8,file_mode=0777,dir_mode=0777 //192.168.2.10/D /mnt/T3500-D**
The mount returns:
     _mount: (hint) your fstab has been modified, but systemd still uses
           the old version; use 'systemctl daemon-reload' to reload._

pi@RPi2:~ $ **truncate -s 8636071936 /mnt/T3500-D/Bkup-RPi3/test5.img**
pi@RPi2:~ $ **echo "label:dos" | sfdisk /mnt/T3500-D/Bkup-RPi3/test5.img**
Hung.....

Again, this works fine under stretch

lurch, 2025-01-17:
> I've narrowed the issue down for you to 2 simple commands

Thank you for the detailed reproduction steps, I'll give this a try next week.

> The mount returns:
_mount: (hint) your fstab has been modified, but systemd still uses
the old version; use 'systemctl daemon-reload' to reload._

That's a harmless warning, see https://github.com/RPi-Distro/repo/issues/346

> Again, this works fine under stretch

Stretch is now **very** old and no longer supported. We only support Bullseye (security updates only) and Bookworm - see https://www.raspberrypi.com/software/operating-systems/
So if you'd like to give this a test on Bullseye over the weekend, that might be useful.

vaguy, 2025-01-18:
Works on Bullseye

vaguy, 2025-02-02:
lurch - did you ever get a chance to try this?