raspberrypi/bookworm-feedback #359: Desktop icon issue

videosmith, 2025-01-18:
Not certain if this is a software issue, or my hardware.
While attempting to drag an icon in the process of copying the file from Desktop to a folder on an attached USB stick, the icon gets "stuck" to the cursor.
Please see attached screen capture.

Only remedy appears to be a reboot, until it occurs again.

Raspberry Pi 4b Bookworm, latest updates as of 2025-1-10.

https://github.com/user-attachments/assets/7cb92ea1-0f24-4953-a0e7-f78ea0aebe74

qrp73, 2025-01-24:
I also had such issue when tried to move picture from ```pcmanfm``` to github issue page in firefox browser.

lurch, 2025-01-28:
I've been trying to reproduce this myself, but I don't seem able to - drag'n'drop always works properly 😕 
Do you have a reliable way of reproducing this, or is it something that was a one-off? If you **can** reproduce this reliably, please provide detailed steps. Thanks.

videosmith, 2025-01-31:
Happens irregurarly unfortunately. Havent been able to force it to occur yet.

DeeJay, 2025-02-22:
Current Forum discussion ongoing here - [https://forums.raspberrypi.com/viewtopic.php?t=384004#p2294654](https://forums.raspberrypi.com/viewtopic.php?t=384004#p2294654)

(I can't trace the prior discussion, but I thought it was concluded that the timeout for 'lingering' or 'hovering' over something was too sensitive? Or vice-versa.  I thought it was fixed, perhaps around Nov 2024, maybe this is a regression?)

lurch, 2025-02-24:
Ahhh! Thanks for the link to the forum-thread @DeeJay  - I've now found a reliable way of reproducing this:
1. On the desktop, right-click and select "New File..." to create an empty file on your desktop named `New` (this works with either a File or Folder, I'm just using a File for simplicity)
2. Long-press on the `New` icon with your left mouse button for a second or two, and whilst continuing to keep the left mouse button pressed, move the mouse
3. A menu will pop up ("Open" / "Text Editor" / "Open With..." etc.) and the file-icon will simultaneously snap to the mouse cursor
4. As soon as you let go of the left mouse button, the menu disappears but the icon remains stuck to the mouse cursor (and has disappeared from the Desktop)
5. When you next left-click on the Desktop, the icon appears back in its previous location, but _also_ remains stuck to the mouse cursor. (And you can no longer drag&drop `New` to a different place on the Desktop)

As a temporary fix (if you're using the labwc desktop), you can run `pkill pcmanfm` in a Terminal, which is quicker than rebooting the Raspberry Pi.
As a permanent workaround, you can edit `~/.config/libfm/libfm.conf` and at the bottom of the `[config]` section, change `gestures_touch_only=0` to `gestures_touch_only=1` and then save the file and run `pkill pcmanfm` to restart the desktop.

Explanation: This bug is due to an unfortunate interaction between two separate UI flows.
* Flow 1 is to left-click on the `New` icon and then (quickly) drag it to a different screen location before releasing the left mouse button again (i.e. the typical drag&drop behaviour)
* Flow 2 is to left-click on the `New` icon, keep the left button held down for a second or two, and then (without moving the mouse) release the left mouse button. This gets picked up by the "gesture recogniser" and interpreted as a right-click event, and so the usual right-click menu appears ("Open" / "Text Editor" / "Open With..." etc.). This feature is enabled by default, so that a user of either the [Touch Display](https://www.raspberrypi.com/products/raspberry-pi-touch-display/) or [Touch Display 2](https://www.raspberrypi.com/products/touch-display-2/) (or any 3rd-party touchscreen) can "long press" on the screen in order to emulate a mouse right-button click.

Unfortunately, it looks like in the rare cases where "Flow 1" and "Flow 2" happen at the same time, pcmanfm gets confused and you end up with the drag&drop icon permanently stuck to the mouse cursor. By editing `~/.config/libfm/libfm.conf` this disables the gesture recogniser, and so "Flow 2" then can't happen.

DeeJay, 2025-02-25:
Thank you for investigating and suggesting a workaround.

But I have two consequential issues arising from the proposed modification to ~/.config/libfm/libfm.conf

The first is that the .conf file begins -

> # Configuration file for the libfm version 1.3.2.
> # Autogenerated file, don't edit, your changes will be overwritten.
> 

That isn't an idle threat. Some process or update does indeed seem to overwrite that file and remove my modification to it.

Under what circumstances does libfm.conf get rewritten under OS control?

The second is that on an installation of 64-bit RasPiOS Bookworm dated 11th Nov 24 using the 'Desktop (only)' image, NOT the Full 'Desktop and Apps' image, there is no ~/config/libfm directory.  That subdirectory and the .conf file DO exist on two similar installations based on the Full ('Desktop+Apps') image. 

Do you know what Debian (or RPT) package provides the tools involved here, please? ( apt-cache search libfm returns a list too long to experiment with. )

Schmarn, 2025-02-25:
Here's the Solution from my AI that worked for me!

To be clear, when I tried to drag a folder, and occasionally a file, to a different location on my desktop, a generic folder or file icon would stick to my mouse pointer, and the only recourse was to reboot the OS to shake it off. But there are better ways...

The problem is the Wayland display server that the OS has recently switched to, it's a simple 5 minute or less fix, follow:
    - Switch from Wayland back to X11
        - open a terminal window
        - call the raspi-config tool: **sudo raspi-config**
        - use the arrow keys to navigate to **Advanced Options** and press **Enter**
        - find the _Wayland_ option and press **Enter**
        - in the options presented, find _X11_ and press **Enter**
        - _X11_ is now active, press **Enter** to return to the main config menu
        - navigate to _Finish_ and press **Enter**
        - reboot your Raspberry Pi, select _Yes_, then **Enter**

This process switches the default display server back to X11, and does not remove Wayland from your system. At any moment you begin to feel nostalgic, feel free to switch back for those frustrating stuck folder days of yor, your call!

It's **_Important_** to note that Updates to the Raspberry Pi could change these settings, and maybe even one day, this issue with the Wayland display server might be resolved! Either way, following these steps, I have been unable to reproduce the stuck folder to mouse pointer issue, and I hope it's your problem solver as well! Good Luck! And much thanks to Gemini...

lurch, 2025-02-25:
X11 is "the past" and Wayland with Labwc is "the future". So you're of course welcome to switch back to X11 (see https://www.raspberrypi.com/documentation/computers/configuration.html#wayland ), but please be aware that if you run into any issues whilst using X11, we're unlikely to fix them.

Also, I suspect that using X11 solves this problem for you because it doesn't support the gesture recogniser and so the "Flow 2" that I described above never gets triggered.

DeeJay, 2025-02-26:
I hope this unhelpful AI-generated distraction will not suppress the useful discussion we were having.  There were a couple of outstanding queries subsequent to your investigation.

lurch, 2025-02-26:
@DeeJay I hadn't forgotten - I'm hoping to investigate your queries today or tomorrow 🙂

lurch, 2025-03-04:
> Under what circumstances does libfm.conf get rewritten under OS control?

@DeeJay I've just had a chat with @spl237 about this (apologies for the delay, we've been busy with different things), and he says that there's two different programs that he knows of that modify the `~/.config/libfm/libfm.conf` file:
* The "Edit -> Preferences" option in the "File Manager" (which adds the "# Autogenerated file, don't edit, your changes will be overwritten." comment to the file)
* The "Defaults" tab in the "Appearance Settings" (which doesn't add the aforementioned comment to the file) - the way that this updates `libfm.conf` seems to be a bit buggy

So apparently the workaround for now is to make whatever changes you need in "Appearance Settings" and "File Manager"'s preferences _first_, and **then** change the `gestures_touch_only` setting in `libfm.conf`.

lurch, 2025-03-05:
After further investigation, we've discovered that if you edit `~/.config/libfm/libfm.conf` by hand, and then edit the Preferences in File Manager, File Manager will overwrite whatever changes you made to `~/.config/libfm/libfm.conf` (because it wasn't aware that the file had been manually edited). However, if you manually edit `~/.config/libfm/libfm.conf` and then reboot (or simply `pkill pcmanfm`), then File Manager **is** then aware of the changes you made, and so any changes made in File Manager's Preferences window  _won't_ then overwrite the changes you previously made.
Slighty complex, but I hope that makes sense!

TLDR: If you manually edit `~/.config/libfm/libfm.conf` (e.g. to set `gestures_touch_only=1`) then always reboot your Pi before doing anything else, otherwise you risk your changes being overwritten.

GKSeifert, 2025-05-16:
I have found that the problem only occurs (for me) when using a wireless mouse or the touchpad on a Logitech K400r keyboard. And it occurs extremely often with the wireless controllers. With a wired mouse I CAN NOT reproduce the problem. Timing? Also, I have found that logging out and back in again will detach the icons being dragged on the mouse pointer. The description by Lurch on Feb 24 is exactly what I see. And the bypass (gestures_touch_only=1) does work. Thanks!