raspberrypi/bookworm-feedback #360: XVideo not available on Pi 3

zocker-160, 2025-01-29:
When running on a Pi 3 in X11 mode, XVideo is not available.

```bash
$ cat /etc/debian_version 
12.7
$ cat /proc/device-tree/model 
Raspberry Pi 3 Model B Plus Rev 1.3
$ xvinfo
X-Video Extension version 2.2
screen #0
 no adaptors present
```
meanwhile on a Pi 4
```bash
$ cat /etc/debian_version 
12.7
$ cat /proc/device-tree/model 
Raspberry Pi 4 Model B Rev 1.2
$ xvinfo
X-Video Extension version 2.2
screen #0
  Adaptor #0: "GLAMOR Textured Video"
    number of ports: 16
    port base: 94
    operations supported: PutImage 
    supported visuals:
      depth 24, visualID 0x21
    number of attributes: 5
      "XV_BRIGHTNESS" (range -1000 to 1000)
              client settable attribute
              client gettable attribute (current value is 0)
      "XV_CONTRAST" (range -1000 to 1000)
              client settable attribute
              client gettable attribute (current value is 0)
      "XV_SATURATION" (range -1000 to 1000)
              client settable attribute
              client gettable attribute (current value is 0)
      "XV_HUE" (range -1000 to 1000)
              client settable attribute
              client gettable attribute (current value is 0)
      "XV_COLORSPACE" (range 0 to 1)
              client settable attribute
              client gettable attribute (current value is 0)
    maximum XvImage size: 8192 x 8192
    Number of image formats: 3
      id: 0x32315659 (YV12)
        guid: 59563132-0000-0010-8000-00aa00389b71
        bits per pixel: 12
        number of planes: 3
        type: YUV (planar)
      id: 0x30323449 (I420)
        guid: 49343230-0000-0010-8000-00aa00389b71
        bits per pixel: 12
        number of planes: 3
        type: YUV (planar)
      id: 0x3231564e (NV12)
        guid: 4e563132-0000-0010-8000-00aa00389b71
        bits per pixel: 12
        number of planes: 2
        type: YUV (planar)
```

My memory could serve me wrong, but afaik it did work on the same hardware in Debian 11 (I sadly cannot test it myself right now).

Why is it not available and could it be added?

6by9, 2025-01-29:
Focus is now on labwc rather than X11.

Pi4 printing `Adaptor #0: "GLAMOR Textured Video"` is probably be the key point. Is glamor enabled on your Pi3?

It wasn't by default as due to the lack of an IOMMU for textures, it ends up fragmenting the CMA heap, and then allocation failures.

It used to be possible to enable via raspi-config, but I can't see the option quickly.
The code for what it did is at https://github.com/RPi-Distro/raspi-config/blob/bookworm/raspi-config#L2531-L2532
```
    systemctl enable glamor-test.service &&
    systemctl start glamor-test.service &&
```
It looks like the interactive option went with https://github.com/RPi-Distro/raspi-config/commit/1089abb821ee0f32c8451fcd62b9df88f047ea01

I think you can use `sudo raspi-config nonint do_glamor 1` to use non-interactive mode to achieve run the standard commands.

zocker-160, 2025-01-29:
> Focus is now on labwc rather than X11.

I know, sadly there is no good way (to my knowledge) to render video into an Qt application with wayland and gstreamer without an additional copy which kills performance on Pi3 and 4.

Currently I just use `x(v)imagesink` with `set_window_handle` and it renders directly into my window with very good performance.

> I think you can use sudo raspi-config nonint do_glamor 1 to use non-interactive mode to achieve run the standard commands.

Maybe I did something wrong, but the glamor service is already running:
```bash
$ sudo systemctl status glamor-test.service
● glamor-test.service - Check for glamor
     Loaded: loaded (/lib/systemd/system/glamor-test.service; enabled; preset: enabled)
     Active: active (exited) since Mon 2024-10-07 21:02:46 CEST; 3 months 22 days ago
    Process: 477 ExecStart=/usr/lib/systemd/scripts/glamor_test.sh (code=exited, status=0/SUCCESS)
   Main PID: 477 (code=exited, status=0/SUCCESS)
        CPU: 289ms

Okt 07 21:02:46 systemd[1]: Starting glamor-test.service - Check for glamor...
Okt 07 21:02:46 systemd[1]: Finished glamor-test.service - Check for glamor.
$ sudo raspi-config nonint do_glamor 1
$ xvinfo
X-Video Extension version 2.2
screen #0
 no adaptors present
$ sudo systemctl status glamor-test.service
● glamor-test.service - Check for glamor
     Loaded: loaded (/lib/systemd/system/glamor-test.service; enabled; preset: enabled)
     Active: active (exited) since Mon 2024-10-07 21:02:46 CEST; 3 months 22 days ago
   Main PID: 477 (code=exited, status=0/SUCCESS)
        CPU: 289ms

Okt 07 21:02:46 systemd[1]: Starting glamor-test.service - Check for glamor...
Okt 07 21:02:46 systemd[1]: Finished glamor-test.service - Check for glamor.
```

6by9, 2025-01-29:
> I'm afraid that it's not something that I know much about myself, but is https://github.com/jc-kynesim/drmu at all useful here?

Unlikely as that is rendering directly to DRM, not talking to a QT application.

A quick search for "gstreamer waylandsink qt render into window" did bring up https://github.com/dothanhtrung/qt-waylandsink-example and https://github.com/xleng/qt-waylandsink-sample.
The former hasn't been touched for 4 years, and the simple switch of `gst_is_wayland_display_handle_need_context_message` to `gst_is_wl_display_handle_need_context_message`, and `gst_wayland_display_handle_context_new` to `gst_wl_display_handle_context_new` then throws linker errors
```
videowidget.cpp:(.text+0x68): undefined reference to `gst_is_wl_display_handle_need_context_message'
/usr/bin/ld: videowidget.cpp:(.text+0xd8): undefined reference to `gst_wl_display_handle_context_new'
```

From the GStreamer Discord archive - https://discourse.gstreamer.org/t/embedding-gstreamer-into-wayland-window/3661/6
> The gst_wl_display_handle_context_new() API is not available via the bindings yet. You’d have to link your application to libgstwayland-1.0 and declare the function locally and then call it via FFI.

So it looks like things are a little early for that bit.

The latter compiles if you replace the explicit `x86_64-linux-gnu`s for `aarch64-linux-gnu` in qtoverlay.pro
It throws a number of GStreamer critical asserts though, and I only get a red window. There's no handling on the creation of pipeline, window, or vsink, so one or more of those is likely failing.

There are works in progress that will offload some layers to the hardware rendering, which reduces the load on the compositor significantly, but it does depend on applications and any frameworks they use providing resources in an sensible manner (ie there's no gain if the application/framework already decided to squash all layers before giving them to Wayland).

6by9, 2025-01-29:
A couple more tweaks to https://github.com/xleng/qt-waylandsink-sample and I'm getting a black window and the error from https://github.com/GStreamer/gst-plugins-bad/blob/master/ext/wayland/gstwaylandsink.c#L958-L962

I've not got time to look at this any further, but that looks fairly close. How it performs I don't know.

zocker-160, 2025-01-29:
Thank you @6by9 for your efforts, I highly appreciate it.

I have looked at that code example using waylandsink over a year ago and I could not get it to work either. 

Furthermore any of the other sinks like `qmlglsink` or `gtkwaylandsink` work, but have terrible performance. Using an appsink works too, but is also slow.

Given the original issue regarding XVideo on Pi 3, I suspect this is a wontfix then, which is fine with me, I just wanted to know if I did something wrong or if this is expected to not be working.

closed 2025-01-29 by zocker-160