raspberrypi/bookworm-feedback #361: chromium high lagging

qrp73, 2025-01-29:
Sometimes I catch very high lagging in chromium browser, it leads to complete system freeze for 5-10 seconds, then it unfreezes for some milliseconds and freezes again. Mouse cursor is stopped and keyboard don't respond during these freeze. 

There is no specific error in dmesg, but when I run chromium from console I see the following errors during these freezes:
```
[4696:11:0129/090909.013763:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
[4696:11:0129/090909.015901:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
[4696:11:0129/090912.009292:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
[4696:11:0129/090912.010009:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
[4696:11:0129/090912.017318:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
[4696:11:0129/090912.020399:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
[1922:4749:0129/090912.481507:ERROR:v4l2_video_decoder.cc(120)] InitializeBackend failed at Initialize@media/gpu/v4l2/v4l2_video_decoder.cc:345 
[1922:4750:0129/090912.483808:ERROR:v4l2_video_decoder.cc(120)] InitializeBackend failed at Initialize@media/gpu/v4l2/v4l2_video_decoder.cc:345 
[4696:11:0129/090912.695073:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
[4696:11:0129/090912.696549:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
close object 1: Invalid argument
close object 5: Invalid argument
close object 6: Invalid argument
close object 5: Invalid argument
close object 6: Invalid argument
close object 7: Invalid argument
close object 1: Invalid argument
close object 8: Invalid argument
close object 9: Invalid argument
[4826:11:0129/090953.631080:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
[4826:11:0129/090953.635139:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
[4826:11:0129/090953.832555:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
close object 1: Invalid argument
close object 5: Invalid argument
close object 6: Invalid argument
[1863:1863:0129/091015.395334:ERROR:interface_endpoint_client.cc(725)] Message 1 rejected by interface blink.mojom.WidgetHost
[1863:1863:0129/091015.399841:ERROR:interface_endpoint_client.cc(725)] Message 1 rejected by interface blink.mojom.WidgetHost
[1863:1863:0129/091015.400345:ERROR:interface_endpoint_client.cc(725)] Message 1 rejected by interface blink.mojom.WidgetHost
[4904:11:0129/091020.337599:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
[4904:11:0129/091020.376551:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
[4904:11:0129/091020.379450:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
[4904:11:0129/091020.380611:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
[4904:11:0129/091020.385187:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
[4904:11:0129/091020.553003:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
[4904:11:0129/091020.647981:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
[1922:5022:0129/091101.448323:ERROR:v4l2_video_decoder.cc(120)] InitializeBackend failed at Initialize@media/gpu/v4l2/v4l2_video_decoder.cc:345 
[1922:5028:0129/091101.583623:ERROR:v4l2_video_decoder.cc(120)] InitializeBackend failed at Initialize@media/gpu/v4l2/v4l2_video_decoder.cc:345 
[1922:5027:0129/091101.583292:ERROR:v4l2_video_decoder.cc(120)] InitializeBackend failed at Initialize@media/gpu/v4l2/v4l2_video_decoder.cc:345 
[4904:11:0129/091121.764352:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
[4904:11:0129/091121.879882:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
[1863:1863:0129/205215.061035:ERROR:interface_endpoint_client.cc(725)] Message 1 rejected by interface blink.mojom.WidgetHost
close object 1: Invalid argument
```

Each time when I catch these high lags I see this error in the console log:
```
[4696:11:0129/090909.013763:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
```

Chromium browser version: 130.0.6723.116-1~deb12u1+rpt1
```
$ uname -a
Linux raspi 6.6.62+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.62-1+rpt1 (2024-11-25) aarch64 GNU/Linux
```

lurch, 2025-01-29:
Can you please provide detailed reproduction steps? Thanks.

qrp73, 2025-01-31:
it was stable reproduced on some videos playback in the browser. But I didn't catch exact video link.

lurch, 2025-02-11:
Closing as reproduction steps not supplied. Please let me know if you're able to reproduce this in future, and give example URLs, and I'll re-open the issue.

closed 2025-02-11 by lurch