raspberrypi/bookworm-feedback #362: Unecessary AAAA DNS lookups in IPv4 only network

scred47, 2025-01-30:
The raspberry OS seems to be doing AAAA lookups even when running just ping queries to a name,

 According to manpage of resolv.conf manpage the default DNS timeout is 5 seconds.

https://www.man7.org/linux/man-pages/man5/resolv.conf.5.html

when I run 

ping google.com 

I can actually count to 5 before the first ping is even sent and we can see the timestamp between the dns lookup and the first ping being 5 seconds 

```
11:07:41.821340 IP 192.168.2.187.57450 > pi.hole.domain: 29346+ A? google.com. (28)
11:07:41.821395 IP 192.168.2.187.57450 > pi.hole.domain: 52819+ AAAA? google.com. (28)
11:07:41.849260 IP pi.hole.domain > 192.168.2.187.57450: 29346 1/0/0 A 142.250.179.174 (44)
11:07:41.914650 IP 192.168.2.187.45426 > pi.hole.domain: 36150+ PTR? 187.2.168.192.in-addr.arpa. (44)
11:07:41.928596 IP pi.hole.domain > 192.168.2.187.45426: 36150 NXDomain* 0/0/0 (44)
11:07:41.929075 IP 192.168.2.187.48648 > pi.hole.domain: 21276+ PTR? 3.3.168.192.in-addr.arpa. (42)
11:07:41.941446 IP pi.hole.domain > 192.168.2.187.48648: 21276* 1/0/0 PTR pi.hole. (63)
11:07:46.825492 IP 192.168.2.187.57450 > pi.hole.domain: 29346+ A? google.com. (28)
11:07:46.841263 IP pi.hole.domain > 192.168.2.187.57450: 29346 1/0/0 A 142.250.179.174 (44)
11:07:46.841438 IP 192.168.2.187.57450 > pi.hole.domain: 52819+ AAAA? google.com. (28)
11:07:46.870100 IP pi.hole.domain > 192.168.2.187.57450: 52819 1/0/0 AAAA 2a00:1450:400e:802::200e (56)
11:07:46.871484 IP 192.168.2.187 > ams15s41-in-f14.1e100.net: ICMP echo request, id 6, seq 1, length 64
11:07:46.879693 IP ams15s41-in-f14.1e100.net > 192.168.2.187: ICMP echo reply, id 6, seq 1, length 64

```

This delay is causing timeouts in some utilities that have sensitive timeout tresholds , and also generates unecesary noise on the network given that the host is not ipv6 capable anyways.

The problem is seen on is running a fresh and up to date Raspberry OS on a raspberry pi zero 2W configured to use the builtin wifi and dhcp

```
$ cat /etc/os-release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"

```

lurch, 2025-01-30:
Are you sure that this is a problem specifically with Raspberry Pi OS, and not just a generic Linux networking thing? Or perhaps it's a problem specific to your network?

The Raspberry Pi office network uses IPv6, which'll make this tricky to test 😉  If I access https://test-ipv6.com from a Pi connected to the office network, it reports 10/10 (and from a terminal I'm able to `ping google.com` with no delay).

However I was able to use Raspberry Pi Connect to connect to one of my home Pi's (on a domestic Virgin Media cable connection), and from there https://test-ipv6.com reports 0/10, but from a terminal I'm still able to `ping google.com` without any big delay.

scred47, 2025-01-30:
Hi lurch, 

thank you for checking.

The Raspberry OS Lite is freshly installed using the using the Pi imager method and using the custom settings feature of the tool to configure the wifi, enable ssh and  configure user and password. I do this to use it headless. 

> Are you sure that this is a problem specifically with Raspberry Pi OS, and not just a generic Linux networking thing? Or perhaps it's a problem specific to your network?

I'm not certain of anything . It may very well be a general Linux network thingy , I'm keen on  reading  any source that explains why these 2 things are normal. 

1)  I still have an autoconfigured link local ipv6 on my wireless NIC even after I disabled ipv6 on sysctl to attempt to work this around.
2) I'm doing AAAA lookups although I don't need to 

 I guess having the first bullet answered may  help correct the second.

6by9, 2025-01-30:
Which systemctl calls have you used to disable ipv6?

I've just created `/etc/sysctl.d/90-disable_ipv6.conf` with `net.ipv6.conf.all.disable_ipv6 = 1` as suggested in https://linuxconfig.org/how-to-disable-ipv6-on-linux, and that removes all IPv6 properties from my network interfaces (after kick with `sudo systctl -p /etc/sysctl.d/90-disable_ipv6.conf`).

The other global nuke option for IPv6 would be to add `ipv6.disable=1` to the start of /boot/firmware/cmdline.txt.

scred47, 2025-01-30:
Hi 6by9,

thank you.

I followed that method and indeed ipv6 is gone on the NIC.

Yet I'm still doing AAAA queries

```
$ sudo tcpdump -i wlan0 port 53 or icmp
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on wlan0, link-type EN10MB (Ethernet), snapshot length 262144 bytes
17:20:19.518097 IP 192.168.2.187.36889 > pi.hole.domain: 49653+ A? google.com. (28)
17:20:19.518154 IP 192.168.2.187.36889 > pi.hole.domain: 26342+ AAAA? google.com. (28)
17:20:19.534234 IP pi.hole.domain > 192.168.2.187.36889: 49653 1/0/0 A 142.251.36.14 (44)
17:20:19.610209 IP 192.168.2.187.56382 > pi.hole.domain: 50075+ PTR? 187.2.168.192.in-addr.arpa. (44)
17:20:19.625433 IP pi.hole.domain > 192.168.2.187.56382: 50075 NXDomain* 0/0/0 (44)
17:20:19.634012 IP 192.168.2.187.54275 > pi.hole.domain: 3998+ PTR? 3.3.168.192.in-addr.arpa. (42)
17:20:19.645723 IP pi.hole.domain > 192.168.2.187.54275: 3998* 1/0/0 PTR pi.hole. (63)
17:20:24.522467 IP 192.168.2.187.36889 > pi.hole.domain: 49653+ A? google.com. (28)
17:20:24.537318 IP pi.hole.domain > 192.168.2.187.36889: 49653 1/0/0 A 142.251.36.14 (44)
17:20:24.537446 IP 192.168.2.187.36889 > pi.hole.domain: 26342+ AAAA? google.com. (28)
17:20:24.561999 IP pi.hole.domain > 192.168.2.187.36889: 26342 1/0/0 AAAA 2a00:1450:400e:811::200e (56)
17:20:24.579144 IP 192.168.2.187 > ams15s44-in-f14.1e100.net: ICMP echo request, id 1, seq 1, length 64
17:20:24.589090 IP 192.168.2.187.45161 > pi.hole.domain: 3014+ PTR? 14.36.251.142.in-addr.arpa. (44)
17:20:24.591174 IP ams15s44-in-f14.1e100.net > 192.168.2.187: ICMP echo reply, id 1, seq 1, length 64
```

and the ping query is still sent 5 seconds after I already received the A lookup result.

I can certainly try to fix why my dns server is taking so long to answer the AAAA lookup but in any case why would I do an AAAA lookup  if I don't have ipv6 at all anyways?

lurch, 2025-01-30:
I wonder if you get any different results / behaviour if you disable PiHole?

andrum993, 2025-01-30:
On a Pi 5 which does not have Pihole installed on it, I have ipv6.disable=1 set in cmdline.txt, but I am also seeing it do AAAA lookups.

I did a quick test on an amd64 Debian 12 VM on my laptop, and adding ipv6.disable=1 to the kernel command line disabled AAAA lookups. The VM was installed using the standard Debian installer, with most packages selected using tasksel. (I added samba and ZFS afterwards). So it looks like this behaviour might be Raspberry Pi-specific. I had a quick nosey at what the Debian 12 VM was using for network configuration, and it seems to be using Network Manager, which is what Raspberry Pi OS uses, so I'm guessing there is some Raspberry Pi configuration somewhere that is causing IPv6 to not be fully disabled.

lurch, 2025-01-30:
@scred47 Looking a bit closer at your `tcpdump` output, I see that you have hosts in both the `192.168.2.*` range and `192.168.3.*` range - could you provide a bit more background about your network topology please, in case that's relevant?

@andrum993 Have you been able to reproduce the "5 second ping delays" that the OP is complaining about?

andrum993, 2025-01-30:
> [@andrum993](https://github.com/andrum993) Have you been able to reproduce the "5 second ping delays" that the OP is complaining about?

No - I've never had any problems with pings on any of my Pis. It might be specific to the use of wifi - I don't use wifi on any of my Pis (except for one that is deployed at my parents' house as a network music player). Or perhaps the OP has changed a settting that has exposed this behaviour.

scred47, 2025-01-31:
Hi lurch ,

Sure , here are some further tests.

No AAAA lookup no custom timeouts No Pihole

```
# Dynamic resolv.conf(5) file for glibc resolver(3) generated by resolvconf(8)
#    DO NOT EDIT THIS FILE BY HAND -- YOUR CHANGES WILL BE OVERWRITTEN
# 127.0.0.53 is the systemd-resolved stub resolver.
# run "resolvectl status" to see details about the actual nameservers.
#nameserver 192.168.3.3
#nameserver 1.1.1.1
options no-aaaa
#options timeout:2
nameserver 1.1.1.1

```

ping is sent within milliseconds of the response.

```
sudo tcpdump -ni wlan0 port 53 or icmp
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on wlan0, link-type EN10MB (Ethernet), snapshot length 262144 bytes
10:31:44.145319 IP 192.168.2.187.54409 > 1.1.1.1.53: 9229+ A? google.com. (28)
10:31:44.162949 IP 1.1.1.1.53 > 192.168.2.187.54409: 9229 1/0/0 A 142.250.179.142 (44)
10:31:44.163832 IP 192.168.2.187 > 142.250.179.142: ICMP echo request, id 4, seq 1, length 64
10:31:44.171430 IP 142.250.179.142 > 192.168.2.187: ICMP echo reply, id 4, seq 1, length 64

```

No Pihole , AAAA lookups allowed, no custom timeout

```
:~ $ cat /etc/resolv.conf
# Dynamic resolv.conf(5) file for glibc resolver(3) generated by resolvconf(8)
#    DO NOT EDIT THIS FILE BY HAND -- YOUR CHANGES WILL BE OVERWRITTEN
# 127.0.0.53 is the systemd-resolved stub resolver.
# run "resolvectl status" to see details about the actual nameservers.
#nameserver 192.168.3.3
#nameserver 1.1.1.1
#options no-aaaa
#options timeout:2
nameserver 1.1.1.1
```

ping is sent 5 seconds after the A record response is received

```
:~ $ sudo tcpdump -ni wlan0 port 53 or icmp
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on wlan0, link-type EN10MB (Ethernet), snapshot length 262144 bytes
11:12:18.669457 IP 192.168.2.187.39768 > 1.1.1.1.53: 7918+ A? google.com. (28)
11:12:18.669516 IP 192.168.2.187.39768 > 1.1.1.1.53: 60034+ AAAA? google.com. (28)
11:12:18.690870 IP 1.1.1.1.53 > 192.168.2.187.39768: 7918 1/0/0 A 142.250.179.142 (44)
11:12:23.674134 IP 192.168.2.187.39768 > 1.1.1.1.53: 7918+ A? google.com. (28)
11:12:23.694051 IP 1.1.1.1.53 > 192.168.2.187.39768: 7918 1/0/0 A 142.250.179.142 (44)
11:12:23.694197 IP 192.168.2.187.39768 > 1.1.1.1.53: 60034+ AAAA? google.com. (28)
11:12:23.710068 IP 1.1.1.1.53 > 192.168.2.187.39768: 60034 1/0/0 AAAA 2a00:1450:400e:801::200e (56)
11:12:23.727193 IP 192.168.2.187 > 142.250.179.142: ICMP echo request, id 5, seq 1, length 64
11:12:23.735397 IP 142.250.179.142 > 192.168.2.187: ICMP echo reply, id 5, seq 1, length 64
```

No Pihole , AAAA lookups allowed, custom timeout of 2 seconds

```
:~ $ cat /etc/resolv.conf
# Dynamic resolv.conf(5) file for glibc resolver(3) generated by resolvconf(8)
#    DO NOT EDIT THIS FILE BY HAND -- YOUR CHANGES WILL BE OVERWRITTEN
# 127.0.0.53 is the systemd-resolved stub resolver.
# run "resolvectl status" to see details about the actual nameservers.
#nameserver 192.168.3.3
#nameserver 1.1.1.1
#options no-aaaa
options timeout:2
nameserver 1.1.1.1

```

I takes 2 seconds for ping to be sent after A record response is received.

```
:~ $ sudo tcpdump -ni wlan0 port 53 or icmp
tcpdump: verbose output suppressed, use -v[v]... for full protocol decode
listening on wlan0, link-type EN10MB (Ethernet), snapshot length 262144 bytes
11:15:33.264368 IP 192.168.2.187.38072 > 1.1.1.1.53: 36281+ A? google.com. (28)
11:15:33.264427 IP 192.168.2.187.38072 > 1.1.1.1.53: 37514+ AAAA? google.com. (28)
11:15:33.283103 IP 1.1.1.1.53 > 192.168.2.187.38072: 36281 1/0/0 A 142.250.179.206 (44)
11:15:35.266366 IP 192.168.2.187.38072 > 1.1.1.1.53: 36281+ A? google.com. (28)
11:15:35.285140 IP 1.1.1.1.53 > 192.168.2.187.38072: 36281 1/0/0 A 142.250.179.206 (44)
11:15:35.285280 IP 192.168.2.187.38072 > 1.1.1.1.53: 37514+ AAAA? google.com. (28)
11:15:35.301463 IP 1.1.1.1.53 > 192.168.2.187.38072: 37514 1/0/0 AAAA 2a00:1450:400e:80d::200e (56)
11:15:35.318563 IP 192.168.2.187 > 142.250.179.206: ICMP echo request, id 6, seq 1, length 64
11:15:35.326315 IP 142.250.179.206 > 192.168.2.187: ICMP echo reply, id 6, seq 1, length 64

```