raspberrypi/bookworm-feedback #363: Password not recognized when entered in login screen ...

AlainB34270, 2025-02-05:
Hello,
I hope I'm in the right place ... If not, maybe you can help me find that right place.
A few days ago, I installed the latest **32bit Bookworm (wayland-based) image for pi 4 via the Raspi Imager with Auto-login on the desktop and Keyboard=fr, 105, UTF-8**. 
This worked well at first. I did **sudo upgrade and full-upgrade then installed nmon, putty and Candle.** I had had to reboot a few times and was auto-logged to the desktop OK.
**But yesterday, I booted up and got the login screen** (!!!). 
What could have triggered that? It had already happened to me on other previous bookworm versions and I hadn't been able to sort it out then (only re-install and pray that the lightdm login screen would not appear, as I knew that I'd then be f....d!) but my requirement for UI was not then what it is now.
**I entered my password (user was OK) but was not recognized by the system**. Entered my password on ssh OK. Ctrl-Alt F1 to open a terminal. login on console OK too.
I changed **xkb_layout** from **gb** to **fr** in **/etc/wayfire/template.ini**
I assume it replaces **$HOME/.config/wayfire.ini**  ...
Checked **/etc/defaut/keyboard**
Tried with and without **xkb_variant= azerty** in both files.
I also tried **sudo raspi-config** to change keyboard and change back again. Even used it to change from **labw** to **wayfire** and even **X11** ... and back.
Nothing works ... 
Will I have to reinstall and pray ?
Isn't this problem supposed to have been resolved at least a year ago? Is it still an issue with inconsistent keyboard definitions ? And I don't even have multiple users on that system. Just one but who expects the system to realize that he (very probably) doesn't change keyboards just for logging in ;-(
I've probably missed many things but I don't know what to do anymore.
Please let me know if you have solutions, investigation procedures, or any idea that could help me resolve this issue
Have a great one

AlainB34270, 2025-02-10:
Issue reproduction : 
-  32-bit bookworm out of the box, kb=fr, passwd="xxxx" (litterally).
- sudo upgrade
- sudo raspi-config to get to desktop with user login + reboot (login OK)
- another reboot. passwd no longer recognized!

If i use "erty" as a password (common to QWERTY and AZERTY keyboards) I seem to be able to login, no matter how often I reboot and whether I am auto logged or require userid/psswd ...
Is this the expected system behaviour?
Good day

XECDesign, 2025-02-10:
Many thanks for the clear summary. I can see that the keyboard layout setting gets reset to GB when raspberrypi-ui-mods gets updated, which would cause issues like this.

One thing that doesn't make sense to me though - isn't `xxxx` the same on GB and FR layouts? If I set that at the password through raspi-config, although the keyboard layout is wrong on the login screen, I can still log in using that password.

AlainB34270, 2025-02-10:
Hi, I probably wasn't fully awake this morning, I meant "wwww" ... Apologies

XECDesign, 2025-02-10:
No worries, that makes more sense. We should have a fix out soon. In the meanwhile, you can edit /usr/share/labwc/environment to fix the layout.

closed 2025-02-10 by AlainB34270