raspberrypi/bookworm-feedback #364: Raspberry PI's ISP (libcamera/picamera2) crops IMX500's inference detection region when video stream is cropped

vikasmalitest, 2025-02-06:
Hello,

I am using picamera2 library (Raspberry Pi's ISP) to crop and show video feed of IMX500 based on people detection (person detected coordinates) obtained from AI inference on IMX500 using object detection model. 

Ideal situation, when I am using ISP cropping method using picamera2 for showing cropped video feed output (x,y,w,h) of custom coordinates, it should show cropped video feed and at the same time, AI detection running on the full resolution which is 4056 * 3040. But this is not the case. When I feed (x,y,w,h) coordiantes for showing cropped video streaming, it runs AI inferencing on (x,y,w,h) coordinates rather than running inference on full resolution (4056*3040).

I am using picamera2.setcontrols("ScalerCrop":(x,y,w,h,))
Please let me know if I am missing something or API dedictaed for above application or this is a missing functionality in using ISP of PI.

lurch, 2025-02-11:
ping @davidplowman and @naushir

davidplowman, 2025-02-11:
Hi, this has been moved to the Picamera2 repo, [here](https://github.com/raspberrypi/picamera2/issues/1204).

closed 2025-02-11 by lurch