raspberrypi/bookworm-feedback #366: SDL2 apps blocking display sleep when running in minimized window with wayland driver

qrp73, 2025-02-12:
Steps to reproduce:

1) create file ```test.c``` and put the following code:
```
// sudo apt install libsdl2-dev
// gcc -o test test.c -Wall -lGL -lSDL2
#include <SDL2/SDL.h>
#include <SDL2/SDL_opengl.h>
#include <stdio.h>

#define WINDOW_WIDTH 640
#define WINDOW_HEIGHT 480

int main(int argc, char *argv[]) {
    // Initialize SDL with Wayland driver if available
    if (SDL_VideoInit("wayland") < 0) {
        printf("Wayland driver not available, falling back to default. SDL_Error: %s\n", SDL_GetError());
        return -1;
    }
    printf("SDL_GetCurrentVideoDriver(): %s\n", (const char*)SDL_GetCurrentVideoDriver());

    // Create an SDL window with OpenGL context
    SDL_Window *window = SDL_CreateWindow("SDL2 OpenGL Window", 
                                          SDL_WINDOWPOS_CENTERED, SDL_WINDOWPOS_CENTERED, 
                                          WINDOW_WIDTH, WINDOW_HEIGHT, 
                                          SDL_WINDOW_OPENGL | SDL_WINDOW_SHOWN);
    if (!window) {
        printf("Window could not be created! SDL_Error: %s\n", SDL_GetError());
        SDL_Quit();
        return -1;
    }

    // Create OpenGL context
    SDL_GLContext glContext = SDL_GL_CreateContext(window);
    if (!glContext) {
        printf("OpenGL context could not be created! SDL_Error: %s\n", SDL_GetError());
        SDL_DestroyWindow(window);
        SDL_Quit();
        return -1;
    }
    // Enable VSync
    if (SDL_GL_SetSwapInterval(1) < 0) {
        printf("Warning: Unable to set VSync! SDL_Error: %s\n", SDL_GetError());
    }

    int running = 1;
    SDL_Event event;

    glClearColor(0.5f, 0.5f, 0.5f, 1.0f);

    // Main loop
    while (running) {
        while (SDL_PollEvent(&event)) {
            if (event.type == SDL_QUIT) {
                running = 0;
            }
        }
        glClear(GL_COLOR_BUFFER_BIT);
        SDL_GL_SwapWindow(window);
    }

    // Cleanup
    SDL_GL_DeleteContext(glContext);
    SDL_DestroyWindow(window);
    SDL_Quit();
    return 0;
}
```

2) compile it with command ```gcc -o test test.c -Wall -lGL -lSDL2```

3) open separate terminal and run the command:
```
swayidle -w timeout 10 'wlopm --off \*' resume 'wlopm --on \*'
```
it should put display into sleep mode after 10 seconds idle. Make sure that it works.

4) start compiled test app with command ```./test```, it should open OpenGL window filled with gray color.

5) minimize test app window

6) wait for 10 seconds

**Expected result:** display going to sleep mode

**Actual result:** swayidle don't trigger idle event until you close test app, despite the fact that it is running in minimized state.

**Note:** when using x11 driver to initialize SDL2, it works ok - as expected, the display going to sleep. The issue happens with wayland driver only.

Consolatis, 2025-02-12:
That happens because SDL2 requests an idle inhibitor. Why SDL does that I have no clue. There is a bunch of bug reports about random SDL applications preventing screensavers all around the web.

`SDL_VIDEO_ALLOW_SCREENSAVER=1 ./test` fixes it: https://wiki.libsdl.org/SDL2/SDL_HINT_VIDEO_ALLOW_SCREENSAVER

I think you can also set the hint in your sources but I am no SDL dev.

lurch, 2025-02-12:
Thanks for the info @Consolatis ! Sounds like this resolves the original problem so I'll close the issue.

closed 2025-02-12 by lurch