raspberrypi/bookworm-feedback #372: Halved fps on mouse move in fullscreen mode issue

qrp73, 2025-02-13:
There is an issue which leads to drop down fps to a half value of display refresh rate, it happens when you move mouse in fullscreen mode. 

First I reported this issue about 2 years ago. The fix was made, but not published and then the issue was deleted and completely disappears.

About a year ago I reported it again. And again the fix was made for private testing in some separate branch and it was possible to install it for test. But now I cannot see this issue. It looks that again it was deleted and all info disappears.

But the bug is still there and still not fixed in Raspberry Pi OS Bookworm with all latest updates. And it is very annoying as it leads to a low fps in games which happens when you move mouse in fullscreen mode.

Please fix this halved fps issue.

### Steps to reproduce:
1) Create ```test-vblank4.c``` file and copy/paste code (see below)
2) Compile with ```gcc -o test-vblank4 test-vblank4.c -Wall -lGL -lSDL2 -lSDL2_ttf -lSDL2_image```
3) Run with ```./test-vblank4```. This test app shows realtime graph with time interval between buffer swaps. It should be stable and equals to display refresh rate.
4) Press F11 to switch into fullscreen mode
5) Move mouse

**Expected result:** buffer swap time interval is stable

**Actual result:** buffer swap time interval raise up and leads to drop down update rate. As result app fps drops down to a half value of display refresh rate.

[image]

Since the fix is already available and tested, please just fix the issue.

### Test code

<details><summary>test-vblank4.c</summary>

```c
// sudo apt install libsdl2-dev libsdl2-ttf-dev libsdl2-image-dev
// sudo pacman -S sdl2 sdl2_ttf sdl2_image
// gcc -o test-vblank4 test-vblank4.c -Wall -lGL -lSDL2 -lSDL2_ttf -lSDL2_image
#include <stdio.h>
#include <string.h>
#include <time.h>
#include <limits.h>
#include <GL/gl.h>
#include <SDL2/SDL.h>
#include <SDL2/SDL_opengl.h>
#include <SDL2/SDL_ttf.h>
#include <SDL2/SDL_image.h>

#ifdef _WIN32
#include <limits.h>
#define MAX_PATH_LENGTH MAX_PATH
#else
#include <limits.h>
#define MAX_PATH_LENGTH PATH_MAX
#endif

// Simple queue implementation
typedef struct {
    uint64_t* data;
    size_t front;
    size_t rear;
    size_t size;
    size_t capacity;
} Queue64;
void queue_init(Queue64* q, size_t capacity) {
    q->capacity = capacity;
    q->size = 0;
    q->front = 0;
    q->rear = 0;
    q->data = (uint64_t*)malloc(capacity * sizeof(uint64_t));
}
void queue_free(Queue64* q) {
    free(q->data);
}
void queue_resize(Queue64* q, size_t new_capacity) {
[... 648 lines cut ...]
        ts1 = ts2;
        _frameCounter++;

        queue_enqueue(&_queue, dt_ns);
        SDL_GetWindowSize(window, &width, &height);
        ssize_t excess_elements = (ssize_t)queue_getSize(&_queue) - width;
        if (excess_elements > 0) {
            queue_trimFront(&_queue, excess_elements);
        }
    }

    queue_free(&_queue);
    freeGlyphs();

    IMG_Quit();
    TTF_CloseFont(font);
    TTF_Quit();
    SDL_GL_DeleteContext(glContext);
    SDL_DestroyWindow(window);
    SDL_Quit();
    return 0;
}
```

</details>

popcornmix, 2025-02-13:
> First I reported this issue about 2 years ago. It was analyzed and the patch was made, but not published and didn't applied to the code, then the issue was deleted and completely disappears.

> Since the patch is already available in private branches and tested, please just apply that patch to fix the issue.

I don't believe we've ever deleted a github issue.
I don't remember the details of a two year old issue. 
Do you remember if the patch was to the kernel/firmware/mesa/something else?

qrp73, 2025-02-13:
As I remember the patch was for mesa. And according to discussion it happens because there is something reserve some resource to support two 4K displays, but as a side effect it leads to issue for any display and the patch was implemented in that ways that it reserve this resource only when the system really uses two 4K displays and for other cases it allows to not slow down the display update.

If I remember correctly it was discussed with @ghollingworth

qrp73, 2025-02-13:
I found last discussion, here is the patch: https://github.com/labwc/labwc/issues/2290#issuecomment-2460498166

Could you please apply it to the main branch?

Consolatis, 2025-02-13:
> The original issue was disappears for some unknown reason so I cannot find it in history.

https://github.com/issues should show all of the issues you created.

qrp73, 2025-02-13:
> There is a bump to mesa 24.2.4 in testing in the beta forums. It may be worth [joining that](https://forums.raspberrypi.com/viewtopic.php?t=345124) and checking if the fix is present.

I know this forum and was participated in this beta. But there was some moderator on this forum who decided to joke on me and unfortunately administrators didn't responded on my messages for a long time. After all they responded, but it was too late, I decided to leave it. I promised myself to not return. It was before the administrators contacted me. Later they contacted with me and the issue was settled, so to speak, but since I promised to myself to not return, now this forum is taboo for me. I won't respect myself if I go back there.

This is issue with halved fps is very old, it is in test stage more than one year. It seems too much, isn't it? Hope it will be rolled out with update soon...

popcornmix, 2025-02-13:
When I get a chance, I'll try your test program with the beta mesa version.

popcornmix, 2025-02-14:
Cannot reproduce with my system using beta repo:
```
$ apt policy mesa-common-dev
mesa-common-dev:
  Installed: (none)
  Candidate: 24.2.4-1~bpo12+1~rpt1
  Version table:
     24.2.8-1~bpo12+1 100
        100 http://deb.debian.org/debian bookworm-backports/main arm64 Packages
     24.2.4-1~bpo12+1~rpt1 500
        500 http://buildbot.pitowers.org:3143 bookworm/untested arm64 Packages
     23.2.1-1~bpo12+rpt3 500
        500 http://buildbot.pitowers.org:3143 bookworm/main arm64 Packages
     22.3.6-1+deb12u1 500
        500 http://deb.debian.org/debian bookworm/main arm64 Packages
```
I only see (narrow) spikes in the test app when pressing F11. No effect from moving mouse in either windowed or fullscreen mode.

qrp73, 2025-02-14:
> Cannot reproduce with my system using beta repo

When this version will be rolled out to the main branch?

popcornmix, 2025-02-14:
The usual answer is "when it's ready".
I'll ask when I see someone who may know, but I suspect that's the response I'll get.

popcornmix, 2025-02-17:
As expected, "when it's ready.
There are some positive reports of bugs fixes and (significantly) higher performance achieved, but there is one outstanding regression that affects performance in full screen mode. It's being investigated but no fix yet.