raspberrypi/bookworm-feedback #373: Apparmor: aa-logprof and aa-genprof not working with class="file" in apparmor message

MagicSimi, 2025-02-15:
`uname -r
6.6.74+rpt-rpi-2712
`

lurch, 2025-02-18:
I'm afraid I know nothing about Apparmor myself, but one of my colleagues pointed me at https://gitlab.com/apparmor/apparmor/-/commit/af6dfe5b810d5b34e6d1c0f06ce01c2a96df0dc4

Is your "other debian system" also running Debian Bookworm, with the same package versions as Raspberry Pi OS?

On my "out of the box" installation of Raspberry Pi OS Bookworm it seems that I have `aa-enabled`, `aa-exec`, `aa-features-abi`, `aa-remove-unknown`, `aa-status` and `aa-teardown` commands; but not `aa-logprof` nor `aa-genprof` ?

lurch, 2025-02-18:
Please feel free. If nothing else, upstream will have a much better understanding of Apparmor than I do! 🤣 
(but please do report the upstream issue link here afterwards)

MagicSimi, 2025-03-05:
After some research, I am pretty sure, the problem stems from a mismatch of kernel and apparmor-utils. While the newer kernel adds the class field to the audit messages (see commit [8c4b785a86be](https://github.com/torvalds/linux/commit/8c4b785a86be)),  apparmor-utils in the version 3.0.8 doesn't have this field implemented yet. This should be fixed in debian testing, and will therefore also be fixed in the next stable Raspberry Pi OS release. In the meantime, any user can use my sed line from above (or one with even better syntax) the feed the log entries into apparmor.

I think the report can be closed.

closed 2025-03-05 by MagicSimi

lurch, 2025-03-06:
Thank you for the update.