raspberrypi/bookworm-feedback #374: Chromium ignores `--force-dark-mode` and follows gsettings instead

Botspot, 2025-02-18:
This is an update on https://github.com/RPi-Distro/chromium/issues/41 and https://github.com/RPi-Distro/repo/issues/364.
It would appear that something related to Bookworm causes Chromium to follow the gsettings preferred color theme over the command-line flag.

On Bullseye this was enough to make Chromium launch with a dark mode:
```
chromium-browser --force-dark-mode
```
But on Bookworm the flag is ignored and this is required:
```
gsettings set org.gnome.desktop.interface color-scheme 'prefer-dark'
chromium
```
I discovered this gsettings workaround today.

It is worth noting that during a 2-week period last year, chromium received no updates but this issue was fixed. At the end of the 2 weeks the bad behavior returned.
For a while I assumed this was a local issue to my install, but then [this user opened an issue](https://github.com/Botspot/pi-apps/issues/2713#issuecomment-2616691798) reporting the same issue to me.

Botspot, 2025-02-18:
Preventing Chromium from talking to dbus is one possible workaround that avoids changing system-wide settings.
```
env --unset=DBUS_SESSION_BUS_ADDRESS chromium --force-dark-mode
```
or
```
dbus-run-session chromium --force-dark-mode
```
It also is a bad idea and causes a lot of errors.

lurch, 2025-02-18:
ping @cillian64 - is this a bug or expected behaviour?

cillian64, 2025-02-19:
I see the same on vanilla Debian bookworm with debian's chromium package (not raspberry pi os).  I'm not sure why the option isn't doing anything, but this looks like an upstream chromium issue, not an rpi-os thing.  Googling around turns up lots of people having similar issues on other platforms.

closed 2025-02-19 by lurch