raspberrypi/bookworm-feedback #375: Camera not found on Chrome / Firefox

mghadam, 2025-02-18:
### Describe the bug

PipeWire does not detect my Logitech USB camera, causing Chrome and Firefox to display a "Camera Not Found" error. Running `wpctl status` shows no video or audio devices.

As per [this guide](https://jgrulich.cz/2024/12/13/when-your-webcam-doesnt-work-solving-firefox-and-pipewire-issues/), I can get the usb camera working by disabling `PipeWire Camera support` on Chromium or `media.webrtc.camera.allow-pipewire` on Firefox. 

Is this a pipewire issue? 

### Steps to reproduce the behaviour

Try launching a google meet session on Firefox / Chromium 

### Device (s)

Raspberry Pi 5

### System

Linux pi4 6.6.74+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux

Linux home 6.6.74+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux

### Additional context

[image]
[image]

popcornmix, 2025-02-18:
I've moved this to bookworm-feedback as it's clearly not a linux kernel issue (changing settings in browser makes it work).
@cillian64 may be able to explain what should/shouldn't work.

cillian64, 2025-02-18:
I've seen this a few times and it's usually a permissions issue. If you accidentally deny camera permission the first time (easily done by accidentally closing the dialog or hitting esc) then that is remembered forever and the camera never works.  This is the xdg-desktop-portal permission, separate to the permission dialog seen inside the browser per-website.

If it's that then you can fix it by deleting or renaming `~/.local/share/flatpak/db/devices` then rebooting.

We're working on a fix for this issue by either making all camera-denials temporary or adding some sort of indication when camera is being refused because of this.

lurch, 2025-02-18:
See also #354

mghadam, 2025-02-18:
I deleted the `~/.local/share/flatpak/db/devices` and rebooted the Raspberry Pi. It prompted me to allow webcam access again, but after granting permission, the same issue persisted and was not resolved.

cillian64, 2025-02-18:
We released a bad version of libcamera last week which broke pipewire-libcamera if you were unlucky enough to upgrade in the couple of days before we pushed the fix.  Please could you upgrade `libcamera0.4` (`sudo apt update; sudo apt install libcamera0.4`), or check which version you've currently got (paste the output from `apt policy libcamera0.4`) ?

mghadam, 2025-02-18:
I think that is the case then as I have a cronjob in place to update packages every night.

Here is the version I have:
```
# apt policy libcamera0.4
libcamera0.4:
  Installed: 0.4.0+rpt20250213-1
  Candidate: 0.4.0+rpt20250213-1
  Version table:
 *** 0.4.0+rpt20250213-1 500
        500 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages
        100 /var/lib/dpkg/status
root@home:~#
```

cillian64, 2025-02-18:
Ok, libcamera `0.4.0+rpt20250213-1` has the fix so if it's still not working then that's not the problem.

Could you let me know exactly which model of webcam it is?  And are you using 64-bit or 32-bit rpi-os?

Please could you paste or attach the output from `journalctl --user -u pipewire --boot` and `journalctl --user -u wireplumber --boot` after attempting to use the webcam?

And finally, would it be possible to test on a new clean install of raspberry pi OS?

I've successfully used USB webcams via pipewire in chromium and firefox recently, and our nightly firefox/chromium tests cover pipewire-libcamera in browsers, so this /should/ work...