raspberrypi/bookworm-feedback #376: labwc: Missing/incorrect borders

theoldwizard1, 2025-02-22:
From: Jack Patteeuw <jack.patteeuw@gmail.com>
Date: 2/21/25, 10:00
Package: labwc
Version: 0.8.1-1+rpt3
Severity: normal

Dear Maintainer,

-- Problem Description

	**Some windows do not receive any perceptible border.  Most notably, Thunderbird.**

	$ ls -l $HOME/.config/labwc/themerc-override
	-rw-r--r-- 1 jack jack 130 Feb 21 09: 28  /home/jack/.config/labwc/themerc-override

	$ cat $HOME/.config/labwc/themerc-override
	border.width: 3
	window.active.border.color: #fc0303
	window.active.title.bg.color: #87919B
	window.active.label.text.color: #F0F0F0

-- System Information:
Debian Release: 12.9
  APT prefers stable-updates
  APT policy: (500, 'stable-updates'), (500, 'stable-security'), (500, 'stable')
Architecture: arm64 (aarch64)
Foreign Architectures: armhf

Kernel: **Linux 6.6.74+rpt-rpi-2712 (SMP w/4 CPU threads; PREEMPT)**
Kernel taint flags: TAINT_CRAP
Locale: LANG=en_US.UTF-8, LC_CTYPE=en_US.UTF-8 (charmap=UTF-8) (ignored: LC_ALL set to en_US.UTF-8), LANGUAGE=en_US.UTF-8
Shell: /bin/sh linked to /usr/bin/dash
Init: systemd (via /run/systemd/system)

Versions of packages labwc depends on:
ii  libc6                2.36-9+rpt2+deb12u9
ii  libcairo2            1.16.0-7+rpt1
ii  libglib2.0-0         2.74.6-2+deb12u5
ii  libinput10           1.22.1-1
ii  libpango-1.0-0       1.50.12+ds-1
ii  libpangocairo-1.0-0  1.50.12+ds-1
ii  libpixman-1-0        0.42.2-1+rpt3
ii  libpng16-16          1.6.39-2
ii  librsvg2-2           2.54.7+dfsg-1~deb12u1
ii  libwayland-server0   1.23.0-1~bpo12+1+rpt1
ii  libwlroots-0.18      0.18.1-1+rpt11
ii  libxcb1              1.15-1
ii  libxkbcommon0        1.5.0-1
ii  libxml2              2.9.14+dfsg-1.3~deb12u1
ii  xwayland             2:22.1.9-1+rpt1

labwc recommends no packages.

Versions of packages labwc suggests:
ii  kanshi                            1.5.1-1~bpo11+rpt1
ii  lxterminal [x-terminal-emulator]  0.4.0-2+rpt1
pn  mako-notifier                     &lt;none&gt;
pn  swaybg                            &lt;none&gt;
ii  swayidle                          1.8.0-1
pn  waybar                            &lt;none&gt;</pre></div>Subject: 
labwc: Missing/incorrect borders

lurch, 2025-02-24:
@Consolatis Is this a Raspberry Pi OS problem, or an upstream labwc problem?

theoldwizard1, 2025-02-25:
Based on the response

> Subject: Re: Bug#1098503: labwc: Missing/incorrect borders
> From: Birger Schacht <birger@debian.org>
> Date: 2/21/2025, 12:12 PM
> 
> We don't have a **0.8.1-1+rpt3** version of labwc in Debian. It seems to me that this is some kind of downstream raspberry pi
> thing
> Please file the bug report at the project that distributes that package.

It sound like some from Raspberry Pi MODIFIED labwc !

Consolatis, 2025-02-25:
> Some windows do not receive any perceptible border.  Most notably, Thunderbird.

I am not quite sure what the actual issue is. I assume this is about CSD applications (client side deocation). In that case its the application / ui-framework responsibility to render borders / titlebars. There are likely ways to tell Thunderbird to use server-side decorations instead but I am not sure how.

> Is this a Raspberry Pi OS problem, or an upstream labwc problem?

If my assumption is true, its a application issue. I think the only potential for Raspberry to cause this would be if the default config for [decorations](https://labwc.github.io/labwc-config.5.html#core) is set to `client` or if the default Qt / GTK theme is broken.

theoldwizard1, 2025-02-25:
> I am not quite sure what the actual issue is. I **assume** this is about CSD applications (client side deocation). In that case its the application / ui-framework responsibility to render borders / [...]

I would be nice if someone could CONFIRM your hypothesis , instead of just guessing.  I have seen it on other applications that I can not list at the moment.

If RPi OS wants to be considered a "production" OS, consistent "look and feel" is important to the end user !

JamesH65, 2025-02-25:
> If RPi OS wants to be considered a "production" OS, consistent "look and feel" is important to the end user !

It's been a production OS for over 10 years and is used in millions of installations worldwide. 

Occasionally, when moving to new codebases, problems not found during initial testing turn up, and then need to be diagnosed and fixed. This does not happen instantly. So confirmation or not of the hypothesis will happen as and when engineering time allows it to be done.

lurch, 2025-02-25:
@theoldwizard1 We're a small team, and we're all quite busy with multiple things. Please be patient, and we'll get round to looking at this low-priority issue in due course.

(BTW @spl237 has put in a **LOT** of effort over the years giving Raspberry Pi OS a consistent look and feel, but we obviously don't have the time to audit every single 3rd-party Open Source application 😂  Understandably, most of the effort is focused on the applications that we include by default in Raspberry Pi OS.)

theoldwizard1, 2025-02-25:
> [@theoldwizard1](https://github.com/theoldwizard1) We're a small team, and we're all quite busy with multiple things. Please be patient, and we'll get round to looking at this low-priority issue in [...]

**I accept those statements !**  ASSUMING it is the other person's fault _(and then potentially ignoring the stated problem)_, does not project a "warm feeling".

The fact that RPT modified labwc from the mainline used by Debian seem odd.

lurch, 2025-02-25:
> The fact that RPT modified labwc from the mainline used by Debian seems odd.

There are _lots_ of packages that we modify, largely to make them more optimised for Raspberry Pi hardware (which is obviously very different to the standard x86 PC hardware that mainline Debian primarily targets). Eventually we port our local modifications back to the upstream projects, but this obviously takes time.

ghollingworth, 2025-02-25:
> > > Some windows do not receive any perceptible border.  Most notably, Thunderbird.
> > 
> > 
> > I am not quite sure what the actual issue is. I **assume** this is about CSD applications (client side decoration). In that case its the application / ui-framework responsibility to render borders / titlebars. There are likely ways to tell Thunderbird to use server-side decorations instead but I am not sure how.
> 
> I would be nice if someone could CONFIRM your hypothesis , instead of just guessing. I have seen it on other applications that I can not list at the moment.
> 
> If RPi OS wants to be considered a "production" OS, consistent "look and feel" is important to the end user !

I can CONFIRM Consolatis (the guy who maintains labwc) is correct...  When you start Thunderbird the decoration is drawn by it (actually Firefox)

So the problem is actually in the application.

closed 2025-02-25 by ghollingworth

theoldwizard1, 2025-02-25:
> I can CONFIRM Consolatis (the guy who maintains labwc) is correct... When you start Thunderbird the decoration is drawn by it (actually Firefox)
> 
> So the problem is actually in the application.

Thank you !

Seems strange that labwc would go through the effort of setting up setting up "themes" and then application would not use them.

Then again, Thunderbird and other apps have to live on top of many different "compositors" and desktop environments.

GuiFlam, 2025-04-09:
I have the same issue on Arch Linux. As you can see, there's a border around my kitty window, but no border is seen around the firefox window.

[image]

theoldwizard1, 2025-04-09:
Here is how I finally SOLVED this problem !

```
$ cat ~/.config/labwc/themerc-override 
border.width: 3
window.active.border.color: #ED1C24
```

Plus, some applications, notably thunderbird, TURN OFF title/decoration by default !