raspberrypi/bookworm-feedback #377: chromium high lagging

qrp73, 2025-02-24:
The issue with high OS lagging when you open some web pages with video on chromium browser happens very often. It usually happens on telegram web pages. I have video example with which this issue happens, but I don't want to share it due to private content. I can examine video details like codec, format and other details and share it.

In order to reproduce it just needs to open https://t.me/**** page with video, click to play the video and it leads to a high OS lagging, the mouse freezes for 3-4 seconds, then it resumes for some 0.1 second and freezes again in a loop. The same happens with music playing in background on vlc. It's almost impossible to do something during OS freezes, because system is freeze and don't respond, you cannot close browser, cannot open task manager, cannot switch to a virtual terminal, etc. But even if you're lucky to switch to a virtual terminal during 0.1 sec unfreeze interval, it still continue freeze even on virtual terminal, so you cannot type something with keyboard.

During these freeze GPU load graph shows >90% load. 
dmesg don't have any messages during these freeze.

Killing chromium browser leads to stop system freeze and system resume to work as expected. But I found that if I reproduce it several times (two in my case), sometimes the system going into some strange state that background processes continue to works, but display content is frozen and don't update anymore and nothing helps to exit from that state - only system reboot.

Here is chromium terminal output during issue:
```
$ uname -a
Linux raspi 6.6.74+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux

$ chromium --version
Chromium 130.0.6723.116 built on Debian GNU/Linux 12 (bookworm)

$ chromium

(chromium:2492): dbind-WARNING **: 16:50:28.175: AT-SPI: Error retrieving accessibility bus address: org.freedesktop.DBus.Error.ServiceUnknown: The name org.a11y.Bus was not provided by any .service files
close object 1: Invalid argument
close object 2: Invalid argument
close object 3: Invalid argument
close object 1: Invalid argument
close object 3: Invalid argument
close object 4: Invalid argument
[2598:11:0224/165032.311042:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
[2598:11:0224/165032.312127:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
[2492:2492:0224/165112.271621:ERROR:wayland_data_drag_controller.cc(151)] Invalid state when trying to start drag. source=kMouse
^C[2492:2492:0224/165149.016615:ERROR:wayland_event_watcher.cc(47)] libwayland: warning: queue 0x3c000850c0 destroyed while proxies still attached:

[2492:2492:0224/165149.016905:ERROR:wayland_event_watcher.cc(47)] libwayland:   wl_shm_pool#37 still attached
```

the issue is started from this message:
```
[2598:11:0224/165032.311042:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
```

then I tried to switch to a terminal with mouse to stop the process (it's very hard when system freezes, so it takes a long time to switch into terminal window and stop process with Ctrl+C, Ctrl+Z):
```
[2492:2492:0224/165112.271621:ERROR:wayland_data_drag_controller.cc(151)] Invalid state when trying to start drag. source=kMouse
```

I copied video url and downloaded it, this is mp4 file.

I tried to play this file with VLC and it is played ok with no freezes, with good fps and about 20% GPU load.

I tried to open video with chromium browser and it is also played ok with no freezes, with good fps and about 20% GPU load.

It appears that telegram page playing this video using some browser feature which leads to ```[2598:11:0224/165032.311042:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1``` error in the browser and OS lagging. But I'm not sure how to check which feature is used exactly.

Here is mp4 video file details shown in ffprobe:
```
$ ffprobe test.mp4
ffprobe version 5.1.6-0+deb12u1+rpt1 Copyright (c) 2007-2024 the FFmpeg developers
  built with gcc 12 (Debian 12.2.0-14)
  configuration: --prefix=/usr --extra-version=0+deb12u1+rpt1 --toolchain=hardened --incdir=/usr/include/aarch64-linux-gnu --enable-gpl --disable-stripping --disable-mmal --enable-gnutls --enable-ladspa --enable-libaom --enable-libass --enable-libbluray --enable-libbs2b --enable-libcaca --enable-libcdio --enable-libcodec2 --enable-libdav1d --enable-libflite --enable-libfontconfig --enable-libfreetype --enable-libfribidi --enable-libglslang --enable-libgme --enable-libgsm --enable-libjack --enable-libmp3lame --enable-libmysofa --enable-libopenjpeg --enable-libopenmpt --enable-libopus --enable-libpulse --enable-librabbitmq --enable-librist --enable-librubberband --enable-libshine --enable-libsnappy --enable-libsoxr --enable-libspeex --enable-libsrt --enable-libssh --enable-libsvtav1 --enable-libtheora --enable-libtwolame --enable-libvidstab --enable-libvorbis --enable-libvpx --enable-libwebp --enable-libx265 --enable-libxml2 --enable-libxvid --enable-libzimg --enable-libzmq --enable-libzvbi --enable-lv2 --enable-omx --enable-openal --enable-opencl --enable-opengl --enable-sand --enable-sdl2 --disable-sndio --enable-libjxl --enable-neon --enable-v4l2-request --enable-libudev --enable-epoxy --libdir=/usr/lib/aarch64-linux-gnu --arch=arm64 --enable-pocketsphinx --enable-librsvg --enable-libdc1394 --enable-libdrm --enable-vout-drm --enable-libiec61883 --enable-chromaprint --enable-frei0r --enable-libx264 --enable-libplacebo --enable-librav1e --enable-shared
  libavutil      57. 28.100 / 57. 28.100
  libavcodec     59. 37.100 / 59. 37.100
  libavformat    59. 27.100 / 59. 27.100
  libavdevice    59.  7.100 / 59.  7.100
  libavfilter     8. 44.100 /  8. 44.100
  libswscale      6.  7.100 /  6.  7.100
  libswresample   4.  7.100 /  4.  7.100
  libpostproc    56.  6.100 / 56.  6.100
Input #0, mov,mp4,m4a,3gp,3g2,mj2, from 'test.mp4':
  Metadata:
    major_brand     : isom
    minor_version   : 512
    compatible_brands: isomiso2avc1mp41
    copyright       : 
    track           : 0
    artist          : 
    album           : 
    comment         : 
    date            : 0
    genre           : 
    title           : 
    encoder         : Lavf58.44.100
    composer        : 
  Duration: 00:00:26.52, start: 0.000000, bitrate: 4308 kb/s
  Stream #0:0[0x1](und): Video: h264 (Main) (avc1 / 0x31637661), yuv420p(tv, unknown/reserved/reserved, progressive), 688x1280 [SAR 1:1 DAR 43:80], 3993 kb/s, 30 fps, 30 tbr, 15360 tbn (default)
    Metadata:
      handler_name    : VideoHandler
      vendor_id       : [0][0][0][0]
  Stream #0:1[0x2](eng): Audio: aac (LC) (mp4a / 0x6134706D), 44100 Hz, stereo, fltp, 319 kb/s (default)
    Metadata:
      handler_name    : SoundHandler
      vendor_id       : [0][0][0][0]
```

qrp73, 2025-02-24:
> Is this the same as the other issues you've reported with the https://t.me website ( [#212](https://github.com/raspberrypi/bookworm-feedback/issues/212) [#210](https://github.com/raspberrypi/bookworm-feedback/issues/210) [#171](https://github.com/raspberrypi/bookworm-feedback/issues/171) ), where it turned out that it was just slow because the website was using Javascript in a really inefficient way?

No, the problem is not because it shows slow web page. The problem is that it breaks Operating System, it stops all processes (include realtime) for 3-4 seconds, then give a small time slice about 0.1 second and then again stops all processes for 3-4 seconds. As result, the operating system freezes and you even can't move mouse, can't close window, can't do anything with OS. 

This is not website issue, this is major issue of Operating System. Because it affects all processes in the system, not just web page.

Regarding to web page, yes it works slowly, but it's not clear why - because OS freezes or because website was using Javascript in a really inefficient way. But in any case it definitely shows that the web site triggers chromium browser error:
```
[2598:11:0224/165032.311042:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
```

it means that the issue is not with web page and not with javascript, but with chromium browser or with OS code issue, as I understand this is juse som issue with hardware accelerated codec. So the issue is within Raspberry Pi firmware for GPU or with chromium browser which uses it in the wrong way.

> Unfortunately all of the links in those previous issues now just say "Please open Telegram to view this post" and I get an xdg-open popup request. I'm afraid I don't have Telegram installed, and neither do I have a Telegram account.

I have the same message about xdg-open popup, unfortunately there is no way to disable it. Just click cancel and open web page with browser instead of app. There is no need to install Telegram app in order to reproduce the issue, because it happens when you visit web page, not app.

> This seems to indeed indicate that the problem isn't with the video file, but with the website doing something "strange".

No, I'm sure that the issue with codec format used in specific video file, because other mp4 videos are played ok on Telegram web site.

When you run chromium from terminal, you can see its log messages, and when this issue happens you will see this error:
```
[2598:11:0224/165032.311042:ERROR:ffmpeg_common.cc(970)] Unsupported pixel format: -1
```

Web site and javascript cannot call ffmpeg libraries directly. This is a part of chromium browser code calls this code.

When I play other videos on Telegram web site, they are played ok and this error didn't happens.

And by the way exactly the same error with the same OS freeze also happens on other sites which playing video files. I mention Telegram web site just as example which I recorded and can easily reproduce.

qrp73, 2025-02-24:
@lurch If you believe that a web page can cause a complete halt of all operating system kernel processes and consider this normal because it's a problem with the web page, it raises legitimate doubts about your qualifications. Any reasonably competent tester or developer would never write such nonsense.

qrp73, 2025-02-24:
> Is this the same as the other issues you've reported with the https://t.me website ( #212 #210 #171 ), where it turned out that it was just slow because the website was using Javascript in a really inefficient way?

Regarding issues that you mentioned (#212, etc) - No, this issue is not related, they are very different, I was able to close browser and there was no such error message in the log. 

These issues which you mentioned was happened not because "bad wevbsite and inefficient JS" as you false wrote, but due to a bug with memory alignment mistake in Raspberry Pi GPU firmware. As I remember it was used 32 bit alignment on 64-bit system instead of 64-bit alignment. And it was fixed in GPU firmware update, it don't happens anymore.

This bug is much worse, it leads to a total Linux kernel freeze for 3-4 seconds with a small timeslice which happens in a loop.

timg236, 2025-02-24:
Deleted comment which would normally get the ussue closed as 'too heated'

qrp73, 2025-02-25:
@timg236, I have noticed that this user @lurch consistently engages in disruptive behavior—closing issues on questionable grounds, making unhelpful comments, and requiring excessive explanations for basic concepts that should be clear to any competent tester or developer. This pattern raises concerns that their actions may be intentional sabotage or trolling. It happens in almost every issue they comment on, and I don't recall a single good or professional comment from this user.  

Therefore, I suspect he may be a corporate saboteur or something like that, and I would like to escalate this matter to management for review and clarification of their role. Can you help with that?

timg236, 2025-02-25:
Disagreeing with extremely valuable bug triaging does not give anyone the right to insult individuals. 30 day ban for qrp73

ghollingworth, 2025-02-25:
This is just the same issue as before, unless you believe they have fixed their javascript

[image]