raspberrypi/bookworm-feedback #378: gnome-calculator freeze

qrp73, 2025-02-24:
I just started gnome-calculator, it opened the window, but then unexpectedly it freeze and don't respond on mouse, I cannot move window, cannot close it:

[image]

I can select context menu in labwc pi-panel by clicking on calculator icon, but selecting Close don't works, it stays in that frozen state.

I attached to the process with gdb and here is callstack taken in this state:
```
$ ps aux | grep gnome-calculator
pi         62021  0.9  3.8 1002648 149216 ?      Sl   01:05   0:01 gnome-calculator
pi         62345  0.0  0.0   6220  1920 pts/1    S+   01:08   0:00 grep --color=auto gnome-calculator
pi@raspi:~ $ gdb -p 62021
GNU gdb (Debian 13.1-3) 13.1
Copyright (C) 2023 Free Software Foundation, Inc.
License GPLv3+: GNU GPL version 3 or later <http://gnu.org/licenses/gpl.html>
This is free software: you are free to change and redistribute it.
There is NO WARRANTY, to the extent permitted by law.
Type "show copying" and "show warranty" for details.
This GDB was configured as "aarch64-linux-gnu".
Type "show configuration" for configuration details.
For bug reporting instructions, please see:
<https://www.gnu.org/software/gdb/bugs/>.
Find the GDB manual and other documentation resources online at:
    <http://www.gnu.org/software/gdb/documentation/>.

For help, type "help".
Type "apropos word" to search for commands related to "word".
Attaching to process 62021
[New LWP 62024]
[New LWP 62025]
[New LWP 62027]
[New LWP 62034]
[New LWP 62036]
[Thread debugging using libthread_db enabled]
Using host libthread_db library "/lib/aarch64-linux-gnu/libthread_db.so.1".
syscall () at ../sysdeps/unix/sysv/linux/aarch64/syscall.S:38
38	../sysdeps/unix/sysv/linux/aarch64/syscall.S: No such file or directory.
(gdb) bt full
#0  syscall () at ../sysdeps/unix/sysv/linux/aarch64/syscall.S:38
#1  0x0000007faa1cd06c in  () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#2  0x0000007fa92452c8 in  () at /lib/aarch64-linux-gnu/libsoup-3.0.so.0
#3  0x0000007faa0b71f8 in g_closure_invoke () at /lib/aarch64-linux-gnu/libgobject-2.0.so.0
#4  0x0000007faa0cb5ac in  () at /lib/aarch64-linux-gnu/libgobject-2.0.so.0
#5  0x0000007faa0d18dc in g_signal_emit_valist () at /lib/aarch64-linux-gnu/libgobject-2.0.so.0
#6  0x0000007faa0d1af0 in g_signal_emit () at /lib/aarch64-linux-gnu/libgobject-2.0.so.0
#7  0x0000007faa0bb568 in  () at /lib/aarch64-linux-gnu/libgobject-2.0.so.0
#8  0x0000007faa0be9b4 in g_object_notify_by_pspec () at /lib/aarch64-linux-gnu/libgobject-2.0.so.0
#9  0x0000007fa922a604 in  () at /lib/aarch64-linux-gnu/libsoup-3.0.so.0
#10 0x0000007faa177614 in g_main_context_dispatch () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#11 0x0000007faa1779e0 in  () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#12 0x0000007faa177a84 in g_main_context_iteration () at /lib/aarch64-linux-gnu/libglib-2.0.so.0
#13 0x0000007fa96ff3a8 in g_application_run () at /lib/aarch64-linux-gnu/libgio-2.0.so.0
#14 0x0000005588839f8c in main ()
(gdb) 
```

Trying to kill process and start it again leads to the same issue... Previously it was worked ok.

qrp73, 2025-02-24:
Some update. Just inspected with WireShark, when I open gnome-calculator, it opens HTTPS connection with www.imf.org

What the heck???

lurch, 2025-02-25:
Closing as this is an upstream issue, not specific to Raspberry Pi OS.

(also, Raspberry Pi OS includes `galculator` by default, rather than `gnome-calculator`)

closed 2025-02-25 by lurch

timg236, 2025-02-25:
Currency conversion rates
https://gitlab.gnome.org/GNOME/gnome-calculator/-/blob/40e931855701a6d64cdd88dd86b24ad235a14f9a/lib/currency-provider.vala#L376

wobbo, 2025-03-11:
I have [GNOME](https://forums.raspberrypi.com/viewtopic.php?t=373028) installed on Raspberry Pi OS Lite and I have the same problem.

```
gnome-calculator 

** (gnome-calculator:4587): WARNING **: 10:18:49.288: currency-provider.vala:161: Couldn't download IMF currency rate file: HTTP/2 Error: INTERNAL_ERROR

(gnome-calculator:4587): libsoup-WARNING **: 10:18:49.288: (../libsoup/soup-session.c:334):soup_session_dispose: runtime check failed: (soup_connection_manager_get_num_conns (priv->conn_manager) == 0)

(gnome-calculator:4587): libsoup-WARNING **: 10:18:49.288: (../libsoup/soup-connection-manager.c:78):soup_host_free: runtime check failed: (host->conns == NULL)
```

lurch, 2025-03-11:
As already mentioned, that is an upstream problem, not something specific to Raspberry Pi OS. When the fix lands in Debian Bookworm, Raspberry Pi OS will also pick up that fixed version.

JedMeister, 2025-06-16:
> As a temporary workaround you can disconnect ethernet cable when starting gnome-calculator.

To save others a click and a bit of reading; (IMO) a superior workaround is:

```
dconf write /org/gnome/calculator/refresh-interval 0
```