raspberrypi/bookworm-feedback #379: Network Manager MAC randomisation causes denial of service

miniwg, 2025-02-26:
Situation:
On a network with a small pool of DHCP addresses, MAC randomisation, which causes the MAC address to change on each restart or reconnection to the network can quicky exhaust the pool of DHCP addresses on a network with a small DHCP pool.

Problem:
RPI% with two NIC cards.  While there are several guides to disabling MAC randomisation on eth0, there is no guidance to how to do so on a second NIC eth1.

Solution:
Wired eth connections should default to fixed MAC addresses as in the vast majority of cases the will be connected to an internal LAN where MAC randomisation is neither necessary nor appropriate,

lurch, 2025-02-26:
I'v just done a number of reboot and power-cycle loops on a Raspberry Pi 5 running the latest version of Raspberry Pi OS, and `ifconfig` reports that I always get the same MAC and IP addresses on both the `eth0` and `wlan0` connection.

Can you please confirm which model of Raspberry Pi you're using, which OS and version you're using, and which "NIC cards" you're using? And what changes have you made to the "out of the box" OS configuration?
Thanks!

6by9, 2025-02-26:
AFAIK MAC address randomisation is not enabled by default on Raspberry Pi OS. It certainly isn't on any of my recent installs, and I do use an RTL8153 USB network adapter as well as the standard ethernet port on my normal development setup.

Knowing what network adapter you're using (presumably `lsusb`, but could be `lspci`), and the kernel version you're on (`uname -a`) will be key.

There was a bug in the AX88179 driver a while back that resulted in not reading the MAC address from the EEPROM, hence a random one was assigned. That was fixed back in April 2024 (https://github.com/raspberrypi/linux/commit/426366d577e9712b2cabc95fc899d4f6f85105ba and https://github.com/raspberrypi/linux/commit/bc835b83899997c01084aeb92ce9d8c25cc13c12)

miniwg, 2025-02-27:
In answer to the questions above the board is a RPi 5 with 8GB RAM.  The affected NIC is a second NIC eth1 connected to PCI. lshw reports it as: RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller.  As observed above the main NIC on the Pi board itself is not affected. lspci reports the same but adds (rev15).

uname -a  returns : 6.6.74+rpt-rpi-2712 #1 SMP PREEMPT Debian 1:6.6.74-1+rpt1 (2025-01-27) aarch64 GNU/Linux

It is a completely vanilla install of Bookworm

lurch, 2025-02-27:
I did some searching and found https://fixes.brecht-schule.hamburg/linux/linux-tutorials/set-persistent-mac-rtl8111
Does that help?

6by9, 2025-02-27:
> I did some searching and found https://fixes.brecht-schule.hamburg/linux/linux-tutorials/set-persistent-mac-rtl8111 Does that help?

My suspicion was that it may have been a limitation of the network adapter, and that page implies that it isn't uncommon on cards using the r8169 driver.
Do you get the `can't read MAC address, setting random one` error logged in the kernel logs? That was added via https://github.com/torvalds/linux/commit/c782e204f7343b6f4527e953a5ad265be4abd627 back in July 2019, so it's not new.

Does this card pick up a valid MAC address when used on any other system?

Having had a quick look at the r8169 driver code, it is calling `of_get_mac_address` to allow configuring the MAC address via device tree. That means you could create a dtoverlay that defined a suitable MAC address as `mac-address` or `local-mac-address` for the relevant device tree node. The location of that node is going to be something like `/axi/pcie@1100000/<something>`.
The firmware does a similar thing to set the MAC address on SMSC9514 and LAN7800 using `.../usb/usb-port@1/ethernet@1`

miniwg, 2025-02-27:
The `can't read MAC address, setting random one` error is logged in the kernel logs.

Thanks for the link to the set persistent MAC address - I think that is the solution for the moment.  

In the interim, the workaround I have used it to get the hardware MAC address using `lshw`. Then use `nmtui`, while user root, to edit the connection which on my Pi, with the setup listed above, is shown as 'Wired connection 2'.  Tab down and show the ethernet section, then enter the hardware MAC address in the 'Cloned MAC address' field and save the changes.

6by9, 2025-02-27:
If the Pi can't read the MAC address, then I'd be surprised if any Linux system can using the standard drivers.
Can you provide a link to the specific card you're using?

A little more googling found me https://github.com/LekKit/RVVM/issues/118#issuecomment-1874714434
If I read that right, the MAC address can either be burnt into the Realtek chip or in an external EEPROM. Linux only reads the value from the Realtek chip.

It looks like there was a patch to read the EEPROM (see https://patchwork.ozlabs.org/project/netdev/patch/48D9FE65.40709@redhat.com/), but it was reverted due to issues (https://lkml.indiana.edu/hypermail/linux/kernel/0810.3/0838.html). 2008 is a long time ago!

What details does `lshw` provide for your LAN adapter? It seems odd if that is giving a MAC address and yet the driver can't read it. 
I'm afraid it's likely to remain a quirk of this particular driver though - it's not the sort of patch we have a major need to try to fix and upstream.

miniwg, 2025-02-27:
The interface is provided as part of this kit:
[https://www.waveshare.com/product/raspberry-pi/boards-kits/raspberry-pi-5/pi5-module-box.htm?sku=27831](url)

`lshw`:
`        *-network
             description: Ethernet interface
             product: RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller
             vendor: Realtek Semiconductor Co., Ltd.
             physical id: 0
             bus info: pci@0000:01:00.0
             logical name: eth1
             version: 15
             serial: 9a:0c:53:87:xx:xx
             size: 1Gbit/s
             capacity: 1Gbit/s
             width: 64 bits
             clock: 33MHz
             capabilities: pm msi pciexpress msix bus_master cap_list ethernet physical tp mii 10bt 10bt-fd 100bt 100bt-fd 1000bt-fd autonegotiation
             configuration: autonegotiation=on broadcast=yes driver=r8169 driver version=6.6.74+rpt-rpi-2712 duplex=full firmware=rtl8168h-2_0.0.2 02/26/15 ip=192.168.1.13 latency=0 link=yes multicast=yes port=twisted pair speed=1Gbit/s
             resources: irq:38 memory:1b80004000-1b80004fff memory:1b80000000-1b80003fff`

I appreciate that this is in reality an edge case hardware issue, but it would be good if there was a readme / howto post or article somewhere that other people who face the same issue don't have to flounder around for some time like I did.

Best

6by9, 2025-02-27:
> The interface is provided as part of this kit: https://www.waveshare.com/product/raspberry-pi/boards-kits/raspberry-pi-5/pi5-module-box.htm?sku=27831
> 
> ```
> `lshw`:
> `        *-network
>              description: Ethernet interface
>              product: RTL8111/8168/8411 PCI Express Gigabit Ethernet Controller
>              vendor: Realtek Semiconductor Co., Ltd.
>              physical id: 0
>              bus info: pci@0000:01:00.0
>              logical name: eth1
>              version: 15
>              serial: 9a:0c:53:87:xx:xx
>              size: 1Gbit/s
>              capacity: 1Gbit/s
>              width: 64 bits
>              clock: 33MHz
>              capabilities: pm msi pciexpress msix bus_master cap_list ethernet physical tp mii 10bt 10bt-fd 100bt 100bt-fd 1000bt-fd autonegotiation
>              configuration: autonegotiation=on broadcast=yes driver=r8169 driver version=6.6.74+rpt-rpi-2712 duplex=full firmware=rtl8168h-2_0.0.2 02/26/15 ip=192.168.1.13 latency=0 link=yes multicast=yes port=twisted pair speed=1Gbit/s
>              resources: irq:38 memory:1b80004000-1b80004fff memory:1b80000000-1b80003fff`
> ```

That's cheap enough I'll grab one to see where that serial number is coming from.

9a:0c:53 has bit 1 set in the top octet, so is a locally administered address, not a uniquely assigned one. https://en.wikipedia.org/wiki/MAC_address#Address_details

> I appreciate that this is in reality an edge case hardware issue, but it would be good if there was a readme / howto post or article somewhere that other people who face the same issue don't have to flounder around for some time like I did.

That would be for the Waveshare product description or Wiki.

lurch, 2025-02-28:
> That would be for the Waveshare product description or Wiki.

Agreed. I'll close this issue as a workaround has been identified.

closed 2025-02-28 by lurch