raspberrypi/bookworm-feedback #380: gnome-terminal fails to launch - Error calling StartServiceByName for org.gnome.Terminal: Timeout was reached

Botspot, 2025-03-01:
At least 3 separate users have all reported the same thing.
- https://github.com/Botspot/bvm/issues/15
- https://github.com/Botspot/pi-apps/issues/2707#issuecomment-2614074998
- https://github.com/Botspot/pi-apps/issues/2717

gnome-terminal, once installed, will make itself the default x-terminal-emulator but fails to launch in some circumstances.
It seems to be caused by the locale not being UTF8. I think the core issue is figuring out why the locale is not being set correctly.
On my system the output of `locale` is
```
LANG=en_US.UTF-8
LANGUAGE=en_US.UTF-8
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=en_US.UTF-8
```
But on an affected user's system, the locale is
```
LANG=en_US
LANGUAGE=en_US
LC_CTYPE="en_US"
LC_NUMERIC="en_US"
LC_TIME="en_US"
LC_COLLATE="en_US"
LC_MONETARY="en_US"
LC_MESSAGES="en_US"
LC_PAPER="en_US"
LC_NAME="en_US"
LC_ADDRESS="en_US"
LC_TELEPHONE="en_US"
LC_MEASUREMENT="en_US"
LC_IDENTIFICATION="en_US"
LC_ALL=en_US
```

Botspot, 2025-03-01:
Seems related to https://github.com/raspberrypi/bookworm-feedback/issues/123 not ever fixing existing users' installs.

lurch, 2025-03-03:
I've had a bit of a poke-about....
If you use `sudo raspi-config` to change your locale (which just [invokes](https://github.com/RPi-Distro/raspi-config/blob/bookworm/raspi-config#L804) `dpkg-reconfigure locales`), then you **only** get offered locales with UTF-8; but if you use the "Raspberry Pi Configuration" GUI tool to change your locale, then you get separate options for "Language", "Country" and "Character Set".

So if a user happens to choose a non-UTF8 character set, and that doesn't work with your software, then I guess you need to update your documentation to advise users to ensure that they've selected a UTF8 character set first? (As you've seen yourself, the default locale in Raspberry Pi OS _does_ use UTF-8.)

Closing as a user-configuration issue, rather than a problem in Raspberry Pi OS.

closed 2025-03-03 by lurch