raspberrypi/bookworm-feedback #381: DRI3 only works on HDMI output

Silent-Hunter, 2025-03-09:
### Describe the bug

Anything that depends on DRI3 seems to not work unless HDMI and _only_ HDMI is the video output. `vkcube` outputs this:
```
Selected GPU 0: V3D 7.1.7.0, type: IntegratedGpu
vulkan: No DRI3 support detected - required for presentation
Note: you can probably enable DRI3 in your Xorg config
Could not find both graphics and present queues
```

`mpv` outputs this:
```
(+) Video --vid=1 (*) (hevc 1440x1080 23.976fps)
 (+) Audio --aid=1 --alang=eng (*) (ac3 6ch 48000Hz)
 (+) Subs  --sid=1 (*) (subrip)
File tags:
 Title: 101 Dalmatians
[vo/gpu/opengl] Suspected software renderer or indirect context.
libEGL warning: DRI3: Screen seems not DRI3 capable
libEGL warning: DRI3: Screen seems not DRI3 capable
[vo/gpu/opengl] Suspected software renderer or indirect context.
[vo/gpu] VT_GETMODE failed: Inappropriate ioctl for device
[vo/gpu/opengl] Failed to set up VT switcher. Terminal switching will be unavailable.
[vo/gpu] Failed to commit ModeSetting atomic request (-13)
[vo/gpu/opengl] Failed to set CRTC for connector 35: Permission denied
[vo/gpu/libplacebo] Found no memory type matching property flags 0xa and type bits 0x1!
[vo/gpu/libplacebo] No slab to serve request for 2380K bytes (with alignment 0x300) in pool 1!
AO: [pipewire] 48000Hz 5.1(side) 6ch floatp
VO: [gpu] 1440x1080 yuv420p
AV: 00:04:13 / 01:19:16 (5%) A-V:  0.000 Cache: 534s/150MB

Exiting... (Quit)
```

Kodi 21 just displays a black screen on labwc, or a weird flickering version of the main menu on sway.

Updating the eeprom also didn't help, I tried that as a last ditch effort when raspinfo said it was out of date.

### Steps to reproduce the behaviour

Use either DSI, composite, or have anything other than _only_ HDMI plugged in, and then try to use vkcube, mpv, kodi, etc.

### Device (s)

Raspberry Pi 5

### System

Here's the output from raspinfo while DSI-1 and HDMI-A-1 are both connected, and labwc is running: https://pastebin.com/bmM735R0

### Additional context

Here's the post I made on the Raspberry Pi forums: https://forums.raspberrypi.com/viewtopic.php?t=385077

At that time, I had thought it was only multi-monitor support in Sway that was broken, but then I noticed the other problems and added to it.

popcornmix, 2025-03-09:
Transferred to bookworm-feedback as this doesn't sound like a linux kernel issue.
Did you have a recent mesa update? (possibly to mesa 24?) that may be a more likely cause.
@cillian64 any ideas?

Silent-Hunter, 2025-03-10:
Ah thanks, I wasn't sure where exactly to put this, but yes I highly suspect Mesa to be the issue. I hope it's not too difficult to fix.

cillian64, 2025-03-10:
@Silent-Hunter you reported issues with sway as well, was that using sway 1.7-6 from the main apt repo or did you build it from source?

This is probably caused by the Mesa 24 release.  There's a change in Mesa to how it tells applications what device to render on, which caused some issues.  For labwc I implemented a workaround in libwlroots-0.18_0.18.2-3+rpt1, but this won't help with X or with the ancient version of sway in the main repo.

Igalia should be working on a workaround in mesa itself which will hopefully fix everything.  Until then your best bet with sway is to build it from source against our libwlroots-0.18-dev.  You'll need to build release `1.10.1` rather than the master branch because the latter requires new (unreleased) wlroots.

6by9, 2025-03-10:
As just confirmed in your tree, we used to have a patch by me that added the "rp1-*" targets to src/gallium/targets/dri/

That isn't present in Mesa 24, probably as it wouldn't immediately apply due to the rename of src/gallium/targets/dri/target.c to dri_target.c, but otherwise I think it is identical.

@XECDesign / @spl237 do we have a repo managing Mesa these days, or is it a collection of patches in the apt repos?

XECDesign, 2025-03-10:
AFAIK, we only have the source in apt. I believe all the patches have been dropped in 24, except for this one:

```patch
From c34c1ed8d7ad30ffbaa4fda32bbb905ab5dcd848 Mon Sep 17 00:00:00 2001
From: Iago Toral Quiroga <itoral@igalia.com>
Date: Wed, 11 Oct 2023 07:16:49 +0100
Subject: [PATCH] dri: check attached outputs and resolution to decide max back
 buffers for vsync

We had a previous fix for this with:
"dri: Limit the max_num_back to 2 on COMPLETE_MODE_FLIP present mode"
which would limit the maximum to 2 buffers instead of the default of 3,
however, it has been observed that this limits vsync performance in
some scenarios. Particularly, fullscreen through XWayland.

The original patch was intended to reduce CMA pressure specifically
for the case where we have two 4K displays attached, so this change
expands the solution by checking whether this is the case and only
limits the number of back buffers to 2 in that scenario, allowing other
setups to run with 3 back buffers for optimal performance.
---
 src/loader/loader_dri3_helper.c | 28 +++++++++++++++++++++++++++-
 1 file changed, 27 insertions(+), 1 deletion(-)

diff --git a/src/loader/loader_dri3_helper.c b/src/loader/loader_dri3_helper.c
index 32135770e9d..b2206003581 100644
--- a/src/loader/loader_dri3_helper.c
+++ b/src/loader/loader_dri3_helper.c
@@ -267,6 +267,32 @@ dri3_fence_await(xcb_connection_t *c, struct loader_dri3_drawable *draw,
    }
 }
 
+static int
+get_max_num_back_vsync(struct loader_dri3_drawable *draw)
+{
+   int hi_count = 0;
+   xcb_randr_get_screen_resources_cookie_t gsr_c =
+      xcb_randr_get_screen_resources(draw->conn, draw->drawable);
+   xcb_randr_get_screen_resources_reply_t *gsr_r =
+      xcb_randr_get_screen_resources_reply(draw->conn, gsr_c, NULL);
+   if (gsr_r) {
+      xcb_randr_crtc_t *rc = xcb_randr_get_screen_resources_crtcs(gsr_r);
+      for (int c = 0; c < gsr_r->num_crtcs; c++) {
+         xcb_randr_get_crtc_info_cookie_t gci_c =
+            xcb_randr_get_crtc_info(draw->conn, rc[c], gsr_r->config_timestamp);
+         xcb_randr_get_crtc_info_reply_t *gci_r =
+            xcb_randr_get_crtc_info_reply(draw->conn, gci_c, NULL);
+         if (gci_r) {
+            if (gci_r->width * gci_r->height > 1920 * 1200)
+               hi_count++;
+            free(gci_r);
+         }
+      }
+      free(gsr_r);
+   }
+   return hi_count > 1 ? 2 : 3;
+}
+
 static void
 dri3_update_max_num_back(struct loader_dri3_drawable *draw)
 {
@@ -275,7 +301,7 @@ dri3_update_max_num_back(struct loader_dri3_drawable *draw)
       if (draw->swap_interval == 0)
          draw->max_num_back = 4;
       else
-         draw->max_num_back = 3;
+         draw->max_num_back = get_max_num_back_vsync(draw);
 
       assert(draw->max_num_back <= LOADER_DRI3_MAX_BACK);
       break;
-- 
2.39.2

```

Silent-Hunter, 2025-03-10:
@cillian64 Yeah I'm using the version of Sway from the main repo, but I should look into building a newer version. Or I could just wait until the Mesa fix is done and use Labwc in the mean time, if that's fixed now. I use Labwc when I have it hooked up to one of my TVs, and Sway with the monitor at my desk, so I can just use Labwc in both scenarios in the mean time.

6by9, 2025-03-11:
This also hits the rpicam-apps when DSI or DPI outputs are active on Pi5.
https://forums.raspberrypi.com/viewtopic.php?t=385137

I've tried the simple conversion of my previous patch, and it's not giving the hoped immediate fix. I'm slightly surprised labwc starts at all without that, as it used not to.

6by9, 2025-03-11:
FWIW, it aborts at https://gitlab.freedesktop.org/mesa/mesa/-/blob/24.2/src/loader/loader_dri3_helper.c?ref_type=heads#L1297
`extension` is non-null, but `extension->present` is 0.
If running without a DSI display, then `extension->present` is 1.

cillian64, 2025-03-12:
I think I might have realised what this is.  Since the mesa 23->24 upgrade, there's a change around how wlroots decides what device to allocate buffers on.  With mesa24, it specifically tries to allocate buffers on whatever the primary display output device is.  If you're only using HDMI then vc4 is the main display device.  If you're using rp1dsi then that usually becomes the main device because it comes up before the HDMI link.  So wlroots will try to allocate buffers on rp1dsi, whatever that means, which will probably give odd results.  I think we always want to allocate on the vc4 device even if HDMI isn't in use at all.

Using the `WLR_DRM_DEVICES` environment variable you can choose which DRM devices wlroots uses, and the first one specified will be the primary one.  So I think you want `WLR_DRM_DEVICES=/dev/dri/card2:/dev/dri/card1` to make this work (but I'm not 100% sure that's the right way round).

lurch, 2025-03-12:
IIRC the `/dev/dri/cardX` numbers get moved about depending on the order they're detected and/or depending what display devices you've got plugged in? `/dev/dri/by-path/*` looks much more useful?

6by9, 2025-03-12:
Starting labwc with `WLR_DRM_DEVICES=/dev/dri/card2:/dev/dri/card1 labwc-pi` makes vkcube and rpicam-hello work again.

Trying to use `WLR_DRM_DEVICES=/dev/dri/by-path/platform-axi\:gpu-card:/dev/dri/by-path/platform-1f00118000.dsi-card labwc-pi` only results in the DSI display working, and rpicam-hello and vkcube fail in the same way as before.

6by9, 2025-03-12:
It does look like there is something missing for the RP1 peripherals though. Testing with an [HX8357D SPI display](https://thepihut.com/products/adafruit-pitft-assembled-480x320-3-5-tft-touchscreen-for-raspberry-pi) that works fine with labwc regardless of whether vc4 is loaded or not, nor the order that the cards are in.

I've just realised that I've nuked my mesa build so will have to recreate the changes I had made.

cillian64, 2025-03-12:
Just to check, you're running with wlroots `0.18.2-3+rpt1` or `0.18.2-3+rpt2`, right? If older than that then https://github.com/raspberrypi/bookworm-feedback/issues/381#issuecomment-2718136497 doesn't apply and things are wacky in different ways on mesa 24

6by9, 2025-03-12:
No, I hadn't done an `apt full-upgrade` as I thought just the mesa libraries were going to be sufficient.
Now doing a full-upgrade.

Silent-Hunter, 2025-03-12:
> Using the `WLR_DRM_DEVICES` environment variable you can choose which DRM devices wlroots uses, and the first one specified will be the primary one. So I think you want `WLR_DRM_DEVICES=/dev/dri/card2:/dev/dri/card1` to make this work (but I'm not 100% sure that's the right way round).

This does fix the issue partially! It makes the DSI display not work, but I can at least leave it plugged in now when using HDMI.

Silent-Hunter, 2025-03-12:
> Using the `WLR_DRM_DEVICES` environment variable you can choose which DRM devices wlroots uses, and the first one specified will be the primary one. So I think you want `WLR_DRM_DEVICES=/dev/dri/card2:/dev/dri/card1` to make this work (but I'm not 100% sure that's the right way round).

This works, partially. If I specify `WLR_DRM_DEVICES=/dev/dri/card2:/dev/dri/card1` or whichever card is the vc4 and then rp1, it doesn't work, but if I omit the rp1 path, the HDMI works.

6by9, 2025-03-13:
After far more faffing with the Debian build system than I expected, this patch appears to resolve the issue.

[0001-dri-Add-RP1-kmsro-devices-to-the-magic-table.patch](https://github.com/user-attachments/files/19227824/0001-dri-Add-RP1-kmsro-devices-to-the-magic-table.patch)

@XECDesign can this be added to builds for testing?

6by9, 2025-03-18:
Patch should now be in the public repos. Please update and retest.

edro123, 2025-03-18:
I was hoping this would fix my issue, but I still get an error. I'm running on a Raspberry Pi 5, with and up to date bookworm and this fix installed this afternoon. My display is non HDMI (I think it's DPI:?). I have this line in my config.txt: dtoverlay=vc4-kms-dpi-hyperpixel4sq

Here's the output I get when I run the imx500_object_detection.py example from the pycamera2 GitHub:

[0:07:15.765026345] [7425]  INFO Camera camera_manager.cpp:327 libcamera v0.4.0+53-29156679
[0:07:15.774467609] [7433]  INFO RPI pisp.cpp:720 libpisp version v1.1.0 e7974a156008 27-01-2025 (21:50:51)
[0:07:15.787880615] [7433]  INFO RPI pisp.cpp:1179 Registered camera /base/axi/pcie@120000/rp1/i2c@80000/imx500@1a to CFE device /dev/media0 and ISP device /dev/media2 using PiSP variant BCM2712_C0

------------------------------------------------------------------------------------------------------------------
NOTE: Loading network firmware onto the IMX500 can take several minutes, please do not close down the application.
------------------------------------------------------------------------------------------------------------------

[0:07:15.822529650] [7425]  INFO Camera camera.cpp:1202 configuring streams: (0) 640x480-XBGR8888 (1) 2028x1520-RGGB_PISP_COMP1
[0:07:15.822794631] [7433]  INFO RPI pisp.cpp:1484 Sensor: /base/axi/pcie@120000/rp1/i2c@80000/imx500@1a - Selected sensor format: 2028x1520-SRGGB10_1X10 - Selected CFE format: 2028x1520-PC1R
Network Firmware Upload: 0.00bytes [00:00, ?bytes/s]QStandardPaths: wrong permissions on runtime directory /run/user/1000, 0770 instead of 0700
libEGL warning: DRI3: Screen seems not DRI3 capable
libEGL warning: DRI3: Screen seems not DRI3 capable
MESA: error: ZINK: vkCreateInstance failed (VK_ERROR_INCOMPATIBLE_DRIVER)
libEGL warning: egl: failed to create dri2 screen
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/picamera2/previews/qt_previews.py", line 36, in createpreview
    qpicamera2 = parent.make_picamera2_widget(cam, width=parent.width, height=parent.height,
                 ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/picamera2/previews/qt_previews.py", line 141, in make_picamera2_widget
    return QGlPicamera2(picam2, width=self.width, height=self.height, transform=self.transform, preview_window=self)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/picamera2/previews/q_gl_picamera2.py", line 97, in __init__
    self.egl = EglState()
               ^^^^^^^^^^
  File "/usr/lib/python3/dist-packages/picamera2/previews/q_gl_picamera2.py", line 27, in __init__
    self.choose_config()
  File "/usr/lib/python3/dist-packages/picamera2/previews/q_gl_picamera2.py", line 45, in choose_config
    check_egl_extensions(self.display, ["EGL_EXT_image_dma_buf_import"])
  File "/usr/lib/python3/dist-packages/picamera2/previews/gl_helpers.py", line 71, in check_egl_extensions
    raise Exception(ext + " missing")
Exception: EGL_EXT_image_dma_buf_import missing

Process ended with exit code -6.

Silent-Hunter, 2025-03-18:
I haven't tested with X11 yet, but as far as labwc is concerned, this has fixed everything; thanks so much! Hopefully edro123's issue isn't too hard to fix.

6by9, 2025-03-18:
There does appear to be an issue with picamera2 rendering through GL with the latest Mesa, but it is NOT related to this issue. I see the same thing with just HDMI output.
I also don't get the `libEGL warning: DRI3: Screen seems not DRI3 capable` error message any more, which is what this issue is about. I therefore consider it a different issue.

> I haven't tested with X11 yet, but as far as labwc is concerned, this has fixed everything; thanks so much!

Thank you for reporting and testing. I'm going to close this down then.

closed 2025-03-18 by 6by9

edro123, 2025-03-18:
my output from  apt policy libgbm1 is:

libgbm1:
  Installed: 24.2.8-1~bpo12+rpt2
  Candidate: 24.2.8-1~bpo12+rpt2
  Version table:
 *** 24.2.8-1~bpo12+rpt2 500
        500 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages
        100 /var/lib/dpkg/status
     22.3.6-1+deb12u1 500
        500 http://deb.debian.org/debian bookworm/main arm64 Packages

edro123, 2025-03-18:
Hmm. After rebooting, I no longer see the issue.