raspberrypi/bookworm-feedback #382: Raspberry Pi OS latest update breaks minecraft-bedrock with libEGL_mesa.so.0: undefined symbol error

JustinS-B, 2025-03-10:
minecraft-bedrock unable to launch after Raspberry Pi OS update

Updated Rpi 5 to latest version using updater GUI, now Minecraft Bedrock won't launch. Running from CLI gives the following errors

```
pi@Pi5:~ $ GALLIUM_HUD=simple,fps ~/.local/bin/MCBedrock.AppImage
qt.core.qobject.connect: QObject::connect(QObject, Unknown): invalid nullptr parameter
qt.core.qobject.connect: QObject::connect(QObject, Unknown): invalid nullptr parameter
/tmp/.mount_MCBedrOgmKQN/usr/bin/msa-ui-qt: symbol lookup error: /lib/aarch64-linux-gnu/libEGL_mesa.so.0: undefined symbol: wl_display_create_queue_with_name
/tmp/.mount_MCBedrOgmKQN/usr/bin/mcpelauncher-ui-qt: symbol lookup error: /lib/aarch64-linux-gnu/libEGL_mesa.so.0: undefined symbol: wl_display_create_queue_with_name
/tmp/.mount_MCBedrOgmKQN/AppRun: line 15: kill: (2341) - No such process
```

cillian64, 2025-03-10:
Ugh, annoying that Mesa keep breaking things with upgrades.  But yes, the solution here is for the app to update

closed 2025-03-10 by lurch

XECDesign, 2025-03-10:
In this case, I believe the problem isn't Mesa itself, but how the AppImage was created.

https://gitlab.freedesktop.org/mesa/mesa/-/issues/11316#note_2472309

XECDesign, 2025-03-10:
> Ahhh, so perhaps similar to the problems that [@tdewey-rpi](https://github.com/tdewey-rpi) ran into with the AppImage version of Raspberry Pi Imager then?

Yes, it's the same issue. The second last version shipped with a fix and worked with both Mesa versions. Then in the last version of rpi-imager, wayland support was removed anyway, but I'm not sure why.

lurch, 2025-03-10:
> Then in the last version of rpi-imager, wayland support was removed anyway, but I'm not sure why.

See https://github.com/raspberrypi/rpi-imager/issues/1019#issuecomment-2707780324

tdewey-rpi, 2025-03-10:
To be clear: I removed Wayland support wholesale because this was the only way to convince the AppImage tooling to not package the libwayland-* libraries. Even using the library exclusion functions didn't do the right thing - because the exclusions are seemingly not being propagated to the LinuxDeploy plugins

XECDesign, 2025-03-10:
Ah, a tool to solve dependency hell issues has issues with dependencies. Thought it might be something like that. 🙁