raspberrypi/bookworm-feedback #383: overlayfs/overlayroot broken on Raspberry Pi 5

sam-vdp, 2025-03-11:
I am trying to setup overlayfs using the current 2024-11-19-raspios-bookworm-arm64-lite image and I believe the process is broken.

Using raspi-config to enable the overlay filesystem performs two steps:

* First it installs the `overlayroot` package:  
`is_installed overlayroot || apt-get install -y overlayroot`
(quoted from the source)
  * This in turn modifies the `initramfs` to include the overlayroot scripts:  
`update-initramfs: Generating /boot/initrd.img-6.6.51+rpt-rpi-v8`     
`'/boot/initrd.img-6.6.51+rpt-rpi-v8' -> '/boot/firmware/initramfs8'`  
(quoted from `apt-get install overlayroot`
* Then raspi-config modifies `/boot/firmware/cmdline.txt` to include `overlayroot=tmpfs`, which actually activates the overlayroot module

This works fine on Raspberry Pi 4 hardware, but there's a catch on Raspberry Pi 5:
As can be seen in the log above, overlayroot regenerates the `initramfs8`. However, on a Raspberry Pi 5, the system boots `kernel_2712.img` and thus also uses `initramfs_2712`, which does not contain the overlay module.

This suspicion is confirmed if I modify `/boot/firmware/config.txt` and add  
`kernel=kernel8.img`  
`initramfs initramfs8`  
then overlayfs works as expected.

I believe this is a bug?

lurch, 2025-03-11:
ping @XECDesign

XECDesign, 2025-03-11:
Was the system updated first? There have been some fixes which I believe should've resolve this already.

closed 2025-03-11 by XECDesign