raspberrypi/bookworm-feedback #384: Mouse Motion Speed won't accept changes

mcunkelman, 2025-03-11:
I open preferences, mouse and keyboard, mouse and change the scroll speed to something other than the default 5.0, hit OK. If I reopen the some pane it is back to 5.0. No changes I make to scroll speed stick. I can change double-click speed and it works as expected. I have updated my system as of this morning.

lurch, 2025-03-11:
I've just checked on fresh installs of both 32-bit and 64-bit PiOS, and both before and after installing the updates, the mouse cursor speed setting gets saved correctly. (note that this affects mouse-cursor movement speed and _not_ scroll speed)

Could you please open a Terminal window and type `cat .config/labwc/rc.xml`, then change both the "Speed" and "Double-click" settings in "Mouse and Keyboard Settings" and then `cat .config/labwc/rc.xml` again, and paste the before and after versions here? Thanks.
Also, what does `apt policy rasputin` and `ls -l .config/labwc/rc.xml` say?

mcunkelman, 2025-03-11:
I did decline to upgrade to 'labwc' as I am using RDP on occasion and din't want to invite anything new to troubleshoot through.  When I run the command you gave it says no such file or directory

lurch, 2025-03-11:
> I did decline to upgrade to 'labwc'

Ahh, that would have been useful information! 😆  Can you please try the tests I mentioned above, but with `.config/wayfire.ini` instead?

mcunkelman, 2025-03-11:
Sorry, my bad....

Before:
`user@machine:~ $ cat .config/wayfire.ini 
[command]
repeatable_binding_volume_up = KEY_VOLUMEUP
command_volume_up = wfpanelctl volumepulse volu
repeatable_binding_volume_down = KEY_VOLUMEDOWN
command_volume_down = wfpanelctl volumepulse vold
binding_mute = KEY_MUTE
command_mute = wfpanelctl volumepulse mute
binding_menu = <super>
command_menu = wfpanelctl smenu menu
binding_terminal = <ctrl> <alt> KEY_T
command_terminal = lxterminal
binding_bluetooth = <ctrl> <alt> KEY_B
command_bluetooth = wfpanelctl bluetooth menu
binding_netman = <ctrl> <alt> KEY_W
command_netman = wfpanelctl netman menu
binding_grim = KEY_SYSRQ
command_grim = grim
binding_orca = <ctrl> <alt> KEY_SPACE
command_orca = gui-pkinst orca reboot
binding_quit = <ctrl> <alt> KEY_DELETE
command_quit = lxde-pi-shutdown-helper
binding_power = KEY_POWER
command_power = pwrkey

[input-device:10-0038 generic ft5x06 (79)]
output = DSI-1

[input-device:6-0038 generic ft5x06 (79)]
output = DSI-1

[input-device:4-0038 generic ft5x06 (79)]
output = DSI-2

[input-device:10-0038 generic ft5x06 (00)]
output = DSI-1

[input-device:6-0038 generic ft5x06 (00)]
output = DSI-1

[input-device:4-0038 generic ft5x06 (00)]
output = DSI-2

[input-device:10-005d Goodix Capacitive TouchScreen]
output = DSI-1

[input-device:6-005d Goodix Capacitive TouchScreen]
output = DSI-1

[input-device:4-005d Goodix Capacitive TouchScreen]
output = DSI-2

[input]
xkb_options=
xkb_model=macintosh
xkb_layout=us
xkb_variant=

[output]
headless_width = 1920
headless_height = 1080
`
Changed Speed from 5.0 to 1.5 and Double-Click from 438 to 1355. 

After:
`user@machine:~ $ cat .config/wayfire.ini 
[command]
repeatable_binding_volume_up = KEY_VOLUMEUP
command_volume_up = wfpanelctl volumepulse volu
repeatable_binding_volume_down = KEY_VOLUMEDOWN
command_volume_down = wfpanelctl volumepulse vold
binding_mute = KEY_MUTE
command_mute = wfpanelctl volumepulse mute
binding_menu = <super>
command_menu = wfpanelctl smenu menu
binding_terminal = <ctrl> <alt> KEY_T
command_terminal = lxterminal
binding_bluetooth = <ctrl> <alt> KEY_B
command_bluetooth = wfpanelctl bluetooth menu
binding_netman = <ctrl> <alt> KEY_W
command_netman = wfpanelctl netman menu
binding_grim = KEY_SYSRQ
command_grim = grim
binding_orca = <ctrl> <alt> KEY_SPACE
command_orca = gui-pkinst orca reboot
binding_quit = <ctrl> <alt> KEY_DELETE
command_quit = lxde-pi-shutdown-helper
binding_power = KEY_POWER
command_power = pwrkey

[input-device:10-0038 generic ft5x06 (79)]
output = DSI-1

[input-device:6-0038 generic ft5x06 (79)]
output = DSI-1

[input-device:4-0038 generic ft5x06 (79)]
output = DSI-2

[input-device:10-0038 generic ft5x06 (00)]
output = DSI-1

[input-device:6-0038 generic ft5x06 (00)]
output = DSI-1

[input-device:4-0038 generic ft5x06 (00)]
output = DSI-2

[input-device:10-005d Goodix Capacitive TouchScreen]
output = DSI-1

[input-device:6-005d Goodix Capacitive TouchScreen]
output = DSI-1

[input-device:4-005d Goodix Capacitive TouchScreen]
output = DSI-2

[input]
xkb_options=
xkb_model=macintosh
xkb_layout=us
xkb_variant=

[output]
headless_width = 1920
headless_height = 1080
`
Additionally:
`user@machine:~ $ apt policy rasputin
rasputin:
  Installed: 0.2
  Candidate: 0.2
  Version table:
 *** 0.2 500
        500 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages
        100 /var/lib/dpkg/status
`

`user@machine:~ $ ls -l .config/wayfire.ini 
-rw-r--r-- 1 user user 1483 Mar 11 10:47 .config/wayfire.ini
`

I replaced my username with 'user' and device name with 'machine' in above outputs.

6by9, 2025-03-11:
> I don't know why the markdown got messed up in my above response

You need 3 backticks before and after blocks you want tagged as code.
Single backticks only works for a single line
https://docs.github.com/en/get-started/writing-on-github/getting-started-with-writing-and-formatting-on-github/basic-writing-and-formatting-syntax#quoting-code

lurch, 2025-03-11:
Weird. I also have `rasputin` version 0.2 installed, and after using `raspi-config` to switch to Wayfire, if I edit the mouse and keyboard settings I get some extra options in the `[input]` section of `.config/wayfire.ini` that aren't present in yours: `mouse_cursor_speed`, `kb_repeat_delay`, `kb_repeat_rate` and `left_handed_mode`.

If you add `mouse_cursor_speed=0.6` to the `[input]` section in your `.config/wayfire.ini`, does "Mouse and Keyboard Settings" then show the Speed as 8.0? And if you change the slider and click OK, does the value in `.config/wayfire.ini` change?

mcunkelman, 2025-03-11:
The value in the UI still continues to reset to 5.0 even with the added line in the .ini

lurch, 2025-03-11:
ping @spl237

spl237, 2025-03-11:
You could try running rasputin at the command line and see if any errors or warnings are generated when you change the setting. 

Also, are you completely sure you are running wayfire and not X? If you are not seeing any changes saved and the value is not being read from wayfire.ini, I suspect your system is not being identified as wayfire by rasputin. What is the output of "export" when you type it in a terminal?

mcunkelman, 2025-03-11:
export yields:
```declare -x COLORTERM="truecolor"
declare -x DBUS_SESSION_BUS_ADDRESS="unix:path=/run/user/1000/bus"
declare -x DESKTOP_SESSION="LXDE-pi"
declare -x DISPLAY=":10.0"
declare -x GPG_AGENT_INFO="/run/user/1000/gnupg/S.gpg-agent:0:1"
declare -x HOME="/home/mcunkelman"
declare -x LANG="en_GB.UTF-8"
declare -x LOGNAME="mcunkelman"
declare -x LS_COLORS="rs=0:di=01;34:ln=01;36:mh=00:pi=40;33:so=01;35:do=01;35:bd=40;33;01:cd=40;33;01:or=40;31;01:mi=00:su=37;41:sg=30;43:ca=00:tw=30;42:ow=34;42:st=37;44:ex=01;32:*.tar=01;31:*.tgz=01;31:*.arc=01;31:*.arj=01;31:*.taz=01;31:*.lha=01;31:*.lz4=01;31:*.lzh=01;31:*.lzma=01;31:*.tlz=01;31:*.txz=01;31:*.tzo=01;31:*.t7z=01;31:*.zip=01;31:*.z=01;31:*.dz=01;31:*.gz=01;31:*.lrz=01;31:*.lz=01;31:*.lzo=01;31:*.xz=01;31:*.zst=01;31:*.tzst=01;31:*.bz2=01;31:*.bz=01;31:*.tbz=01;31:*.tbz2=01;31:*.tz=01;31:*.deb=01;31:*.rpm=01;31:*.jar=01;31:*.war=01;31:*.ear=01;31:*.sar=01;31:*.rar=01;31:*.alz=01;31:*.ace=01;31:*.zoo=01;31:*.cpio=01;31:*.7z=01;31:*.rz=01;31:*.cab=01;31:*.wim=01;31:*.swm=01;31:*.dwm=01;31:*.esd=01;31:*.avif=01;35:*.jpg=01;35:*.jpeg=01;35:*.mjpg=01;35:*.mjpeg=01;35:*.gif=01;35:*.bmp=01;35:*.pbm=01;35:*.pgm=01;35:*.ppm=01;35:*.tga=01;35:*.xbm=01;35:*.xpm=01;35:*.tif=01;35:*.tiff=01;35:*.png=01;35:*.svg=01;35:*.svgz=01;35:*.mng=01;35:*.pcx=01;35:*.mov=01;35:*.mpg=01;35:*.mpeg=01;35:*.m2v=01;35:*.mkv=01;35:*.webm=01;35:*.webp=01;35:*.ogm=01;35:*.mp4=01;35:*.m4v=01;35:*.mp4v=01;35:*.vob=01;35:*.qt=01;35:*.nuv=01;35:*.wmv=01;35:*.asf=01;35:*.rm=01;35:*.rmvb=01;35:*.flc=01;35:*.avi=01;35:*.fli=01;35:*.flv=01;35:*.gl=01;35:*.dl=01;35:*.xcf=01;35:*.xwd=01;35:*.yuv=01;35:*.cgm=01;35:*.emf=01;35:*.ogv=01;35:*.ogx=01;35:*.aac=00;36:*.au=00;36:*.flac=00;36:*.m4a=00;36:*.mid=00;36:*.midi=00;36:*.mka=00;36:*.mp3=00;36:*.mpc=00;36:*.ogg=00;36:*.ra=00;36:*.wav=00;36:*.oga=00;36:*.opus=00;36:*.spx=00;36:*.xspf=00;36:*~=00;90:*#=00;90:*.bak=00;90:*.old=00;90:*.orig=00;90:*.part=00;90:*.rej=00;90:*.swp=00;90:*.tmp=00;90:*.dpkg-dist=00;90:*.dpkg-old=00;90:*.ucf-dist=00;90:*.ucf-new=00;90:*.ucf-old=00;90:*.rpmnew=00;90:*.rpmorig=00;90:*.rpmsave=00;90:"
declare -x NO_AT_BRIDGE="1"
declare -x OLDPWD
declare -x PATH="/home/mcunkelman/bin:/home/mcunkelman/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/local/games:/usr/games"
declare -x PULSE_SCRIPT="/etc/xrdp/pulse/default.pa"
declare -x PWD="/home/mcunkelman"
declare -x QT_QPA_PLATFORMTHEME="qt5ct"
declare -x SAL_USE_VCLPLUGIN="gtk3"
declare -x SHELL="/bin/bash"
declare -x SHLVL="1"
declare -x SSH_AGENT_PID="1962"
declare -x SSH_AUTH_SOCK="/tmp/ssh-7gOFouUffDAl/agent.1882"
declare -x TERM="xterm-256color"
declare -x TEXTDOMAIN="Linux-PAM"
declare -x UID="1000"
declare -x USER="mcunkelman"
declare -x VTE_VERSION="7006"
declare -x XDG_CONFIG_DIRS="/etc/xdg"
declare -x XDG_CONFIG_HOME="/home/mcunkelman/.config"
declare -x XDG_CURRENT_DESKTOP="LXDE"
declare -x XDG_DATA_DIRS="~/.local/share:/usr/local/share:/usr/share/raspi-ui-overrides:/usr/share:/usr/share/gdm:/var/lib/menu-xdg"
declare -x XDG_MENU_PREFIX="lxde-pi-"
declare -x XDG_RUNTIME_DIR="/run/user/1000"
declare -x XDG_SESSION_CLASS="user"
declare -x XDG_SESSION_ID="c1"
declare -x XDG_SESSION_TYPE="x11"
declare -x XRDP_PULSE_SINK_SOCKET="xrdp_chansrv_audio_out_socket_10"
declare -x XRDP_PULSE_SOURCE_SOCKET="xrdp_chansrv_audio_in_socket_10"
declare -x XRDP_SESSION="1"
declare -x XRDP_SOCKET_PATH="/run/xrdp/sockdir"
declare -x _LXSESSION_PID="1882"
```

Running rasputin bought up the Mouse and Keyboard UI, still set to 5.0, I changed it and hit ok, then ran rasputin again and it was back to 5.0.

spl237, 2025-03-11:
OK, you're not running wayfire - you're running X - there are no Wayland environment variables set and your session type is set to X11.

Mouse speed under X is set directly by accessing the xinput pointer device, so I suspect there is something odd about your mouse, possibly due to use via XRDP.

What is the output of "xinput list"?

lurch, 2025-03-11:
If you're running over xrdp, then isn't it the mouse speed of your _local_ computer that you want to edit, rather than the mouse-speed on the remote Raspberry Pi?

spl237, 2025-03-11:
As I suspected - you have no physical input devices; only a virtual mouse provided by XRDP, and I suspect it does not have a speed that can be set.

To confirm, please post the output of "xinput list-props 4" and "xinput list-props 6"

mcunkelman, 2025-03-11:
```xinput list-props 4
Device 'Virtual core XTEST pointer':
	Device Enabled (119):	1
	Coordinate Transformation Matrix (120):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	XTEST Device (232):	1
mcunkelman@rpidev:~ $ xinput list-props 6
Device 'xrdpMouse':
	Device Enabled (119):	1
	Coordinate Transformation Matrix (120):	1.000000, 0.000000, 0.000000, 0.000000, 1.000000, 0.000000, 0.000000, 0.000000, 1.000000
	Device Accel Profile (236):	0
	Device Accel Constant Deceleration (237):	1.000000
	Device Accel Adaptive Deceleration (238):	1.000000
	Device Accel Velocity Scaling (239):	10.000000
```

spl237, 2025-03-11:
Yes, neither of those have the relevant "Accel Speed" parameter, so their speed cannot be set by the user.

closed 2025-03-11 by spl237