raspberrypi/bookworm-feedback #385: GPU plug-in on task bar of raspi os stays at 0% even if gpu is working regularly.

wobey83, 2025-03-11:
Using a Raspberry Pi 5 with the lastest (stable) 64-bit raspi os. 
uname -a gives "Linux 6.6.74-rpt-rpi-2712 Debian ... aarch64 GNU/Linux"
Updates by apt-get are executed on a daily base. 
Incremental backups are generated weekly (rsync -aAXHv).
On 11th of March 2025 I realized that the GPU plug-in 
at the taskbar of the graphical user interface remains at
0% even if software is using the gpu (chromium/youtube/video).

After some trials I restored the backup of 23rd of February 2025.
Now the GPU plug-in showed the right values for the GPU usage.

After that, I updated the OS with "sudo apt-get update" and 
"sudo apt-get upgrade". With the updated OS the GPU plug-in of
the tastbar of the graphical user interface stuck again at 0%.

I assume a failure in the OS that happened somewhen between
23rd of Feb. and 11th of March. I suppose this failure happened
already 3 or 4 days earlier, but I have no exact date.

The OS update on 11th of March still remains with this failure.

lurch, 2025-03-11:
Hmmm, I've just tested on a fresh install of RPiOS 64-bit on a Pi 5, and after installing all the apt-updates, the GPU taskbar plug-in still works for me! `uname -a` also shows that I'm running kernel 6.6.74+rpt-rpi-2712.

If you run the Accessories -> Task Manager program, does that show GPU usage? What does `apt policy wfplug-gpu` say? Are you still running the default Labwc desktop?

wobey83, 2025-03-11:
One of the following install packages seem to create this behaviour:
libegl-mesa0, libgbm1, libgl1-mesa-dev, libgl1-mesa-dri, libglapi-mesa,
libglx-mesa0, libwlroots-0.18, mesa-vdpau-drivers or mesa-vulkan-drivers.
Tommorrow I will try to isolate which one is responsible. Needs some time
to install packages one by one and reboot and check gpu. Thank you for your
immediate reply.

lurch, 2025-03-12:
As I mentioned yesterday, on a freshly-installed and fully-apt-updated version of 64-bit Raspberry Pi OS running on a Raspberry Pi 5, the GPU usage plugin is still working for me. But just for the avoidance of doubt:
```console
$ apt policy libegl-mesa0 libgbm1 libgl1-mesa-dev libgl1-mesa-dri libglapi-mesa libglx-mesa0 libwlroots-0.18 mesa-vdpau-drivers mesa-vulkan-drivers wfplug-gpu
libegl-mesa0:
  Installed: 24.2.8-1~bpo12+rpt1
  Candidate: 24.2.8-1~bpo12+rpt1
  Version table:
 *** 24.2.8-1~bpo12+rpt1 500
        500 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages
        100 /var/lib/dpkg/status
     22.3.6-1+deb12u1 500
        500 http://deb.debian.org/debian bookworm/main arm64 Packages
libgbm1:
  Installed: 24.2.8-1~bpo12+rpt1
  Candidate: 24.2.8-1~bpo12+rpt1
  Version table:
 *** 24.2.8-1~bpo12+rpt1 500
        500 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages
        100 /var/lib/dpkg/status
     22.3.6-1+deb12u1 500
        500 http://deb.debian.org/debian bookworm/main arm64 Packages
libgl1-mesa-dev:
  Installed: 24.2.8-1~bpo12+rpt1
  Candidate: 24.2.8-1~bpo12+rpt1
  Version table:
 *** 24.2.8-1~bpo12+rpt1 500
        500 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages
        100 /var/lib/dpkg/status
     22.3.6-1+deb12u1 500
        500 http://deb.debian.org/debian bookworm/main arm64 Packages
libgl1-mesa-dri:
  Installed: 24.2.8-1~bpo12+rpt1
  Candidate: 24.2.8-1~bpo12+rpt1
  Version table:
 *** 24.2.8-1~bpo12+rpt1 500
        500 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages
        100 /var/lib/dpkg/status
     22.3.6-1+deb12u1 500
        500 http://deb.debian.org/debian bookworm/main arm64 Packages
libglapi-mesa:
  Installed: 24.2.8-1~bpo12+rpt1
  Candidate: 24.2.8-1~bpo12+rpt1
  Version table:
 *** 24.2.8-1~bpo12+rpt1 500
        500 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages
        100 /var/lib/dpkg/status
     22.3.6-1+deb12u1 500
        500 http://deb.debian.org/debian bookworm/main arm64 Packages
libglx-mesa0:
  Installed: 24.2.8-1~bpo12+rpt1
  Candidate: 24.2.8-1~bpo12+rpt1
  Version table:
 *** 24.2.8-1~bpo12+rpt1 500
        500 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages
        100 /var/lib/dpkg/status
     22.3.6-1+deb12u1 500
        500 http://deb.debian.org/debian bookworm/main arm64 Packages
libwlroots-0.18:
  Installed: 0.18.2-3+rpt2
  Candidate: 0.18.2-3+rpt2
  Version table:
 *** 0.18.2-3+rpt2 500
        500 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages
        100 /var/lib/dpkg/status
mesa-vdpau-drivers:
  Installed: 24.2.8-1~bpo12+rpt1
  Candidate: 24.2.8-1~bpo12+rpt1
  Version table:
 *** 24.2.8-1~bpo12+rpt1 500
        500 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages
        100 /var/lib/dpkg/status
     22.3.6-1+deb12u1 500
        500 http://deb.debian.org/debian bookworm/main arm64 Packages
mesa-vulkan-drivers:
  Installed: 24.2.8-1~bpo12+rpt1
  Candidate: 24.2.8-1~bpo12+rpt1
  Version table:
 *** 24.2.8-1~bpo12+rpt1 500
        500 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages
        100 /var/lib/dpkg/status
     22.3.6-1+deb12u1 500
        500 http://deb.debian.org/debian bookworm/main arm64 Packages
wfplug-gpu:
  Installed: 0.2
  Candidate: 0.2
  Version table:
 *** 0.2 500
        500 http://archive.raspberrypi.com/debian bookworm/main arm64 Packages
        100 /var/lib/dpkg/status
```