raspberrypi/bookworm-feedback #386: WiFi menu unaware of 6G WiFi connections

geerlingguy, 2025-03-11:
I've been testing WiFi 6E and WiFi 7 on the CM4, CM5, and Pi 5 lately, and I noticed that while 6 GHz connections (for both WiFi 6E and WiFi 7) work great, and provide much greater speeds than equivalent protocols (BE/AX) on 5 GHz bands, the Pi menu only shows '5G' even though the connection is '6G':

[image]

Confirming the connection details with `iw`:

[image]

I presume this is a UI bug that should be fixed—but may need to also wait on a NetworkManager update to 1.46 or later (Pi OS currently ships with 1.42.3), as that version added full support for displaying 6 GHz band information. (Though I'm not sure what underlying libraries the UI widget uses...).

lurch, 2025-03-11:
> (Though I'm not sure what underlying libraries the UI widget uses...).

I'll double-check tomorrow, but it's probably this? https://github.com/raspberrypi-ui/pplug-netman

geerlingguy, 2025-03-11:
@lurch - Oh cool! It looks like the code can be updated for that menu with _relative_ ease, as it's using `iw` already, and not `nmcli`, to get the channel/frequency info... https://github.com/search?q=repo%3Araspberrypi-ui%2Fpplug-netman%205G&type=code — see:

https://github.com/raspberrypi-ui/pplug-netman/blob/2f9c034063bdfd45fdd7e37c6aa68b01894222ff/src/netman.c#L63

Hmm... but then the actual panel menulet gets it from nm I think?

https://github.com/raspberrypi-ui/pplug-netman/blob/2f9c034063bdfd45fdd7e37c6aa68b01894222ff/src/ap-menu-item.c#L314

lurch, 2025-03-12:
ping @spl237 because he's the maintainer of that code 😉

spl237, 2025-04-09:
I've asked for a new icon to represent 6G networks; once that is ready, we'll update the plugin to do this - it's only the lack of the icon which means we can't do it yet!