raspberrypi/bookworm-feedback #387: Xorg very busy probing EDID at each new frame after upgrade to Mesa 24

sebastiendu, 2025-03-13:
A few of our Pi 4 run an OpenGL3/GLFW program on Xorg.
The following mesa packages are installed:

libegl-mesa0
libgbm1
libgl1-mesa-dri
libglapi-mesa
libglx-mesa0

After upgrading these packages from version
`23.2.1-1~bpo12+rpt3`
to version
`24.2.8-1~bpo12+rpt1`,
Xorg takes about 100 % of one CPU and fills its log file with an EDID probe for every single new frame, as long as the GUI runs.

Eventually /var/log/Xorg.0.log fills the SD card partition, therefore further trouble can be triggered by this issue.

Downgrading the packages to version `23.2.1-1~bpo12+rpt3` fixes this.

Here is the endless and fast repeating pattern found in /var/log/Xorg.0.log :
```
[  5779.009] (II) modeset(0): EDID vendor "ROW", prod id 0
[  5779.009] (II) modeset(0): Using hsync ranges from config file
[  5779.009] (II) modeset(0): Using vrefresh ranges from config file
[  5779.009] (II) modeset(0): Printing DDC gathered Modelines:
[  5779.009] (II) modeset(0): Modeline "1024x768"x0.0   65.00  1024 1048 1184 1344  768 771 777 806 -hsync -vsync (48.4 kHz eP)
[  5779.009] (II) modeset(0): Modeline "1280x720"x0.0   74.25  1280 1720 1760 1980  720 725 730 750 +hsync +vsync (37.5 kHz e)
[  5779.009] (II) modeset(0): Modeline "1920x1080i"x0.0   74.25  1920 2448 2492 2640  1080 1084 1094 1125 interlace +hsync +vsync (28.1 kHz e)
[  5779.009] (II) modeset(0): Modeline "720x480"x0.0   27.00  720 736 798 858  480 489 495 525 -hsync -vsync (31.5 kHz e)
[  5779.009] (II) modeset(0): Modeline "720x576"x0.0   27.00  720 732 796 864  576 581 586 625 -hsync -vsync (31.2 kHz e)
[  5779.009] (II) modeset(0): Modeline "1920x1080"x0.0  148.50  1920 2008 2052 2200  1080 1084 1089 1125 +hsync +vsync (67.5 kHz e)
[  5779.009] (II) modeset(0): Modeline "1920x1080"x0.0  148.50  1920 2448 2492 2640  1080 1084 1089 1125 +hsync +vsync (56.2 kHz e)
[  5779.009] (II) modeset(0): Modeline "1920x1080i"x0.0   74.25  1920 2008 2052 2200  1080 1084 1094 1125 interlace +hsync +vsync (33.8 kHz e)
[  5779.009] (II) modeset(0): Modeline "1280x720"x0.0   74.25  1280 1390 1430 1650  720 725 730 750 +hsync +vsync (45.0 kHz e)
[  5779.009] (II) modeset(0): Modeline "1440x576i"x0.0   27.00  1440 1464 1590 1728  576 580 586 625 interlace -hsync -vsync (15.6 kHz e)
[  5779.009] (II) modeset(0): Modeline "1440x480i"x0.0   27.00  1440 1478 1602 1716  480 488 494 525 interlace -hsync -vsync (15.7 kHz e)
[  5779.009] (II) modeset(0): Modeline "640x480"x0.0   25.18  640 656 752 800  480 490 492 525 -hsync -vsync (31.5 kHz e)
[  5779.009] (II) modeset(0): Modeline "800x600"x0.0   40.00  800 840 968 1056  600 601 605 628 +hsync +vsync (37.9 kHz e)
[  5779.009] (II) modeset(0): Modeline "800x600"x0.0   36.00  800 824 896 1024  600 601 603 625 +hsync +vsync (35.2 kHz e)
[  5779.009] (II) modeset(0): Modeline "640x480"x0.0   31.50  640 656 720 840  480 481 484 500 -hsync -vsync (37.5 kHz e)
[  5779.009] (II) modeset(0): Modeline "640x480"x0.0   31.50  640 664 704 832  480 489 492 520 -hsync -vsync (37.9 kHz e)
[  5779.009] (II) modeset(0): Modeline "720x400"x0.0   28.32  720 738 846 900  400 412 414 449 -hsync +vsync (31.5 kHz e)
[  5779.009] (II) modeset(0): Modeline "1024x768"x0.0   78.75  1024 1040 1136 1312  768 769 772 800 +hsync +vsync (60.0 kHz e)
[  5779.009] (II) modeset(0): Modeline "1024x768"x0.0   75.00  1024 1048 1184 1328  768 771 777 806 -hsync -vsync (56.5 kHz e)
[  5779.009] (II) modeset(0): Modeline "800x600"x0.0   49.50  800 816 896 1056  600 601 604 625 +hsync +vsync (46.9 kHz e)
[  5779.009] (II) modeset(0): Modeline "800x600"x0.0   50.00  800 856 976 1040  600 637 643 666 +hsync +vsync (48.1 kHz e)
[  5779.062] (II) modeset(0): EDID vendor "ROW", prod id 0
[  5779.062] (II) modeset(0): Using hsync ranges from config file
[  5779.062] (II) modeset(0): Using vrefresh ranges from config file
[  5779.062] (II) modeset(0): Printing DDC gathered Modelines:
[  5779.062] (II) modeset(0): Modeline "1024x768"x0.0   65.00  1024 1048 1184 1344  768 771 777 806 -hsync -vsync (48.4 kHz eP)
[  5779.062] (II) modeset(0): Modeline "1280x720"x0.0   74.25  1280 1720 1760 1980  720 725 730 750 +hsync +vsync (37.5 kHz e)
[  5779.062] (II) modeset(0): Modeline "1920x1080i"x0.0   74.25  1920 2448 2492 2640  1080 1084 1094 1125 interlace +hsync +vsync (28.1 kHz e)
[  5779.062] (II) modeset(0): Modeline "720x480"x0.0   27.00  720 736 798 858  480 489 495 525 -hsync -vsync (31.5 kHz e)
[  5779.062] (II) modeset(0): Modeline "720x576"x0.0   27.00  720 732 796 864  576 581 586 625 -hsync -vsync (31.2 kHz e)
[  5779.062] (II) modeset(0): Modeline "1920x1080"x0.0  148.50  1920 2008 2052 2200  1080 1084 1089 1125 +hsync +vsync (67.5 kHz e)
[  5779.062] (II) modeset(0): Modeline "1920x1080"x0.0  148.50  1920 2448 2492 2640  1080 1084 1089 1125 +hsync +vsync (56.2 kHz e)
[  5779.062] (II) modeset(0): Modeline "1920x1080i"x0.0   74.25  1920 2008 2052 2200  1080 1084 1094 1125 interlace +hsync +vsync (33.8 kHz e)
[  5779.062] (II) modeset(0): Modeline "1280x720"x0.0   74.25  1280 1390 1430 1650  720 725 730 750 +hsync +vsync (45.0 kHz e)
[  5779.062] (II) modeset(0): Modeline "1440x576i"x0.0   27.00  1440 1464 1590 1728  576 580 586 625 interlace -hsync -vsync (15.6 kHz e)
[  5779.062] (II) modeset(0): Modeline "1440x480i"x0.0   27.00  1440 1478 1602 1716  480 488 494 525 interlace -hsync -vsync (15.7 kHz e)
[  5779.062] (II) modeset(0): Modeline "640x480"x0.0   25.18  640 656 752 800  480 490 492 525 -hsync -vsync (31.5 kHz e)
[  5779.062] (II) modeset(0): Modeline "800x600"x0.0   40.00  800 840 968 1056  600 601 605 628 +hsync +vsync (37.9 kHz e)
[  5779.062] (II) modeset(0): Modeline "800x600"x0.0   36.00  800 824 896 1024  600 601 603 625 +hsync +vsync (35.2 kHz e)
[  5779.062] (II) modeset(0): Modeline "640x480"x0.0   31.50  640 656 720 840  480 481 484 500 -hsync -vsync (37.5 kHz e)
[  5779.062] (II) modeset(0): Modeline "640x480"x0.0   31.50  640 664 704 832  480 489 492 520 -hsync -vsync (37.9 kHz e)
[  5779.062] (II) modeset(0): Modeline "720x400"x0.0   28.32  720 738 846 900  400 412 414 449 -hsync +vsync (31.5 kHz e)
[  5779.062] (II) modeset(0): Modeline "1024x768"x0.0   78.75  1024 1040 1136 1312  768 769 772 800 +hsync +vsync (60.0 kHz e)
[  5779.062] (II) modeset(0): Modeline "1024x768"x0.0   75.00  1024 1048 1184 1328  768 771 777 806 -hsync -vsync (56.5 kHz e)
[  5779.062] (II) modeset(0): Modeline "800x600"x0.0   49.50  800 816 896 1056  600 601 604 625 +hsync +vsync (46.9 kHz e)
[  5779.062] (II) modeset(0): Modeline "800x600"x0.0   50.00  800 856 976 1040  600 637 643 666 +hsync +vsync (48.1 kHz e)
[  5779.115] (II) modeset(0): EDID vendor "ROW", prod id 0
[  5779.115] (II) modeset(0): Using hsync ranges from config file
[  5779.115] (II) modeset(0): Using vrefresh ranges from config file
[  5779.115] (II) modeset(0): Printing DDC gathered Modelines:
[  5779.115] (II) modeset(0): Modeline "1024x768"x0.0   65.00  1024 1048 1184 1344  768 771 777 806 -hsync -vsync (48.4 kHz eP)
```

lurch, 2025-03-13:
ping @cillian64

cillian64, 2025-03-13:
@lurch I'm afraid I only do Wayland support, not X.  I recommend asking Dave and Dom if they have any ideas, and if not, asking Igalia as they're responsible for our Mesa support in general

lurch, 2025-03-13:
ping @6by9 and @popcornmix 😉

lurch, 2025-03-13:
@sebastiendu I've now tried reproducing this myself on a Raspberry Pi 4 B, running a fresh 64-bit Raspberry Pi OS image, fully up-to-date with APT. After using `raspi-config` to switch to X11 and rebooting, I _don't_ see loads of extraneous lines in the Xorg log? `wc -l /var/log/Xorg.0.log` only reports 512 lines. I also apt-installed `mesa-utils` and ran `glxgears` but that didn't result in anything being added to `Xorg.0.log` either.

Are you able to reproduce this on an "idle desktop" or could your "OpenGL3/GLFW program" be doing something to provoke all these additional EDID probes? If you can tell me how to reproduce your problem, that'd be very useful!

lpopo0856, 2025-03-13:
I ran into the same issue. Hoping this [thread](https://github.com/raspberrypi/firmware/issues/1951) will help!

lurch, 2025-03-13:
@lpopo0856 I'm afraid that I'm unwilling to build a custom pi-gen image, just to investigate problems with Xorg. But if you can provide me with step by step instructions for how to reproduce / test your issue starting from a fresh install of a Raspberry Pi OS image (either 32-bit or 64-bit) then I'll take a closer look. Thanks.

teafella, 2025-03-13:
Ive been pointed here from  [my post on the pi forums](https://forums.raspberrypi.com/viewtopic.php?p=2302669) and I wanted to report that I am having similar issues with direct rendering without X involved at all (/var/log/Xorg.0.log doesnt even exist on my system). Would be great to figure out whats happening here, I will also be rolling back for the time being but I suspect that X is not at the root of the issue.

ak-rex, 2025-03-14:
I'm having the same issue on Labwc. llvmpipe is being used instead of V3D. If I downgrade the mesa packages from `24.2.8-1~bpo12+rpt1` to version `23.2.1-1~bpo12+rpt3` Labwc starts using V3D.
Even when using `24.2.8-1~bpo12+rpt1` on the same install where Labwc will only use llvmpipe, KDE Plasma wayland will use V3D.

lurch, 2025-03-14:
@teafella and @ak-rex I wonder if you're actually suffering from #381 rather than this issue?

lurch, 2025-03-14:
@ak-rex If you can provide step by step instructions on how to reproduce your issue, and how to "detect" if you're using llvmpipe vs V3D, that'd be very useful! (this is an area of the software stack that I'm not very familiar with yet)

ak-rex, 2025-03-14:
I have a device running Waydroid, after an apt upgrade it started having bad frame rate skips in Labwc. Switching over to KDEplasma solved it. I just updated and started seeing slowdowns.
Running `glmark2-es2-wayland` shows Labwc llvmpipe as the GL_RENDERER https://ibb.co/RGyc4Hp9
Same SD install switching to KDEplasma uses V3D as the GL_RENDERER https://ibb.co/HpfhmzPT
Downgrading or moving to a backup before the update uses V3D as the GL_RENDERER https://ibb.co/ksc8cCyH

ak-rex, 2025-03-15:
Here's the output of `glmark2-es2-wayland` after loading into Labwc.
<details>
  <summary>Output</summary>
  
```
libEGL debug: using driver drm-rp1-dsi for 6

libEGL debug: MESA-LOADER: device is not located on the PCI bus

libEGL debug: using driver drm-rp1-dsi for 6

libEGL debug: EGL user error 0x3003 (EGL_BAD_ALLOC) in eglInitialize: DRI2: failed to load driver

libEGL debug: did not find extension DRI2_BufferDamage version 1

libEGL debug: did not find extension DRI2_Interop version 1

libEGL debug: did not find extension DRI2_Blob version 1

libEGL debug: did not find extension DRI_MutableRenderBufferDriver version 1

libEGL debug: did not find extension DRI_Kopper version 1

libEGL debug: No DRI config supports native format PIPE_FORMAT_R16G16B16A16_FLOAT
libEGL debug: No DRI config supports native format PIPE_FORMAT_R16G16B16X16_FLOAT
libEGL debug: No DRI config supports native format PIPE_FORMAT_R10G10B10X2_UNORM
libEGL debug: No DRI config supports native format PIPE_FORMAT_R10G10B10A2_UNORM
libEGL debug: No DRI config supports native format PIPE_FORMAT_R8G8B8A8_UNORM
libEGL debug: No DRI config supports native format PIPE_FORMAT_R8G8B8X8_UNORM
libEGL debug: No DRI config supports native format PIPE_FORMAT_B5G5R5A1_UNORM
libEGL debug: No DRI config supports native format PIPE_FORMAT_B5G5R5X1_UNORM
libEGL debug: No DRI config supports native format PIPE_FORMAT_B4G4R4A4_UNORM
libEGL debug: No DRI config supports native format PIPE_FORMAT_B4G4R4X4_UNORM
libEGL debug: eglPrintConfigDebug: nothing to print
libEGL debug: ---------------
libEGL debug: Configurations:
libEGL debug: cho       bf lv colourbuffer dp st  ms           vis  cav  bi     renderable           supported
libEGL debug: sen    id sz  l  r  g  b  a  th cl ns b           id  eat  nd  gl es es2 es3 vg         surfaces
libEGL debug: ---------------
libEGL debug:  30 0x001 32  0 10 10 10  2   0  0  0 0 0x00000000--            y  y   y   y                win,
libEGL debug:  32 0x002 32  0 10 10 10  2  16  0  0 0 0x00000000--            y  y   y   y                win,
libEGL debug:  34 0x003 32  0 10 10 10  2  24  0  0 0 0x00000000--            y  y   y   y                win,
libEGL debug:  36 0x004 32  0 10 10 10  2  24  8  0 0 0x00000000--            y  y   y   y                win,
libEGL debug:  38 0x005 32  0 10 10 10  2  32  0  0 0 0x00000000--            y  y   y   y                win,
libEGL debug:  40 0x006 32  0 10 10 10  2   0  0  4 1 0x00000000--            y  y   y   y                win,
libEGL debug:  42 0x007 32  0 10 10 10  2  16  0  4 1 0x00000000--            y  y   y   y                win,
libEGL debug:  44 0x008 32  0 10 10 10  2  24  0  4 1 0x00000000--            y  y   y   y                win,
libEGL debug:  46 0x009 32  0 10 10 10  2  24  8  4 1 0x00000000--            y  y   y   y                win,
libEGL debug:  48 0x00a 32  0 10 10 10  2  32  0  4 1 0x00000000--            y  y   y   y                win,
libEGL debug:  20 0x00b 30  0 10 10 10  0   0  0  0 0 0x00000000--            y  y   y   y                win,
libEGL debug:  21 0x00c 30  0 10 10 10  0  16  0  0 0 0x00000000--            y  y   y   y                win,
libEGL debug:  22 0x00d 30  0 10 10 10  0  24  0  0 0 0x00000000--            y  y   y   y                win,
libEGL debug:  23 0x00e 30  0 10 10 10  0  24  8  0 0 0x00000000--            y  y   y   y                win,
libEGL debug:  24 0x00f 30  0 10 10 10  0  32  0  0 0 0x00000000--            y  y   y   y                win,
libEGL debug:  25 0x010 30  0 10 10 10  0   0  0  4 1 0x00000000--            y  y   y   y                win,
libEGL debug:  26 0x011 30  0 10 10 10  0  16  0  4 1 0x00000000--            y  y   y   y                win,
libEGL debug:  27 0x012 30  0 10 10 10  0  24  0  4 1 0x00000000--            y  y   y   y                win,
libEGL debug:  28 0x013 30  0 10 10 10  0  24  8  4 1 0x00000000--            y  y   y   y                win,
libEGL debug:  29 0x014 30  0 10 10 10  0  32  0  4 1 0x00000000--            y  y   y   y                win,
libEGL debug:  31 0x015 32  0  8  8  8  8   0  0  0 0 0x00000000--            y  y   y   y                win,
libEGL debug:  33 0x016 32  0  8  8  8  8  16  0  0 0 0x00000000--            y  y   y   y                win,
libEGL debug:  35 0x017 32  0  8  8  8  8  24  0  0 0 0x00000000--            y  y   y   y                win,
libEGL debug:  37 0x018 32  0  8  8  8  8  24  8  0 0 0x00000000--            y  y   y   y                win,
libEGL debug:  39 0x019 32  0  8  8  8  8  32  0  0 0 0x00000000--            y  y   y   y                win,
libEGL debug:  41 0x01a 32  0  8  8  8  8   0  0  4 1 0x00000000--            y  y   y   y                win,
libEGL debug:  43 0x01b 32  0  8  8  8  8  16  0  4 1 0x00000000--            y  y   y   y                win,
libEGL debug:  45 0x01c 32  0  8  8  8  8  24  0  4 1 0x00000000--            y  y   y   y                win,
libEGL debug:  47 0x01d 32  0  8  8  8  8  24  8  4 1 0x00000000--            y  y   y   y                win,
libEGL debug:  49 0x01e 32  0  8  8  8  8  32  0  4 1 0x00000000--            y  y   y   y                win,
libEGL debug:  10 0x01f 24  0  8  8  8  0   0  0  0 0 0x00000000--            y  y   y   y                win,
[... 11 lines cut ...]
libEGL debug:   2 0x02b 16  0  5  6  5  0  24  0  0 0 0x00000000--            y  y   y   y                win,
[... cut ...]
    GL_RENDERER:    llvmpipe (LLVM 15.0.6, 128 bits)
    GL_VERSION:     OpenGL ES 3.2 Mesa 24.2.8-1~bpo12+rpt1
    Surface Config: buf=32 r=8 g=8 b=8 a=8 depth=32 stencil=0 samples=0
    Surface Size:   800x600 windowed
=======================================================
[build] use-vbo=false:^C FPS: 198 FrameTime: 5.072 ms
=======================================================
                                  glmark2 Score: 197 
=======================================================

  ```
</details>

sebastiendu, 2025-03-17:
After more research, here is some useful information.
 
- glxgears triggers the same Xorg log flood as my program
- we still only use the _32-bit_ Raspberry Pi OS Lite, I have not yet tried this on the 64-bits one.
- we install **lightdm** and configure it with an autologin to a custom **i3-wm** session setup with no decoration.
- So far I could not reproduce the bug on a pure Xorg session, so, maybe, the bug is in i3 and was just not visible before Mesa 24.

Below is how to corrupt a fresh and pure _Raspberry Pi OS Lite (32-bit)_ so to witness the bad behavior.

```
apt update
apt full-upgrade
apt install i3 lightdm mesa-utils
systemctl set-default graphical.target
cat <<EOF >/usr/share/lightdm/lightdm.conf.d/local.conf
[Seat:*]
autologin-user=pi
autologin-user-timeout=0
autologin-session=i3
EOF
```
Edit /etc/i3/config so to modify the last few lines:
remove the "bar" section (3 lines)
```
#bar {
#        status_command i3status
#}
```
and replace `exec i3-config-wizard` with these two lines:
```
# exec i3-config-wizard
default_border none
exec glxgears
```

then `reboot`.
You should see the glxgears in full screen.

See the poor refresh rate, less than 19 fps instead of 60 on a 1920×1080 HDMI display: `tail -f .xsession-errors`

Check the log flood: `tail -f /var/log/Xorg.0.log`

mijutu, 2025-04-14:
The same problem occurs on 64bit too with a simple libvlc application. No window manager needed.

- apt install libvlc-dev vlc
- copy code from https://wiki.videolan.org/LibVLC_Tutorial/#Sample_libVLC_code and set path or url
- compile with -lvlc
- start Xorg with X -ac -nolisten tcp -retro -noreset
- run libvlc sample: Xorg uses a lot of cpu.
- run plain vlc with the same file: No exessive cpu usage.