raspberrypi/bookworm-feedback #388: gtk-nop starts automatically but doesn't seem to be useful

Intranox, 2025-03-13:
I recently noticed this package (gtk-nop) that weighs more than 50MB in main memory but seems useless. I don't think I installed it but rather I think it is included in the system and starts automatically. Is it possible to remove it without problems?

lurch, 2025-03-13:
IIRC it's something to do with the power-button handling on Raspberry Pi 5? ping @spl237

closed 2025-03-13 by spl237

Intranox, 2025-03-13:
> Per impostazione predefinita, il pulsante di accensione hardware di un Raspberry Pi 5 spegne il sistema non appena viene premuto.
> 
> In genere, questo non è un bene, perché se lo si preme inavvertitamente, si rischia di perdere il lavoro non salvato ecc.
> 
> Per evitare ciò, il suo effetto viene inibito con un systemd-inhibit mentre il desktop è in esecuzione, il che significa che invece di arrestare direttamente il sistema, il suo effetto è quello di attivare un gestore di tasti nel gestore delle finestre, che apre la finestra di dialogo delle opzioni di arresto.
> 
> Un'inibizione systemd deve essere associata a un'applicazione, in modo che l'inibizione esista mentre l'applicazione è in esecuzione.
> 
> gtk-nop è, come suggerisce il nome, un'applicazione GTK (ovvero Windows) che non fa nulla: non apre una finestra; rimane lì come istanza di un'applicazione GTK. Ma poiché è un'applicazione GTK che richiede un gestore di finestre, si chiuderà automaticamente quando il gestore di finestre termina, il che a sua volta significa che se si esce dal desktop, l'inibizione si chiude e il tasto di accensione riprende il suo comportamento predefinito una volta tornati alla console. (Nella console, non è possibile far apparire una finestra di dialogo di conferma, quindi non ha senso inibire il comportamento predefinito.)
> 
> In conclusione, se non stai usando un Pi 5, non fa nulla di utile, dato che non hai comunque un pulsante di accensione. Se stai usando un Pi 5 e sei contento che se premi il pulsante di accensione, il tuo Pi si spegnerà senza alcuna conferma, allora puoi andare avanti e rimuoverlo: la cosa migliore da fare è semplicemente eliminare il file /etc/xdg/autostart/pwrkey.desktop, che impedirà quindi l'avvio del processo di inibizione.

Many thanks. Excellent and complete explanation. Having a pi4 I will remove the automatic startup file, also because in any case 50MB in main memory, depending on the model of the raspberry pi in use, are enough and saving them does not hurt

> The hardware power button on a Raspberry Pi 5 will by default shut the system down as soon as it is pressed.
> 
> This is generally not a good thing, as it will mean you can lose unsaved work etc if you press it inadvertently.
> 
> In order to avoid this, its effect is inhibited with a systemd-inhibit while the desktop is running, which means instead of directly shutting down the system, its effect is to trigger a key handler in the window manager, which pops up the shutdown options dialog.
> 
> A systemd-inhibit has to be associated with an application, such that the inhibit exists while that application is running.
> 
> gtk-nop is, as the name suggests, a GTK (i.e. windows) application which does nothing - it doesn't open a window; it just sits there being an instance of a GTK application. But because it is a GTK application which requires a window manager, it will automatically close when the window manager terminates, which in turn means that if you exit the desktop, the inhibit closes and the power key resumes its default behaviour once back at the console. (In the console, there is no ability to pop up a confirmation dialog, so there is no point in inhibiting the default behaviour.)
> 
> Bottom line - if you are not running on a Pi 5, it doesn't do anything useful, as you don't have a power button anyway. If you are running a Pi 5 and you are happy that if you hit the power button, your Pi will shut down with no confirmation, then you can go ahead and remove it - the best thing to do is just to delete the file /etc/xdg/autostart/pwrkey.desktop, which will then prevent the inhibit process from starting.

Many thanks. Excellent and complete explanation. Having a pi4 I will remove the autostart file, also because 50MB in main memory, depending on the model of the raspberry pi in use, are enough and saving them does not hurt. I understand well however that in the case of the raspberry pi 5 it can be very useful