raspberrypi/bookworm-feedback #392: raspberrypi-ui-mods.postinst has incorrect regular expression

hackerb9, 2025-03-22:
The latest update to raspberrypi-ui-mods attempts to use the standard /etc/default/keyboard settings, which is very good, but unfortunately the regular expression it uses is broken. Here is a patch which fixes it:

```patch
--- /var/lib/dpkg/info/raspberrypi-ui-mods.postinst~	2025-03-18 04:55:58.000000000 -0700
+++ /var/lib/dpkg/info/raspberrypi-ui-mods.postinst	2025-03-21 20:41:57.221576729 -0700
@@ -23,10 +23,10 @@
           fi
 
 	  if pgrep labwc > /dev/null ; then
-	    MODEL=$(grep XKBMODEL /etc/default/keyboard | cut -d= -f2 | tr -d '"')
-	    LAYOUT=$(grep XKBLAYOUT /etc/default/keyboard | cut -d= -f2 | tr -d '"')
-	    VARIANT=$(grep XKBVARIANT /etc/default/keyboard | cut -d= -f2 | tr -d '"')
-	    OPTIONS=$(grep XKBOPTIONS /etc/default/keyboard | cut -d= -f2 | tr -d '"')
+	    MODEL=$(grep ^XKBMODEL= /etc/default/keyboard | cut -d= -f2 | tr -d '"')
+	    LAYOUT=$(grep ^XKBLAYOUT= /etc/default/keyboard | cut -d= -f2 | tr -d '"')
+	    VARIANT=$(grep ^XKBVARIANT= /etc/default/keyboard | cut -d= -f2 | tr -d '"')
+	    OPTIONS=$(grep ^XKBOPTIONS= /etc/default/keyboard | cut -d= -f2 | tr -d '"')
 	    mkdir -p "$HOME_DIR/.config/labwc/"
 	    chown -R $USER: "$HOME_DIR/.config/labwc/"
 	    UFILE="$HOME_DIR/.config/labwc/environment"
``` 

As you can see, the current script greps for _any_ and _all_ occurrences of the XKB phrases. 

<details><summary>For example, click here to see my /etc/default/keyboard:</summary>
<ul>

```shell
# Check /usr/share/doc/keyboard-configuration/README.Debian for
# documentation on what to do after having modified this file.

# The following variables describe your keyboard and can have the same
# values as the XkbModel, XkbLayout, XkbVariant and XkbOptions options
# in /etc/X11/xorg.conf.

# For XKBOPTIONS, see /usr/share/X11/xkb/rules/

XKBMODEL="pc105"
XKBLAYOUT="us"
XKBVARIANT=""
XKBOPTIONS="ctrl:nocaps,compose:rctrl,compose:rwin,compose:multikey,shift:both_capslock_cancel,keypad:pointerkeys,terminate:ctrl_alt_bksp"

# Comments from /usr/share/X11/xkb/rules/base.lst on b9's preferences:
#  ctrl:nocaps                  Caps Lock as Ctrl
#  compose:rctrl                Right Ctrl is Compose Key
#  compose:rwin                 Right Win [RasberryPi has no rctrl]
#  shift:both_capslock          Both Shift together toggle Caps Lock
#  shift:both_capslock_cancel   Same as above, but one Shift key disables it
#  keypad:pointerkeys           Shift + Num Lock enables PointerKeys
#  terminate:ctrl_alt_bksp      Kill the X server with Ctrl+Alt+Backspace

# If you don't want to use the XKB layout on the console, you can
# specify an alternative keymap.  Make sure it will be accessible
# before /usr is mounted.
# KMAP=/etc/console-setup/defkeymap.kmap.gz
BACKSPACE="guess"
```

</ul>
</details>

The important point to note is that the phrase "XKBOPTIONS" is used in a comment. The postinst script is mistakenly treating the comment as a variable and dying like so:

```shell
sed -i 's/XKB_DEFAULT_OPTIONS.*/XKB_DEFAULT_OPTIONS=#' For XKBOPTIONS, see /usr/share/X11/xkb/rules/ ctrl:nocaps,compose:rctrl,compose:rwin,compose:multikey,shift:both_capslock_cancel,keypad:pointerkeys,terminate:ctrl_alt_bksp/ /root//.config/labwc/environment
sed: -e expression #1, char 45: unterminated `s' command
```

The solution, as seen in my patch, is to change it to only match if the variable name is found at the beginning of the line and directly followed by `=` (to ensure it is not a prefix for a longer variable). My patch does not handle the (unlikely) case where a variable is set more than once, but if one wanted to be extra sure, simply pipe from grep into `tail -1`.

____

<details><summary><h3>Sidenotes</h3></summary>

The following considerations are less important, but related:

1. It seems like a bad idea to have a system script which writes into user directories, obliterating any customization the user may have previously set up. Wouldn't it make more sense to modify `/etc/xdg/labwc/environment` to change the system defaults? It would seem a lot simpler than looping over all the home directories.
1. Modifying labwc's config files, whether in the user or system directories, is not an ideal solution as doing so requires the Raspberry Pi owner to know a hidden extra step after modifying /etc/default/keyboard.
    1. The best solution is to patch labwc to read /etc/default/keyboard in addition its own config files.
    2. If that is not possible, please consider putting a note in the default /etc/default/keyboards letting people know to re-run `dpkg-reconfigure raspberrypi-ui-mods` after making changes.
1.  Why is the postinst script only making modification if labwc happens to be running? Wouldn't it be better to check if labwc is installed? 

</details>

hackerb9, 2025-03-24:
### Sidenote 4

By the way, fixing this script to edit `/etc/xdg/labwc/environment` instead of every user's config file would also have the side benefit of fixing the issue where the keyboard layout is set incorrectly at graphical login. This is the bug where some people — although  they can login fine from a virtual console — cannot login from pi-greeter. 

The default /etc/xdg/labwc/environment currently contains:

```
XKB_DEFAULT_MODEL=pc105
XKB_DEFAULT_LAYOUT=gb
```

Most likely these users are using a US layout and their password contains <kbd>@</kbd> or <kbd>#</kbd> which invisibly become <kbd>"</kbd> or <kbd>£</kbd>.

spl237, 2025-04-01:
I've pulled in your fix - many thanks.

As for your other points, I don't believe there is an ideal solution to this problem - in the past we did address this by editing the files in /etc, but this then made it more awkward when we needed to make changes to those files that we wanted to apply to all users, and that is still a likely contingency as labwc is under active development. On balance, for now I think it is better to remain with the system we are currently using, but I will keep this under review for a future change.

closed 2025-04-01 by spl237

hackerb9, 2025-04-01:
> I've pulled in your fix - many thanks.

You are quite welcome. I hope to contribute more in the future.

> As for your other points, I don't believe there is an ideal solution to this problem - in the past we did address this by editing the files in /etc, but this then made it more awkward when we needed [...]

Yes, I left /etc vs $HOME as a sidenote because it seems like a tricky problem. I should probably mention, though, that the time when a program like labwc is under active development is the best time to get things like this right.