raspberrypi/bookworm-feedback #397: Setting waveshare 3.5 inch on Raspberry Pi Compute Module 3+ bookworm lite

CJ-012, 2025-03-31:
I have been trying to set up the LCD for days now, and I'm totally stumped. I realized that they had the manuals for bookworm, but I tried everything I saw.

Now, does anyone know how to set it up from scratch?

https://www.waveshare.com/wiki/3.5inch_RPi_LCD_(A)_Manual_Configuration

lurch, 2025-03-31:
I'm afraid that this is something that you'll have to ask Waveshare about. You might be able to get support from the community at https://forums.raspberrypi.com/

(Also sending @6by9 a quick ping, in case this is a screen that he has experience with)

closed 2025-03-31 by lurch

6by9, 2025-03-31:
fbcp is dead.

Sorry, but I won't go decompiling random dtoverlays from vendors.

Looking at the images off https://www.waveshare.com/product/3.5inch-RPi-LCD-A.htm, it's a clone of the Ozmaker PiScreen.

`dtoverlay=piscreen` should work, although recently I've found the need to drop the SPI frequency to 18MHz using `,speed=18000000`, and generally go for the DRM driver rather than the deprecated FBTFT driver by adding `drm`.
ie complete command of `dtoverlay=piscreen,speed=18000000,drm`. That's certainly worked with my display that looks incredibly similar (I really should make a note of where I buy these things from!)

CJ-012, 2025-03-31:
So should I not install the drivers?

6by9, 2025-03-31:
No additional drivers are needed as they're included in the standard kernel image.

And we won't recommend disabling `dtoverlay=vc4-kms-v3d` as that disables all graphics acceleration. fbcp will never work on Pi5 as the relevant firmware hooks aren't there.

CJ-012, 2025-03-31:
So, don’t download any drivers. Just add the dtoverlay=piscreen,speed=18000000,drm at the end of /boot/firmware/config.txt

6by9, 2025-04-01:
Yup.

CJ-012, 2025-04-01:
When I did that, I rebooted and my tv went black and the lcd stayed white. Also, when I went to check if the spi was on, it said not found. But, when I commented it out and reboot, it would say that spi is on and it displays on the tv again.

6by9, 2025-04-01:
The order of DRM devices being probed can vary, and whichever enumerates first will take /dev/fb0 and the console.
You should be able to add `dtparam=drm_fb0_vc4` to force vc4 (HDMI outputs) to take /dev/fb0 and hence get the console rendered to it. (Note that if there are no outputs of vc4 connected, then /dev/fb0 will NOT be assigned to any other display device, and you will not have a visible console).

SPI access from userspace will be disabled if you load a kernel driver that uses it - that is intentional as you don't want userspace sending random rubbish to a device the kernel is managing.

Did you check the kernel logs for any errrors? (`dmesg`).
Or check with `kmsprint` as to whether the display was initialised? (you may need to add `--device=/dev/dri/card2` if vc4 is also active).

All I can say is that I tested my panel (actually an ELEGOO panel from https://www.amazon.co.uk/ELEGOO-480x320-Raspberry-Monitor-Interface/dp/B01MRQTMTD) using that dtoverlay command yesterday, and it worked fine.

CJ-012, 2025-04-01:
When I did dmesg | grep spi it says
 [drm] Initialized ili9486 1.0.0 20200118 for spi0.0 on minor 0
 ili9486 spi0.0: [drm] fb1: ili9486drmfb frame buffer device 
ads7846 spi0.1: supply vcc not found, using dummy regulator 
ads7846 spi0.1: touchscreen, irq 184 input: ADS7846 Touchscreen as /devices/platform/soc/3f204000.spi/spi_master/spi0/spi0.1/input/input3

6by9, 2025-04-01:
Sounds fine.

Now test it with `kmstest`, or `kmstest --device=/dev/dri/card2` if that initial test put the test card up on your HDMI output.

6by9, 2025-04-01:
Seeing as you've filed this under Bookworm-feedback, I am assuming you are using Bookworm and therefore have the latest kernel (should now be 6.12.20, as reported by `uname -a`).

CJ-012, 2025-04-01:
Yes

CJ-012, 2025-04-01:
[image]

6by9, 2025-04-01:
That's as I'd expect, but I'd also expect to get a coloured test pattern on the SPI display whilst running kmstest.

Have you ever had this display working? If not, then it is possible that the display itself is just duff.

CJ-012, 2025-04-01:
I haven't ever had it display anything. It's just a white screen. Do you want to see how I have it connected on the board?

CJ-012, 2025-04-01:
Never mind that was for the hdmi. However, I don’t have another board on me atm.

6by9, 2025-04-01:
Pi5 with the previously linked screen and that same dtoverlay line.

[image]

I'd suspect your wiring is the issue.

CJ-012, 2025-04-01:
This is how I connected it
2 5V 
6 GND
11 TP_IRQ touch screen interrupt -- GPIO(18)
18 LCD_RS data register selection -- GPIO(25)
19 LCD_SI SPI data input of LCD -- GPIO(10)
21 TP_SO SPI data output of touchscreen -- GPIO(9)
22 RST reset -- GPIO(27)
23 LCD_SCK SPI clock of LCD -- GPIO(11)
24 LCD_CS chip LCD selection -- GPIO(8)
26 TP_CS chip touchscreen selection -- GPIO(7)

CJ-012, 2025-04-02:
I finally got it working after re-wiring the pins. Thank you so much for your help!