raspberrypi/bookworm-feedback #399: GUI overwrites ~/.config/labwc/rc.xml

theoldwizard1, 2025-03-31:
When using the standard desktop, Raspberry icon -> Preferences -> Keyboard and Mouse and then changing any settings, **~/.config/labwc/rc.xml gets OVERWRITTEN** during the next reboot.

**BEFORE**

```
$ cat ~/.config/labwc/rc.xml
<?xml version="1.0" encoding="UTF-8"?>
<labwc_config>
	<placement>
		<policy>cascade</policy>
	</placement>
</labwc_config>
```

AFTER

```
$ cat ~/.config/labwc/rc.xml.2
<?xml version="1.0" encoding="UTF-8"?>
<openbox_config xmlns="http://openbox.org/3.4/rc"><libinput><device category="default"><pointerSpeed>0.000000</pointerSpeed><leftHanded>no</leftHanded></device></libinput><mouse><doubleClickTime>400</doubleClickTime></mouse><keyboard><repeatRate>25</repeatRate><repeatDelay>600</repeatDelay></keyboard></openbox_config>
```

You can not append the original file to the new one because the configuration is "closed".  "openbox_config" does not "understand" <placement>.

lurch, 2025-04-01:
Yes, I can reproduce your example. It's unfortunate that any existing `<labwc_config>` section gets wiped and replaced with a `<openbox_config>` section. **However** if I use:
```xml
<?xml version="1.0"?>
<openbox_config>
    <placement>
        <policy>cascade</policy>
    </placement>
</openbox_config>
```
and then use Raspberry icon -> Preferences -> Keyboard and Mouse and then changing any settings, then `~/.config/labwc/rc.xml` gets updated to:
```xml
<?xml version="1.0"?>
<openbox_config>
    <placement>
        <policy>cascade</policy>
    </placement>
<libinput><device category="default"><pointerSpeed>-0.220000</pointerSpeed></device></libinput><mouse><doubleClickTime>927</doubleClickTime></mouse></openbox_config>
```
which I think is the behaviour that you're after? (and yes, the "window cascading" sill works)

@spl237 When updating the `~/.config/labwc/rc.xml` file, would i be possible to merge any `<labwc_config>` section into the `<openbox_config>` section, instead of just overwriting it?

theoldwizard1, 2025-04-01:
SUCCESS !

Why is the new rc.xml file written out as a `<openbox_config>` section ?

lurch, 2025-04-01:
I **think** when I asked Simon about this quite a while ago, he said something like `<labwc_config>` and `<openbox_config>` are functionally equivalent, and he was using `<openbox_config>` to maintain backwards compatibility? But I may be misremembering...

spl237, 2025-04-01:
> [@spl237](https://github.com/spl237) When updating the `~/.config/labwc/rc.xml` file, would it be possible to merge any `<labwc_config>` section into the `<openbox_config>` section, instead of just overwriting it?

All things are possible...but some are easier than others... ;)

At present, I don't know - if libxml has a convenient function for merging different parts of an XML tree, then it should be relatively straightforward. If it doesn't, it probably isn't!

lurch, 2025-04-01:
> At present, I don't know - if libxml has a convenient function for merging different parts of an XML tree, then it should be relatively straightforward. If it doesn't, it probably isn't!

It seems _very_ unlikely that a file would contain both an `<openbox_config>` section and a `<labwc_config>` section, so I wonder if a hacky solution might be to load the XML file, and then if there's a `labwc_config` root node but not an `openbox_config` root node, simply rename the `labwc_config` node to `openbox_config` before continuing? (If that works, I believe that would resolve the OP's original complaint?)

theoldwizard1, 2025-04-02:
Thanks for the QUALITY replies !

I used to be a sys admin on multiple UN*X box about 20 years ago. UN*X as in Solaris and OSF/1.  Never saw Linux back then.  I forgot at least half of what I used to know and RPi OS/Linux has MANY different feature like wayland, labwc and even sytemctl and using XML to configure things !  I have a good friend who recently retired and IS experienced in Linux (SUSE?)  who is helping be a lot.

Maybe Simon will accept the change of merging XML trees.  _(I still think it is a bit odd that the replacement rc.xml is an `<openbox_config>` section.)_

**How do you setup a system wide labwc rc.xml ?**  I used to know how to do this for CDE _(yes, I am that old)_.

lurch, 2025-04-02:
> Thanks for the QUALITY replies !

😁 

> How do you setup a system wide labwc rc.xml ?

You're probably looking for `/etc/xdg/labwc/rc.xml` ?

> I used to know how to do this for CDE _(yes, I am that old)_.

The horror! 😱  I remember using CDE at Uni.

lurch, 2025-04-02:
@theoldwizard1 Simon's comment [here](https://github.com/raspberrypi/bookworm-feedback/issues/400#issuecomment-2772081463) is also slightly applicable to this issue.

theoldwizard1, 2025-04-02:
I added `<placement>cascade</placement>` to `/etc/xdg/labwc/rc.xml`.  I know it will get whacked if the is and `update/upgrade`.