raspberrypi/bookworm-feedback #400: Policy for configuration files

hackerb9, 2025-04-01:
<blockquote>
<details><summary><h3>Sidenotes</h3></summary>

The following considerations are less important, but related:

1. It seems like a bad idea to have a system script which writes into user directories, obliterating any customization the user may have previously set up. Wouldn't it make more sense to modify `/etc/xdg/labwc/environment` to change the system defaults? It would also seem a lot simpler than looping over all the home directories.
1. Modifying labwc's config files, whether in the user or system directories, is not an ideal solution as doing so requires the Raspberry Pi owner to know a hidden extra step is needed after modifying /etc/default/keyboard.
    1. The best solution is to patch labwc to read /etc/default/keyboard in addition to its own config files.
    2. If that is not possible, please consider adding a note in /etc/default/keyboard letting people know to re-run `dpkg-reconfigure raspberrypi-ui-mods` after making changes. 

</details>

> As for your other points, I don't believe there is an ideal solution to this problem - in the past we did address this by editing the files in /etc, but this then made it more awkward when we needed to make changes to those files that we wanted to apply to all users, and that is still a likely contingency as labwc is under active development. On balance, for now I think it is better to remain with the system we are currently using, but I will keep this under review for a future change. 

 _Originally posted by @spl237 in [#392](https://github.com/raspberrypi/bookworm-feedback/issues/392#issuecomment-2768948039)_
</blockquote>

I would like to suggest the eventual creation of a policy on how Raspberry Pi OS will handle configuration files in the future. 

In particular, the current situation where a package's install script can replace every user's personal preferences in each user's home directory should be addressed. In my humble opinion, it is a mistake. It prevents users from being able to have lasting changes. Also, some apps, like labwc, do not check the system config files (in /etc) if a personal config file is found (in $HOME). The reasonable presumption is the file exists because the user created it to override the system defaults.

The question is, if not by modifying every user's config file, how can Raspberry Pi OS Devs make changes they want to apply to all users? It's not an easy problem, but it has been tackled by others and it may make sense to look at historical practice.
_____

I believe there are (at least) four main actors who might want to write to configuration files:

1. **User**. The person running the software; most impacted by choices, least level of expertise. 
2. **Admin**. The system administrator (root) in charge of choices that affect all users on a single machine (e.g., keyboard layout). On Raspberry Pi OS this is usually presumed to be the same as the user.
3. **OS Devs**. The folks who choose the defaults for anyone who downloads a distribution. 
4. **App Devs**. The people who write programs like labwc that need to be configurable by any of the above and fallback to sane defaults.

UNIX has traditionally followed a scheme formalized in the Linux File System Hierarchy Standard:

4. App Devs place config files that shouldn't be changed under /user/share (or /usr/lib). Example config files for the local Admin to edit go in /etc.
3. OS Devs modify what the App Devs did, if necessary. 
2. The Admin edits files in /etc only.
1. The User edits files in $HOME only.

Debian (dpkg) handles config files in /etc specially because there could be a conflict during an upgrade if a file has been modified by the local Admin and a newer version from the Devs is available. Debian stops the upgrade and asks the Admin how to resolve the situation, defaulting to keeping the current config file. This gives the Admin the chance to merge recent updates with the local system's preferences.

This is not a desirable situation, so Debian often will allow Admins to put only the preferences they've changed inside a directory (/etc/foo.conf.d) instead of editing a single, monolithic config file (/etc/foo.conf).

____

So, if Raspberry Pi OS wanted to change an app's defaults in an update, they would modify only the system configuration file in /etc. Typically, only an expert user would create a personal config file in $HOME. That means the app will use the system default for nearly everyone. (Experts presumably know what they are doing and can figure it out if it breaks.)

Labwc, in particular, automatically uses the config file in /etc if no personal config file exists in $HOME. Unfortunately, personal configs are currently sure to exist, due to the automated creation by raspberry-pi-uimods. I suggest modifying the install script to remove automatically generated config files so labwc can once again use /etc/xdg/labwc/labwc.conf.

lurch, 2025-04-02:
A couple of "subtleties" that immediately spring to mind:
* You mention that some applications ignore any settings in `/etc` if there's a user-specific config file in $HOME. However I believe there might be _some_ applications which merge any user-specific config "on top of" the settings in `/etc`?
* It's not always immediately obvious which things are "system level" settings, and which things are "user level" settings? e.g. you've asserted that the keyboard layout is a system-level thing; but if there are multiple users on the Pi, one user might want their locale settings set to French and another might want them set to Spanish? (and of course things like the Desktop background will always be user-specific)

But I'm sure that Simon has already thought about all of this in much more depth than I have... 😉

lurch, 2025-04-02:
> there would be no temptation for RPi OS to write to the user's home directory if changes can be made in /etc.

I'm not sure what you mean by this? To my mind, "RPi OS" includes both the system-level packages like the `raspberrypi-ui-mods` you've mentioned, but also the user-level packages like `raspi config`, "Appearance Settings", "Screen Configuration", "Raspberry Pi Configuration" etc. And some of those things (e.g. the "Interfaces" tab in "Raspberry Pi Configuration") are definitely a system-level setting, whereas others (e.g. the "Desktop" tab in "Appearance Settings") are definitely a user-level setting, and some things might be somewhere inbetween?

I'm not disagreeing with anything you're saying, I'm just pointing out that this can all be quite complex and not have a "single obvious solution" ? 🤔 

> Imagine a Hungarian developer who writes an app that defaults to her locale, RPi OS packages it in a way that sets the default to "UK", the admin of a Pi 500 at a school overrides the system default to US, and a teacher chooses Japanese for his account on the Pi.

And herein lies another "subtlety" - as you've pointed out, a lot of Raspberry Pi setups are single-user, in which case the "User" and the "Admin" are the same person. And Raspberry Pi OS obviously has to be very beginner-friendly, so most people might not even realise that "system level config" and "user level config" are two different things - they expect everything to "just work". And so somehow we have to balance this requirement with also offering the ability to cater to more advanced / complex setups, e.g. the "multi-user Pi in a school setup" example that you provided 😉