raspberrypi/bookworm-feedback #401: /dev/serial0 missing

Milliways2, 2025-04-03:
Upgrading to the latest firmware 6.12.20+rpt-rpi-v8 has broken
/dev/serial0 -> ttyAMA0

The udev rule 60-serial.rules (which is where I assume it is set) is unchanged.

lurch, 2025-04-03:
Can you provide more details please? What model of Raspberry Pi are you using? Which version of Raspberry Pi OS are you using? What does your `/boot/firmware/config.txt` say?

Milliways2, 2025-04-03:
Pi4, Pi3B+, Pi2  	Bookworm 12.10 64 bit, except Pi2
Serial is enabled in /boot/firmware/config.txt.
Several others have reported the same on different Pi.

JinShil, 2025-04-03:
I ran it to a similar issue a while back.  See https://github.com/raspberrypi/linux/issues/5566

hobbydevdude, 2025-04-04:
I have the same issue on my Pi3+, after latest apt upgrade there is no /dev/serial0 any longer. 
This broke the usage of my raspbee serial device with deconz. 
I needed to add dtoverlay=disable-bt  in /boot/firmware/config.txt to get deconz working again.

JuriPospelow, 2025-04-10:
I have the same issue on my Pi3+, after latest apt upgrade there is no /dev/serial0 any longer.
Serial is enabled in /boot/firmware/config.txt.
Upgade have about 200 packeges, but only update from:
            linux-headers-rpi-2712 linux-headers-rpi-v8 linux-image-rpi-2712 linux-image-rpi-v8 linux-libc-dev 
will /dev/serial0 brocken.
As workaround:
 sudo apt-mark hold linux-headers-rpi-2712 linux-headers-rpi-v8 linux-image-rpi-2712 linux-image-rpi-v8 linux-libc-dev

timg236, 2025-04-10:
I've submitted a workaround here, tested on Pi3, Pi4, Pi5. It will probably get updated if we can find a better strategy for mapping ttys to of_nodes or minimal kernel patch but I think "this will do" for now

https://github.com/RPi-Distro/raspberrypi-sys-mods

JuriPospelow, 2025-04-12:
I have removed my 100-serials.rules and changed  99-comm.rules with only 1 Line (I have Pi3):
       TTYNODE=$$(readlink /sys/class/tty/%k/device | sed 's/.*\\//\\/soc\\/serial@/' | sed 's/\\..*//' | sed 's/@f/@7/' ); \
  maked reboot and my /dev/serial0 did'nt exit again!
I have: 6.12.20+rpt-rpi-v8
What made i wrong?

lurch, 2025-04-14:
@Milliways2 This should now be fixed in the latest `apt upgrade` - can you confirm?
(see https://github.com/RPi-Distro/raspberrypi-sys-mods/pull/95 and https://github.com/RPi-Distro/raspberrypi-sys-mods/pull/97 if you're curious about the technical details)

lurch, 2025-04-15:
Can confirm, that this also isn't working correctly for me on a Pi 3B+. Thanks for the report. Investigating...

lurch, 2025-04-16:
@JuriPospelow I believe that I've got a fix in https://github.com/RPi-Distro/raspberrypi-sys-mods/pull/98 - would you like to test please?

JuriPospelow, 2025-04-19:
Yes, it work. I have Pi3B+ and tested serial0 to ttyS0, to ttyAMA0 and udevadmtest - all without errors.

lurch, 2025-04-23:
Closing, as the updated `raspberrypi-sys-mods` package has now been shipped in APT.

closed 2025-04-23 by lurch