raspberrypi/bookworm-feedback #402: My Rpi5 is not booting after update and upgrade

ctubig, 2025-04-07:
I have installed latest bookworm 64bit os on my sdcard 64gb. The initial setup was smooth and the desktop is showing with pre installed apps. I tried to run the sudo update and upgrade and the reboot but my Pi is not booting anymore after update and upgrade. If I use SanDisk sdcard 64gb and up. This happens consistently as i tried many times.

I have SanDisk 32gb sdcard where I also installed bookworm os, everything runs smoothly without a problem even after update and upgrade.

What would be the reason why it's not working on sdcard 64gb and up?

timg236, 2025-04-07:
You haven't provided much information about your system e.g. logs, configuration  etc.

For general support questions please see the Troubleshooting forum
https://forums.raspberrypi.com/viewforum.php?f=28&sid=6077096d4c3de455f48f62e254e12825

ctubig, 2025-04-07:
I can't access logs because it freezes during startup. I think i have given enough information to reproduce this bug.

lurch, 2025-04-28:
From the software-side of things, I don't think there's any reason why it'd work fine with a 32 GB SD card but not work with a 64 GB SD card. Is it possible that your 64 GB card is faulty? You can use software like https://h2testw.org/ or https://fight-flash-fraud.readthedocs.io/ to check your SD card is working correctly.