raspberrypi/bookworm-feedback #404: ffmpeg 8:5.1.6-0+deb12u1+rpt3 breaks deinterlace

aCaBctx, 2025-04-08:
Hi,
I've just upgraded ffmpeg and related packages from `8:5.1.6-0+deb12u1+rpt1` to `8:5.1.6-0+deb12u1+rpt3` which has resulted in the inability to play certain content in kodi.

After some tinkering it appears most things work except HD television streams for which the playback is delayed by several seconds and eventually only audio is reproduced.

Disabling deinterlacing (within kodi) makes them play again just fine (ofc if the stream is not actually interlaced).

Downgrading to `-rpt1` entirely fixes the issue.

Willing to provide more details if needed.

Thank you

popcornmix, 2025-04-08:
We do provide kodi and kodi21 packages. Does kodi21 work for you?

aCaBctx, 2025-04-08:
Hi,
indeed upgrading to kodi21 (and restoring ffmpeg `-rpt3`) results in playable HD streams.

Thanks

lurch, 2025-04-28:
Can this issue be closed @aCaBctx ?

closed 2025-04-28 by popcornmix