raspberrypi/bookworm-feedback #405: Please disable -fstack-clash-protection in gcc on armhf

andrew-sayers, 2025-04-09:
gcc lets you set `-fstack-clash-protection` on Pi armhf bookworm, but doing so causes valgrind errors even in trivial programs:

```sh
$ gcc -fstack-clash-protection -x c - <<EOF
void empty_function() {}
int main() {
  empty_function();
  return 0;
}
EOF
$ valgrind ./a.out
==19138== Memcheck, a memory error detector
==19138== Copyright (C) 2002-2022, and GNU GPL'd, by Julian Seward et al.
==19138== Using Valgrind-3.19.0 and LibVEX; rerun with -h for copyright info
==19138== Command: ./a.out
==19138==
==19138== Invalid write of size 4
==19138==    at 0x1041C: main (in /home/andrew/a.out)
==19138==  Address 0x7db5f2a0 is on thread 1's stack
==19138==  8 bytes below stack pointer
==19138==
==19138==
==19138== HEAP SUMMARY:
==19138==     in use at exit: 0 bytes in 0 blocks
==19138==   total heap usage: 0 allocs, 0 frees, 0 bytes allocated
==19138==
==19138== All heap blocks were freed -- no leaks are possible
==19138==
==19138== For lists of detected and suppressed errors, rerun with: -s
==19138== ERROR SUMMARY: 1 errors from 1 contexts (suppressed: 0 from 0)
```

[This does not occur on Debian armhf](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1102495#10), and cannot be fixed there.

Relatedly, `dpkg-dev` sets `-fstack-clash-protection` by default, and therefore generates packages with broken binaries.  Fixing this bug might fix that bug, or might make it even worse.

andrew-sayers, 2025-04-10:
Update: although [the report linked above](https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1102495) has been closed, a new message gives useful background information.  You may want to subscribe over there in case there are further messages.

XECDesign, 2025-04-10:
I think the right answer is to fix valgrind rather than disable hardening options, but in either case this would be a [Raspbian issue](https://bugs.launchpad.net/raspbian/).

timg236, 2025-04-10:
"Relatedly, dpkg-dev sets -fstack-clash-protection by default, and therefore generates packages with broken binaries."

Which packages are broken?

lurch, 2025-04-14:
> Could you update [this project's README](https://github.com/raspberrypi/bookworm-feedback/blob/main/README.md) to link to Raspbian, and to explain which bugs should be reported where? It sounds like the rule is that this project is for architecture-neutral and/or arm64-specific bugs, and armhf-specific bugs go over there?

There are no hard-and-fast rules about where bugs should be reported; but any issues that get reported here will be investigated by a Raspberry Pi engineer, and acted on accordingly (even if that "action" is to advise the OP where that issue ought to be reported to).

Closing this issue, as the problem is with upstream valgrind rather than Raspberry Pi OS.

closed 2025-04-14 by lurch