raspberrypi/bookworm-feedback #407: Realtek RTL8812BU

parkay6666, 2025-04-15:
my usb wi-fi adapter does not seem to be recognized anymore by the raspberry pi 5 after some recent updates.

i just did a full reimage with the latest 64-bit version and i was able to successfully connect to my local wi-fi. if i do a full-upgrade and reboot i get a message that 'no network devices available'. if i re-image and manually install all updates except for linux-headers-*, linux-image-* and linux-libc-dev numbered 2712, i am still able to connect to my local wi-fi network.

after installing the 2712 linux headers/image and rebooting my network adapter is no longer detected. and i do have the line in my config.txt to disable the onboard wi-fi device. i have been using this wi-fi adapter for quite a while as my pi 5 2gb is in a metal case.

thanx

lurch, 2025-04-16:
Yeah, we recently bumped our Linux kernel from 6.6 to 6.12, so that's likely the (unintentional) reason why your WiFi adaptor has stopped working.
@popcornmix Could you transfer this to the Linux repo please?

parkay6666, 2025-04-16:
thanx

6by9, 2025-04-17:
Link to forum thread - https://forums.raspberrypi.com/viewtopic.php?t=386736

lurch, 2025-04-23:
Forum thread identifies this as an upstream Linux kernel driver issue, and not something specific _just_ to Raspberry Pi OS, so I'll close this issue for now.

closed 2025-04-23 by lurch