raspberrypi/bookworm-feedback #410: Can't install kernels or headers - dpkg ovpn-dco compile fails

chindokae, 2025-04-21:
On an rpi5 running bookworm, I am unable to install or remove kernel headers due to compilation failures in ovpn-dco.

According to this post, it should be fixed:   https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1089502

I can't update ovpn-dco or remove it due to the compilation failures.   linux-headers are the same.  The compilation failure blocks all attempts to upgrade or remove the package.

Any advice on how to force the removal would be appreciated.

Here is the data:

lsb_release -a
No LSB modules are available.
Distributor ID: Debian
Description:    Debian GNU/Linux 12 (bookworm)
Release:        12
Codename:       bookworm

dpkg --print-architecture
arm64

file /bin/ls
/bin/ls: ELF 64-bit LSB pie executable, ARM aarch64, version 1 (SYSV), dynamically linked, interpreter /lib/ld-linux-aarch64.so.1, BuildID[sha1]=9f127c37a4c459cf01639f6ded2fcf11a49d3da9, for GNU/Linux 3.7.0, stripped

Setting up openvpn-dco-dkms (0.0+git20231103-1~deb12u1) ...
Removing old ovpn-dco-0.0+git20231103 DKMS files...
Deleting module ovpn-dco-0.0+git20231103 completely from the DKMS tree.
Loading new ovpn-dco-0.0+git20231103 DKMS files...
Building for 6.12.20+rpt-rpi-2712 6.12.20+rpt-rpi-v8
Building initial module for 6.12.20+rpt-rpi-2712
Error! Bad return status for module build on kernel: 6.12.20+rpt-rpi-2712 (aarch64)
Consult /var/lib/dkms/ovpn-dco/0.0+git20231103/build/make.log for more information.
dpkg: error processing package openvpn-dco-dkms (--configure):
 installed openvpn-dco-dkms package post-installation script subprocess returned error exit status 10
Errors were encountered while processing:
 openvpn-dco-dkms
E: Sub-process /usr/bin/dpkg returned an error code (1)

cat /var/lib/dkms/ovpn-dco/0.0+git20231103/build/make.log
DKMS make.log for ovpn-dco-0.0+git20231103 for kernel 6.12.20+rpt-rpi-2712 (aarch64)
Mon 21 Apr 16:48:45 EDT 2025
/var/lib/dkms/ovpn-dco/0.0+git20231103/build/gen-compat-autoconf.sh /var/lib/dkms/ovpn-dco/0.0+git20231103/build/compat-autoconf.h
make -C /lib/modules/6.12.20+rpt-rpi-2712/build M=/var/lib/dkms/ovpn-dco/0.0+git20231103/build PWD=/var/lib/dkms/ovpn-dco/0.0+git20231103/build REVISION=0.0+git20231103 CONFIG_OVPN_DCO_V2=m INSTALL_MOD_DIR=updates/  modules
make[1]: Entering directory '/usr/src/linux-headers-6.12.20+rpt-rpi-2712'
  CC [M]  /var/lib/dkms/ovpn-dco/0.0+git20231103/build/drivers/net/ovpn-dco/main.o
  CC [M]  /var/lib/dkms/ovpn-dco/0.0+git20231103/build/drivers/net/ovpn-dco/bind.o
  CC [M]  /var/lib/dkms/ovpn-dco/0.0+git20231103/build/drivers/net/ovpn-dco/crypto.o
  CC [M]  /var/lib/dkms/ovpn-dco/0.0+git20231103/build/drivers/net/ovpn-dco/ovpn.o
  CC [M]  /var/lib/dkms/ovpn-dco/0.0+git20231103/build/drivers/net/ovpn-dco/peer.o
  CC [M]  /var/lib/dkms/ovpn-dco/0.0+git20231103/build/drivers/net/ovpn-dco/sock.o
/var/lib/dkms/ovpn-dco/0.0+git20231103/build/drivers/net/ovpn-dco/main.c: In function ‘ovpn_setup’:
/var/lib/dkms/ovpn-dco/0.0+git20231103/build/drivers/net/ovpn-dco/main.c:129:47: error: ‘NETIF_F_LLTX’ undeclared (first use in this function); did you mean ‘NETIF_F_LRO’?
  129 |         netdev_features_t feat = NETIF_F_SG | NETIF_F_LLTX |
      |                                               ^~~~~~~~~~~~
      |                                               NETIF_F_LRO
/var/lib/dkms/ovpn-dco/0.0+git20231103/build/drivers/net/ovpn-dco/main.c:129:47: note: each undeclared identifier is reported only once for each function it appears in
In file included from /var/lib/dkms/ovpn-dco/0.0+git20231103/build/drivers/net/ovpn-dco/crypto.h:16,
                 from /var/lib/dkms/ovpn-dco/0.0+git20231103/build/drivers/net/ovpn-dco/addr.h:13,
                 from /var/lib/dkms/ovpn-dco/0.0+git20231103/build/drivers/net/ovpn-dco/peer.h:13,
                 from /var/lib/dkms/ovpn-dco/0.0+git20231103/build/drivers/net/ovpn-dco/ovpn.h:14,
                 from /var/lib/dkms/ovpn-dco/0.0+git20231103/build/drivers/net/ovpn-dco/main.c:12:
/var/lib/dkms/ovpn-dco/0.0+git20231103/build/drivers/net/ovpn-dco/main.c: At top level:
/var/lib/dkms/ovpn-dco/0.0+git20231103/build/include/uapi/linux/ovpn_dco.h:14:22: error: expected declaration specifiers or ‘...’ before string constant
   14 | #define OVPN_NL_NAME "ovpn-dco-v2"
      |                      ^~~~~~~~~~~~~
/var/lib/dkms/ovpn-dco/0.0+git20231103/build/drivers/net/ovpn-dco/main.c:263:26: note: in expansion of macro ‘OVPN_NL_NAME’
  263 | MODULE_ALIAS_GENL_FAMILY(OVPN_NL_NAME);
      |                          ^~~~~~~~~~~~
make[4]: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/scripts/Makefile.build:249: /var/lib/dkms/ovpn-dco/0.0+git20231103/build/drivers/net/ovpn-dco/main.o] Error 1
make[4]: *** Waiting for unfinished jobs....
make[3]: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/scripts/Makefile.build:490: /var/lib/dkms/ovpn-dco/0.0+git20231103/build/drivers/net/ovpn-dco] Error 2
make[2]: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/Makefile:1969: /var/lib/dkms/ovpn-dco/0.0+git20231103/build] Error 2
make[1]: *** [/usr/src/linux-headers-6.12.20+rpt-common-rpi/Makefile:236: __sub-make] Error 2
make[1]: Leaving directory '/usr/src/linux-headers-6.12.20+rpt-rpi-2712'
make: *** [Makefile:59: all] Error 2

 apt policy
Package files:
 100 /var/lib/dpkg/status
     release a=now
 500 https://archive.raspberrypi.com/debian bookworm/main armhf Packages
     release o=Raspberry Pi Foundation,a=stable,n=bookworm,l=Raspberry Pi Foundation,c=main,b=armhf
     origin archive.raspberrypi.com
 500 https://archive.raspberrypi.com/debian bookworm/main arm64 Packages
     release o=Raspberry Pi Foundation,a=stable,n=bookworm,l=Raspberry Pi Foundation,c=main,b=arm64
     origin archive.raspberrypi.com
 500 https://deb.debian.org/debian bookworm-updates/main armhf Packages
     release v=12-updates,o=Debian,a=stable-updates,n=bookworm-updates,l=Debian,c=main,b=armhf
     origin deb.debian.org
 500 https://deb.debian.org/debian bookworm-updates/main arm64 Packages
     release v=12-updates,o=Debian,a=stable-updates,n=bookworm-updates,l=Debian,c=main,b=arm64
     origin deb.debian.org
 500 https://deb.debian.org/debian-security bookworm-security/contrib armhf Packages
     release v=12,o=Debian,a=stable-security,n=bookworm-security,l=Debian-Security,c=contrib,b=armhf
     origin deb.debian.org
 500 https://deb.debian.org/debian-security bookworm-security/contrib arm64 Packages
     release v=12,o=Debian,a=stable-security,n=bookworm-security,l=Debian-Security,c=contrib,b=arm64
     origin deb.debian.org
 500 https://deb.debian.org/debian-security bookworm-security/main armhf Packages
     release v=12,o=Debian,a=stable-security,n=bookworm-security,l=Debian-Security,c=main,b=armhf
     origin deb.debian.org
 500 https://deb.debian.org/debian-security bookworm-security/main arm64 Packages
     release v=12,o=Debian,a=stable-security,n=bookworm-security,l=Debian-Security,c=main,b=arm64
     origin deb.debian.org
 500 https://deb.debian.org/debian bookworm/non-free-firmware armhf Packages
     release v=12.10,o=Debian,a=stable,n=bookworm,l=Debian,c=non-free-firmware,b=armhf
     origin deb.debian.org
 500 https://deb.debian.org/debian bookworm/non-free-firmware arm64 Packages
     release v=12.10,o=Debian,a=stable,n=bookworm,l=Debian,c=non-free-firmware,b=arm64
     origin deb.debian.org
 500 https://deb.debian.org/debian bookworm/non-free armhf Packages
     release v=12.10,o=Debian,a=stable,n=bookworm,l=Debian,c=non-free,b=armhf
     origin deb.debian.org
 500 https://deb.debian.org/debian bookworm/non-free arm64 Packages
     release v=12.10,o=Debian,a=stable,n=bookworm,l=Debian,c=non-free,b=arm64
     origin deb.debian.org
 500 https://deb.debian.org/debian bookworm/contrib armhf Packages
     release v=12.10,o=Debian,a=stable,n=bookworm,l=Debian,c=contrib,b=armhf
     origin deb.debian.org
 500 https://deb.debian.org/debian bookworm/contrib arm64 Packages
     release v=12.10,o=Debian,a=stable,n=bookworm,l=Debian,c=contrib,b=arm64
     origin deb.debian.org
 500 https://deb.debian.org/debian bookworm/main armhf Packages
     release v=12.10,o=Debian,a=stable,n=bookworm,l=Debian,c=main,b=armhf
     origin deb.debian.org
 500 https://deb.debian.org/debian bookworm/main arm64 Packages
     release v=12.10,o=Debian,a=stable,n=bookworm,l=Debian,c=main,b=arm64
     origin deb.debian.org
Pinned packages:

lurch, 2025-04-23:
> According to this post, it should be fixed: https://bugs.debian.org/cgi-bin/bugreport.cgi?bug=1089502

Yeah, fixed in `0.0+git20241121-1`, but as you can see from your logs, Raspberry Pi OS is using `0.0+git20231103`, which is the same version as in [upstream Debian Bookworm](https://packages.debian.org/search?keywords=openvpn-dco-dkms). This problem only occurs downstream in Raspberry Pi OS Bookworm (but not in upstream Debian Bookworm) because we've switched to using the 6.12 Linux kernel.
I think the "proper" fix here would be for us to backport the fix for `openvpn-dco-dkms` to Raspberry Pi OS Bookworm, but we're hoping to release a version of Raspberry Pi OS Trixie at some point, so you might be better off simply waiting for that?

ping @XECDesign for his thoughts.

XECDesign, 2025-04-24:
Backported a newer version of `openvpn-dco-dkms` to Bookworm and that seems to work for me. It should be in apt now.

lurch, 2025-04-24:
Can you test if this works for you please @chindokae and then close this issue if so? Thanks!

chindokae, 2025-04-24:
I will try shortly.

chindokae, 2025-04-24:
That installed and built cleanly.  Thank you all very much.