raspberrypi/bookworm-feedback #411: [BUG]: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE.

karuto12, 2025-04-23:
### What happened?

I am trying to install ultralytics and opencv but i got this error of hash mismatch; but i was able to install opencv using sudo apt install but cannot do so for ultralytics.

I tried
- purging the cache
- updating pip, setuptools
- also tried [stackflow](https://stackoverflow.com/questions/40183108/pip-says-these-packages-do-not-match-the-hashes-from-the-requirements-file-if)

i was not sure where to put this error; hope this is correct; please can someone help

### Version

1.9.2 (Default)

### What host operating system were you using?

Windows

### Host OS Version

Version 24H2 (OS Build 2600.3775)

### Selected OS

Raspbian GNU/Linux 32-bit

### Which Raspberry Pi Device are you using?

Raspberry Pi 4B, 400, and Compute Modules 4, 4S

### What kind of storage device are you using?

USB Pen Drive

### OS Customisation

- [ ] Yes, I was using OS Customisation when the bug occurred.

### Relevant log output

```shell
(.venv) pi@raspberrypi:~/Desktop/app $ pip install --upgrade ultralytics
Looking in indexes: https://pypi.org/simple, https://www.piwheels.org/simple
Collecting ultralytics
  Using cached ultralytics-8.3.114-py3-none-any.whl (983 kB)
Collecting numpy<=2.1.1,>=1.23.0
  Using cached https://www.piwheels.org/simple/numpy/numpy-2.1.1-cp311-cp311-linux_armv7l.whl (6.0 MB)
Collecting matplotlib>=3.3.0
  Downloading https://www.piwheels.org/simple/matplotlib/matplotlib-3.10.1-cp311-cp311-linux_armv7l.whl (8.2 MB)
     â”â”â”â”â”â”â”â”â”â”â”â”â”â”â”â”â”â”â”â”â”â•ºâ”â”â”â”â”â”â”â”â”â”â”â”â”â”â”â”â”â” 4.4/8.2 MB 115.9 kB/s eta 0:00:33
ERROR: THESE PACKAGES DO NOT MATCH THE HASHES FROM THE REQUIREMENTS FILE. If you have updated the package versions, please update the hashes. Otherwise, examine the package contents carefully; someone may have tampered with them.
    matplotlib>=3.3.0 from https://www.piwheels.org/simple/matplotlib/matplotlib-3.10.1-cp311-cp311-linux_armv7l.whl#sha256=fbb27b5baaecf602de781a8305cec2798cf0a432696aa99665de15fc66b16a45 (from ultralytics):
        Expected sha256 fbb27b5baaecf602de781a8305cec2798cf0a432696aa99665de15fc66b16a45
             Got        35121efc8756bfedcd968a565705b023cf34fe0bd6ba1ac2cd356716c3478c9c
```

tdewey-rpi, 2025-04-23:
*waves wand*

And in to the correct repository this goes!

Thanks for the report, @karuto12. Unfortunately if you're reporting an issue with an OS provided by rpi-imager, you _must_ report that issue against the OS, not against rpi-imager. In this case I have the power to fix the mistake - but for other OSes I don't have that power, and would have been forced to close this as 'not relevant'.

lurch, 2025-04-23:
Appears to be an issue with https://www.piwheels.org which is outside of Raspberry Pi's control, so you should file a bug-report at https://github.com/piwheels/packages

closed 2025-04-23 by lurch