raspberrypi/bookworm-feedback #413: Raspberry Pi Connect Screen sharing incorrect keyboard mapping

einhugur, 2025-05-13:
Raspberry Pi Connect Screen sharing incorrect keyboard mapping.

If you have Icelandic keyboard on Raspberry Pi and Icelandic Keyboard on Windows that is connecting.

Then Shift 0 (Shift Zero) gives "-" but it should be giving "="

No key will give "="

This has also been longstanding issue with VNC into Raspberry PI, there is just no way to type "=" in remote session if you have Icelandic keyboard.

any1, 2025-07-23:
This should actually work if you have the same keyboard layout on both the server and client side. There are issues if you have multiple keyboard layouts on the server, but multiple layouts are actually supported on wayvnc git master.

rafaeln, 2025-07-24:
It should but it doesn't.

any1, 2025-07-24:
Actually, dvorak is a different issue that that of the original poster. The problem is that the keyboard variant doesn't get passed along in `wayvnc-run.sh`

any1, 2025-07-25:
@einhugur, is your system fully up to date?

Are you using labwc or wayfire?

What does your `/etc/wayvnc/config` look like?

What is the output from `ps ax | grep wayvnc`?

any1, 2025-07-26:
> Only remaining issue is that 2 stroke keys do not work like í ó, ú , etc

Dead keys work for me both with Tiger VNC viewer and Real VNC viewer. What are you using?

any1, 2025-07-26:
> The report is about Raspberry's Pi's own Pi-Connect. So no viewer. Using Pi-Connect viewer system.

Ahh, forgot about that. I can also type é on Connect. I am using Firefox on Linux.

einhugur, 2025-07-26:
I was connecting from Firefox on windows.