raspberrypi/bookworm-feedback #414: Chromium with graphics acceleration shows a white screen when playing a video

wuhk76, 2025-05-13:
Description:

     I have a Raspberry Pi 4 model B (6.12.25+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.12.25-1+rpt1 (2025-04-30) aarch64 GNU/Linux). When I try to play a YouTube video in Chromium version 135 or 136 with graphics acceleration on, it just shows a white screen, but it still has sound. It is interesting because if I turn on graphics acceleration, it does display the video. Also, it is only the Gnome interface it has this problem; it will work in the raspberry pi desktop. Please note that the previous Chromium verson 134, does not have this error on gnome. I suspect it has something to do with the 2d graphics library. When I run Chromium in the terminal with no flags in the terminal, and watch a video, it gives me this error: "ERROR:gpu/command_buffer/service/shared_image/shared_image_manager.cc:408] SharedImageManager::ProduceSkia: Trying to produce a Skia representation from an incompatible backing: CompoundImageBacking." this error tells me that that the graphics library, Skia,  is not compatible with the GPU. I understand that I can switch to an x based interface, but i prefer Gnome for its accessibility, such as its onscreen keyboard. I also want the ability for graphics acceleration for videos because I often use my Raspberry Pi for displaying videos on large TV monitors.

Steps to recreate problem:
     I have tested this error on a fresh install of Raspberry Pi OS, and it gives the same issue.
   1. Install the latest version of Raspberry Pi OS on a Raspberry Pi 4 Model B. I am using Bookworm.
   2. Update repositories: sudo apt update @ sudo apt upgrade.

[image]

   3. Install gnome: sudo apt install gnome. Note that the display manager should be set to gdm3 if asked for in the installation. 
   4. Reboot to start gnome: sudo reboot.
   5. Install Chromium: sudo apt install chromium.
   6. Enable graphics acceleration if needed. It should be in the system option of Chromium settings. 
   7. Run a video in Chromium, such as with YouTube. It should show a white screen, but plays sound. If run in the terminal with command 
    "chromium" it should give the same skia error when displaying a video.

XECDesign, 2025-05-14:
Support for chromium is mostly just limited to labwc in a standard Raspberry Pi OS desktop environment. If there's a known fix, I'm happy to apply it, but it's not something we can investigate right now. I'm guessing Gnome's wayland compositor is missing some required features, but that isn't an environment we test or support.

lurch, 2025-06-30:
> it will work in the raspberry pi desktop

In that case I'm going to have to close this issue. I'm afraid we only have the resources to support the standard labwc desktop environment on the official Raspberry Pi OS images. Anything else is outside of the scope of this repo.

closed 2025-06-30 by lurch

dopreo, 2025-07-05:
I seem to be experiencing this very same issue on Chromium on my Pi 500, despite being under labwc.
The audio plays just fine, but the video remains a white screen.

When running the Chromium browser from the Pi's terminal, the following error appears repeatedly while trying to play back video from youtube.com:
```[2506:2506:0705/105958.891271:ERROR:gpu/command_buffer/service/shared_image/shared_image_manager.cc:396] SharedImageManager::ProduceSkia: Trying to produce a Skia representation from an incompatible backing: CompoundImageBacking```

This is in the standard **Raspberry Pi Desktop** desktop environment, under **labwc** (_Wayland_).
I'd appreciate any advice, and would absolutely be more than happy to provide further diagnostic information.

---

### As reported by `chrome://gpu`:
<details>
    <summary>Graphics Feature Status, Version Information, et cetera</summary>

```
Graphics Feature Status
=======================
*   Canvas: Hardware accelerated
*   Direct Rendering Display Compositor: Disabled
*   Compositing: Hardware accelerated
*   Multiple Raster Threads: Enabled
*   OpenGL: Enabled
*   Rasterization: Hardware accelerated on all pages
*   Raw Draw: Disabled
*   Skia Graphite: Disabled
*   TreesInViz: Disabled
*   Video Decode: Hardware accelerated
*   Video Encode: Software only. Hardware acceleration disabled
*   Vulkan: Disabled
*   WebGL: Hardware accelerated
*   WebGL2: Hardware accelerated
*   WebGPU: Disabled
*   WebNN: Disabled

Version Information
===================
Data exported              : 2025-07-05T10:08:06.366Z
Chrome version             : Chrome/138.0.7204.92
Operating system           : Linux 6.12.34+rpt-rpi-2712
Software rendering list URL: https://chromium.googlesource.com/chromium/src/+/f079b9bc781e3c2adb1496ea1d72812deb0ddb3d/gpu/config/software_rendering_list.json
Driver bug list URL        : https://chromium.googlesource.com/chromium/src/+/f079b9bc781e3c2adb1496ea1d72812deb0ddb3d/gpu/config/gpu_driver_bug_list.json
ANGLE commit id            : unknown hash
2D graphics backend        : Skia/138 a46d5732d9fca93eaec23e502e2eef814b707e6b
Command Line               : /usr/lib/chromium/chromium --js-flags=--no-decommit-pooled-pages --force-renderer-accessibility --enable-remote-extensions --show-component-extension-options --enable-gpu-rasterization --no-default-browser-check --disable-pings --media-router=0 --enable-remote-extensions --load-extension --use-angle=gles --flag-switches-begin --cast-streaming-force-enable-hardware-h264 --cast-streaming-force-enable-hardware-vp8 --cast-streaming-force-enable-hardware-vp9 --ignore-gpu-blocklist --ozone-platform-hint=auto --top-chrome-touch-ui=disabled --enable-features=CapturedSurfaceControl,LinkPreview,ParallelDownloading,TouchTextEditingRedesign,WaylandPerSurfaceScale,WaylandTextInputV3 --disable-features=CWSInfoFastCheck --flag-switches-end --ozone-platform=wayland

Driver Information
==================
- Initialization time: 92
- In-process GPU: false
- Skia Backend: GaneshGL
- Passthrough Command Decoder: false
- Sandboxed: false
- GPU0: VENDOR= 0x0000 [Google Inc. (Broadcom)], DEVICE=0x0000 [ANGLE (Broadcom, V3D 7.1.10.2, OpenGL ES 3.1 Mesa 24.2.8-1~bpo12+rpt3)], DRIVER_VENDOR=Mesa, DRIVER_VERSION=24.2.8 *ACTIVE*
- Optimus: false
- AMD switchable: false
- Pixel shader version: 1.00
- Vertex shader version: 1.00
- Max. MSAA samples: 4
- Machine model name: 
- Machine model version: 
- GL implementation parts: (gl=egl-angle,angle=opengles)
- Display type: ANGLE_OPENGLES
- GL_VENDOR: Google Inc. (Broadcom)
- GL_RENDERER: ANGLE (Broadcom, V3D 7.1.10.2, OpenGL ES 3.1 Mesa 24.2.8-1~bpo12+rpt3)
- GL_VERSION: OpenGL ES 2.0.0 (ANGLE 2.1 git hash: unknown hash)
- GL_EXTENSIONS: GL_AMD_performance_monitor GL_ANGLE_base_vertex_base_instance_shader_builtin GL_ANGLE_blob_cache GL_ANGLE_client_arrays GL_ANGLE_compressed_texture_etc GL_ANGLE_depth_texture GL_ANGLE_framebuffer_blit GL_ANGLE_framebuffer_multisample GL_ANGLE_get_serialized_context_string GL_ANGLE_get_tex_level_parameter GL_ANGLE_instanced_arrays GL_ANGLE_memory_size GL_ANGLE_program_binary_readiness_query GL_ANGLE_program_cache_control GL_ANGLE_request_extension GL_ANGLE_robust_client_memory GL_ANGLE_texture_compression_dxt3 GL_ANGLE_texture_compression_dxt5 GL_ANGLE_texture_external_update GL_ANGLE_translated_shader_source GL_ARM_rgba8 GL_CHROMIUM_bind_generates_resource GL_CHROMIUM_bind_uniform_location GL_CHROMIUM_color_buffer_float_rgba GL_CHROMIUM_copy_texture GL_CHROMIUM_lose_context GL_CHROMIUM_sync_query GL_EXT_base_instance GL_EXT_blend_minmax GL_EXT_clear_texture GL_EXT_color_buffer_half_float GL_EXT_compressed_ETC1_RGB8_sub_texture GL_EXT_debug_label GL_EXT_debug_marker GL_EXT_depth_clamp GL_EXT_discard_framebuffer GL_EXT_draw_buffers GL_EXT_draw_elements_base_vertex GL_EXT_float_blend GL_EXT_frag_depth GL_EXT_instanced_arrays GL_EXT_map_buffer_range GL_EXT_occlusion_query_boolean GL_EXT_polygon_offset_clamp GL_EXT_read_format_bgra GL_EXT_sRGB GL_EXT_sRGB_write_control GL_EXT_shadow_samplers GL_EXT_texture_border_clamp GL_EXT_texture_compression_dxt1 GL_EXT_texture_compression_rgtc GL_EXT_texture_compression_s3tc_srgb GL_EXT_texture_filter_anisotropic GL_EXT_texture_format_BGRA8888 GL_EXT_texture_mirror_clamp_to_edge GL_EXT_texture_rg GL_EXT_texture_sRGB_decode GL_EXT_texture_storage GL_EXT_texture_type_2_10_10_10_REV GL_EXT_unpack_subimage GL_KHR_debug GL_KHR_parallel_shader_compile GL_KHR_texture_compression_astc_ldr GL_KHR_texture_compression_astc [... cut ...]
ii  libglx-mesa0:arm64                   24.2.8-1~bpo12+rpt3                 arm64        free implementation of the OpenGL API -- GLX vendor library
ii  mesa-libgallium:arm64                24.2.8-1~bpo12+rpt3                 arm64        shared infrastructure for Mesa drivers
ii  mesa-vdpau-drivers:arm64             24.2.8-1~bpo12+rpt3                 arm64        Mesa VDPAU video acceleration drivers
ii  mesa-vulkan-drivers:arm64            24.2.8-1~bpo12+rpt3                 arm64        Mesa Vulkan graphics drivers
```

`pi@raspberrypi:~ $ `
</details>

---

*Thank you so much!*