raspberrypi/bookworm-feedback #415: systemd journal config changed from Storage=auto to Storage=persistent

gitbls, 2025-05-25:
### Describe the bug

In the 2024-11-13 and earlier IMGs /etc/systemd/journald.conf had `#Storage=auto`, which by default set the journald storage correctly conditioned on the existence of /var/log/journal

In the 2025-05-13 IMG that setting is changed to `Storage=volatile`.

What was the motivation for this change?

It seems like a less-than-optimal choice for problem solving since logs from all prior boots are lost by default. With the prior setting, those who wanted volatile logs for system optimization reasons are more likely to know to change it.

If this was an intentional change, might I suggest that you revert the change to /etc/systemd/journald.conf and make the modification the correct way:
```
mkdir -p /var/log/journal
mkdir -p /etc/systemd/journald.conf.d
cat > /etc/systemd/journald.conf.d/xxx-raspi-volatile-journal.conf<<EOF
Storage=volatile
EOF

Thanks!

### Steps to reproduce the behaviour

Boot a system created from an 2025-05-13 IMG. Examine /etc/systemd/journald.conf

### Device (s)

Raspberry Pi 5

### System

```
pw/ssdy/work# sdm --explore vanilla/2025-05-06-raspios-bookworm-arm64.img 
* Mount IMG 'vanilla/2025-05-06-raspios-bookworm-arm64.img'
mount: /dev/loop0 mounted on /mnt/sdm.
mount: /dev/loop1 mounted on /mnt/sdm/boot/firmware.
* Enter IMG 'vanilla/2025-05-06-raspios-bookworm-arm64.img'
root@sdm:/# cd /etc/systemd
root@sdm:/etc/systemd# cat journald.conf 
#  This file is part of systemd.
#
#  systemd is free software; you can redistribute it and/or modify it under the
#  terms of the GNU Lesser General Public License as published by the Free
#  Software Foundation; either version 2.1 of the License, or (at your option)
#  any later version.
#
# Entries in this file show the compile time defaults. Local configuration
# should be created by either modifying this file, or by creating "drop-ins" in
# the journald.conf.d/ subdirectory. The latter is generally recommended.
# Defaults can be restored by simply deleting this file and all drop-ins.
#
# Use 'systemd-analyze cat-config systemd/journald.conf' to display the full config.
#
# See journald.conf(5) for details.

[Journal]
Storage=volatile
#Compress=yes
#Seal=yes
#SplitMode=uid
#SyncIntervalSec=5m
#RateLimitIntervalSec=30s
#RateLimitBurst=10000
#SystemMaxUse=
#SystemKeepFree=
#SystemMaxFileSize=
#SystemMaxFiles=100
#RuntimeMaxUse=
#RuntimeKeepFree=
#RuntimeMaxFileSize=
#RuntimeMaxFiles=100
#MaxRetentionSec=
#MaxFileSec=1month
#ForwardToSyslog=yes
#ForwardToKMsg=no
#ForwardToConsole=no
#ForwardToWall=yes
#TTYPath=/dev/console
#MaxLevelStore=debug
#MaxLevelSyslog=debug
#MaxLevelKMsg=notice
#MaxLevelConsole=info
#MaxLevelWall=emerg
#LineMax=48K
#ReadKMsg=yes
#Audit=no
```

popcornmix, 2025-05-25:
Moved issue to bookworm feedback as it's not related to the linux kernel.

XECDesign, 2025-05-27:
I may have some of this wrong, but I think the decision came about when we were looking at reducing overhead and improving boot times. On a well-used by the logs took up a significant chunk of the disk space and contributed to IO an bottleneck while booting.

Most users will not need to look at logs from previous boots, so it was decided that the overhead is not worth it.

While I like the notion of using a .d directory for the conf file, I worry it will lead to some confusion. If a user looks at the top level .conf file and sees one setting, but it gets overridden by another conf file they may not be aware of it will lead to extra issues being raised. We already see a fair bit of that with the sources.list.d files

If there is a strong argument for what makes a subdirectory the more "correct" way, I'm sure we can reconsider that aspect of it.

XECDesign, 2025-05-28:
Makes sense. @roliver-rpi, what do you think?

Changing the existing approach would be a bit of a pain, since it would require a package upgrade to automatically transfer the setting across to the .d directory and restore the original file and/or require raspi-config to work in either case.

We could leave bookworm as is and apply the change to trixie only. Or, we could say it's not worth the hassle and leave it as is.

lurch, 2025-05-28:
> There is definitely a user-education issue here though

Fortunately, the `/etc/systemd/journald.conf` (as displayed above) _does_ already say:
```
# should be created by either modifying this file, or by creating "drop-ins" in
# the journald.conf.d/ subdirectory. The latter is generally recommended.
```
(although the `/etc/systemd/journald.conf.d/` directory doesn't exist by default, which makes things slightly harder than they otherwise could be).

XECDesign, 2025-05-28:
> In an ideal world, we would set our default volatile storage policy in something like `/usr/lib/systemd/journald.conf.d/70-rpi-volatile-storage.conf` and this file would be owned by one of our configuration packages.

Alright, I'll can add it to sys-mods for Trixie.

gitbls, 2025-05-28:
> > In an ideal world, we would set our default volatile storage policy in something like `/usr/lib/systemd/journald.conf.d/70-rpi-volatile-storage.conf` and this file would be owned by one of our configuration packages.
> 
> Alright, I'll can add it to sys-mods for Trixie.

A fixie for Trixie is perfect, since it's coming soon. Thx!

gitbls, 2025-07-01:
I see that the latest Trixie bits via the beta forum have this fix applied in /lib/systemd/journal.conf.d /70-rpi-volatile-storage.conf

Thank you for making this change!

Interestingly, as I was testing it, I ran into an issue where my /etc/systemd/journal.conf.d/030-my.conf did not override the 70-rpi-volatile-storage.conf setting. 

After further investigation It appears that the _base filename_ is also included in the priority ordering decision. That is, /etc/systemd/journald.conf.d/030-my.conf setting is overridden by the /lib/ssytemd/journald.conf.d/70-rpi-....conf setting!

Blew me away! Am I the last one to know this? 😳

Notice in the first output 70... is processed before 999..., and in the second output the order is reversed. 

Output from `systemd-analyze cat-config systemd/journald.conf` with comments removed with my file named 999-my.conf
```

pit~# systemd-analyze cat-config systemd/journald.conf
 
[Journal]
Storage=volatile
 
# /usr/lib/systemd/journald.conf.d/70-rpi-volatile-storage.conf
[Journal]
Storage=volatile
 
# /etc/systemd/journald.conf.d/999-my.conf
[Journal]
Storage=persistent
ForwardToSyslog=no
 
# /usr/lib/systemd/journald.conf.d/syslog.conf
 
[Journal]
ForwardToSyslog=yes
pit~# 
 ```
And here's the output with it named 030-my.conf
```
pit~# systemd-analyze cat-config systemd/journald.conf
# /etc/systemd/journald.conf

[Journal]
Storage=volatile
 
# /etc/systemd/journald.conf.d/030-my.conf
[Journal]
Storage=persistent
ForwardToSyslog=no
 
# /usr/lib/systemd/journald.conf.d/70-rpi-volatile-storage.conf
[Journal]
Storage=volatile
 
# /usr/lib/systemd/journald.conf.d/syslog.conf
 
[Journal]
ForwardToSyslog=yes
pit~#

gitbls, 2025-07-01:
Two additional points:
* It appears that the default /etc/systemd/journald.conf still has the `Storage=volatile` in it. The Debian default is `#Storage=auto` so this should be reverted
* I did find [this systemd issue](https://github.com/systemd/systemd/issues/30439) that discusses this WRT sysctl, and there is a checkin that recommends using less than `50-*` for distros and reserve the upper names (60-90?) for users.

roliver-rpi, 2025-07-02:
The 10-40 guidance is certainly new to me. It looks like it was added to systemd docs for v256; bookworm ships v254.

If you're still on Bookworm, you can use [rpi-systemd-config](https://github.com/raspberrypi/rpi-systemd-config) in addition to 'systemd-analyze cat-config' to verify the final value that's being used by systemd, i.e:
```
$ rpi-systemd-config systemd/journald.conf JRNL_STRG Journal::Storage
JRNL_STRG='volatile'
```