raspberrypi/bookworm-feedback #420: Missing requirement for preinstalled picamera2 package (OpenCV)

dirkcuys, 2025-06-06:
When running the following error when running a python script to take pictures using pyimage2 with a fresh install of bookwork using the Raspberry Pi Imager

```
Traceback (most recent call last):
  File "/home/pi/pi.py", line 7, in <module>
    from picamera2.devices.imx708 import IMX708
  File "/usr/lib/python3/dist-packages/picamera2/devices/__init__.py", line 6, in <module>
    from .imx500 import IMX500
  File "/usr/lib/python3/dist-packages/picamera2/devices/imx500/__init__.py", line 2, in <module>
    from .postprocess_efficientdet_lite0 import \
  File "/usr/lib/python3/dist-packages/picamera2/devices/imx500/postprocess_efficientdet_lite0.py", line 12, in <module>
    from picamera2.devices.imx500.postprocess import (
  File "/usr/lib/python3/dist-packages/picamera2/devices/imx500/postprocess.py", line 13, in <module>
    import cv2
ModuleNotFoundError: No module named 'cv2'
```

Here is the python code I ran:

```python
from picamera2.devices.imx708 import IMX708
from picamera2 import Picamera2
from libcamera import Transform
from libcamera import controls

if __name__ == "__main__":
    imx708 = IMX708(0)
    imx708.set_sensor_hdr_mode(True)
    imx708.close()
    picam2 = Picamera2()
    config = picam2.create_still_configuration()
    picam2.configure(config)
    picam2.start(show_preview=False)
    output_path = "image.jpg"
    picam2.capture_file(output_path)
```

I fixed the issue by installing openCV: `sudo apt install python3-opencv`

dirkcuys, 2025-06-10:
Thanks for the links! I'm not specifically using OpenCV, just reporting the issue :)

lurch, 2025-06-10:
> The imx500 post-processing libraries do require opencv, so this is expected.

In that case should we include `python3-opencv` in the OS image by default and/or add it as a dependency to one of the picamera2 packages?